esphome/issues #2224: TTGO ESP-WROVER-KIT ESP32 Camera not

severeski, 2021-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

core-2021.7.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

LilyGO TTGO T-Camera ESP32-WROVER

**ESPHome version (latest production, beta, dev branch)**

ESPHome latest/beta/dev

**Affected component:**

Camera

**Description of problem:**
Setup of the camera fails and so no image

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: doorbell
  platform: ESP32
  board: esp-wrover-kit

esp32_camera:
  name: Doorbell Camera
  external_clock:
    pin: GPIO23
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO39, GPIO36, GPIO23, GPIO18, GPIO15, GPIO4, GPIO14, GPIO5]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  resolution: 640x480
  jpeg_quality: 10
  vertical_flip: false
  horizontal_mirror: false

i2c:
  sda: GPIO21
  scl: GPIO22

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/doorbell.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/doorbell
Processing doorbell (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
[... 226 lines cut ...]
[21:49:47][C][ota:030]:   Address: doorbell.local:3232
[21:49:47][C][ota:032]:   Using Password.
[21:49:47][C][api:095]: API Server:
[21:49:47][C][api:096]:   Address: doorbell.local:6053
[21:49:47][C][homeassistant.binary_sensor:037]: Homeassistant Binary Sensor 'Doorbell Copy Door Relay'
[21:49:48][C][homeassistant.binary_sensor:038]:   Entity ID: 'switch.door_lock_relay'
[21:49:48][C][wifi_signal.sensor:009]: WiFi Signal 'doorbell WiFi Signal'
[21:49:48][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:49:48][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:49:48][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:49:48][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:49:48][C][homeassistant.text_sensor:022]: Homeassistant Text Sensor 'occupancy'
[21:49:48][C][homeassistant.text_sensor:023]:   Entity ID: 'input_select.occupancy'
```

**Additional information and things you've tried:**
Different ESPhome versions of the firmware compiled on separate computer 1.17.x - 1.18.x - 1.19.x From 1.19.0 and up I get this Setup Failed: ERROR

Older versions used to work up to 1.18.x but now also not working, but no errors about hte camera

Used a new esp board strait out of the box (so tried two separate esp boards)

Add remove esphome and intergration

If you need me to try some other things I am all ears.

Sonusss, 2021-07-10:
Hi,

I had a similar problem setting up camera.
I discovered that there are many flavors of camera and boards and GPIOs are not always properly defined.
Check deeply about your real board schematic and camera model.
Also in my case I had to add this parameter to get it working: 
`power_down_pin: GPIO32`

severeski, 2021-07-10:
Hi,

Thanks

This was on the box it came in. As far as i can tell is my current config correct but I might be overlooking something.

[image]

Sonusss, 2021-07-10:
Indeed but the following declaration is missing in your YAML file:
`power_down_pin: GPIO26`

(If I can read it properly from your pic....)

wokkeltje13, 2021-07-10:
I have an ESP32-CAM and have the same errors while compiling.

```
Compiling /data/elektriciteitsmeter/.pioenvs/elektriciteitsmeter/FrameworkArduino/esp32-hal-time.c.o
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/elektriciteitsmeter/.pioenvs/elektriciteitsmeter/FrameworkArduino/esp32-hal-timer.c.o
```

I also don't have a video feed. The board was in the drawer for a while, but I am sure it was working before with an older version of ESPHome and HomeAssitant (sorry have no idea what version, but currently everything is up to date) I just updated it without checking if it was working before. (did test on 2 different boards, 1 used and 1 brand new.

Sonusss, 2021-07-10:
Sorry, but this topic have nothing to do with compiling !
It is related to flashed ESPhome build without camera proper setup...

drthanwho, 2021-07-19:
I have the same problem. It might be a HA issue as I have two of these flashed months ago and suddenly they are not showing a camera preview. All other sensors and internal entities work just fine. 

Trying to reflash one of them I get the camera ERRROR message in the logs

drthanwho, 2021-07-19:
Ok so seems I was correct. Latest HA Core 2021.7.3 fixes the issue. But there is also an additional issue regarding updating the firmware to the latest on the ESPHOME side that ends on an camera setup ERROR

severeski, 2021-07-19:
With the arduino version line as suggested above by @drthanwho my board does not boot anymore. No ip, no screen etc.

Everything works (minus the camera) with firmware version 1.17.2 on the board. 
HA version 20121.7.3 and esphome add-on version: 1.19.4

Unfortunately I do not have a backup for the HA add-on version 1.17.2

So if some might donate me a backup of the HA add-on version 1.17.2 I can check if the camera also works with that version so we can narrow some stuff down.

drthanwho, 2021-07-19:
That's weird. It worked on mine. And there was also a post on the forum where I got the Arduino tip from and I think it worked for them too

severeski, 2021-07-19:
I agree. I read that post too. And tried different things, this is the only combo that works for me (minus camera). Otherwise the board fails. 

And I tried it on two separately bought boards. One used, one brand new.

Do you have the same board?

(I clean build files etc between tries)

drthanwho, 2021-07-19:
Yeah I have the TTGO white ones. One with a fisheye lense and one with normal. It happened that I had taken them down recently for a few weeks so I didn't notice the problem during the beta for 7.0.

So today I got one online and couldn't get the camera entity to work. Tried updating the firmware just in case and then that's when I got the camera ERROR in the ESPHOME logs. Then for the other that I hadn't flashed and was running an older version, it worked as soon as I updated to HA Core 7.3 (was on 7.1). 

For the one that I updated the firmware the camera worked with me adding the older Arduino version in the firmware

severeski, 2021-07-19:
Well some progress, I compiled a fresh firmware 1.19.4  via the docker command line method (not the HA addon):

Inlcuded:
`
esphome:
  arduino_version: 1.0.2

`

Now everything works again, but still no camera. The error remains on both boards:
`
[E][esp32_camera:090]:   Setup Failed: ERROR
`

severeski, 2021-07-22:
Update: With the new version 1.20.0 still the same problem.

petosak, 2021-07-24:
Same problem....:(

mycortex, 2021-08-13:
> Same problem....:(
> Edit: After power off and power on ESP device - camera works for some hours.
> SW reset via ESPHome doesn't help.

This has allways been a problem with this board and its not related to this issue, in my opinion. I solved this with a template sensor that look at the MQTT debug. If it contains the word "Image" it's set to true, if not false.
`"{{ 'Image:' in states.sensor.pircam_kontor_mqtt_debug.state }}"`

Added the restart switch in ESPHome-config for the device. If sensor is false for more than one minute I trigger the restart. It has been stable since :) 

But there is something wrong here. All my excisting TTGO-cameras is working as it should, but if I add a new one with same config it wont work. I can see in the logs that it sends images, but no camera entity found in HA. It does not even get discovered by HA and is not listed under ESPHome in Integrations. In fact none of my new devices in ESPHome get listed in the integration. I can find the sensors in Developer Tools-States, but no camera.

pixeldoc2000, 2021-11-05:
I have a simular Issus with Camera Setup Error.

It works with `arduino_version: 1.0.5` and ESPHome Version: v2021.10.3 (latest) for my setup.

Board: ESP32-CAM with OV2640 Camera

```yaml
esphome:
  platform: esp32
  board: esp-wrover-kit
  arduino_version: 1.0.5

esp32_camera:
  name: "${camera_name}"
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
```

Even without defined `reset_pin` .

pixeldoc2000, 2021-11-06:
Might be related to #435

severeski, 2021-11-10:
Info update.

I tried this and it 'seemed' to work.

After compile and upload in HA ESPHOME console the debug gave no error
during setup.

However. No camera output in HA.

After manually powering off/on the board. The same camera setup error in
the debug output of HA ESPHOME console.

mycortex, 2021-11-11:
I've had problems from time to time with this board, but no problems after I found that in my case it actually was related to the MQTT-component included in the code. When this component is removed it works. Issue: [2643](https://github.com/esphome/issues/issues/2643)

The solution for me was to add `setup_priority: -200` in the camera-component. After I found this solution the camera never fail loading. I don't exactly know, but I always have to be sure I use a good USB-supply to get this board working as it should. So I suspect it's related to power and that it's not that happy with starting all components at boot.

It's worth a try :) 

My code:
```
esp32_camera:
  name: $friendly_name Camera
  external_clock:
    pin: GPIO32
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO18, GPIO23, GPIO36, GPIO39]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  power_down_pin: GPIO26
  resolution: 800x600
  jpeg_quality: 10
  vertical_flip: true
  horizontal_mirror: true
  setup_priority: -200  #(when this line is added the camera loads fine. Without this line it fails)

i2c:
  sda: GPIO21
  scl: GPIO22
```

severeski, 2021-12-20:
I found my issue ... its a bit dumb as fuck.

Its right there in the original configuration ...     "pin: GPIO23" has to be 32

closed 2021-12-20 by severeski

crispywisp, 2022-01-05:
> I've had problems from time to time with this board, but no problems after I found that in my case it actually was related to the MQTT-component included in the code. When this component is removed it works. Issue: [2643](https://github.com/esphome/issues/issues/2643)
> 
> The solution for me was to add `setup_priority: -200` in the camera-component. After I found this solution the camera never fail loading. I don't exactly know, but I always have to be sure I use a good USB-supply to get this board working as it should. So I suspect it's related to power and that it's not that happy with starting all components at boot.
> 
> It's worth a try :)
> 
> My code (ESPHome version 2021.10.3):
> 
> ```
> esp32_camera:
>   name: $friendly_name Camera
>   external_clock:
>     pin: GPIO32
>     frequency: 20MHz
>   i2c_pins:
>     sda: GPIO13
>     scl: GPIO12
>   data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO18, GPIO23, GPIO36, GPIO39]
>   vsync_pin: GPIO27
>   href_pin: GPIO25
>   pixel_clock_pin: GPIO19
>   power_down_pin: GPIO26
>   resolution: 800x600
>   jpeg_quality: 10
>   vertical_flip: true
>   horizontal_mirror: true
>   setup_priority: -200  #(when this line is added the camera loads fine. Without this line it fails)
> 
> i2c:
>   sda: GPIO21
>   scl: GPIO22
> ```

Adding the arduino_version didn’t work for me. But adding setup_priority: -200 fixed my problem.