esphome/issues #2226: bme680_bsec crashes periotically

lanrat, 2021-07-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (NodeMCU)

**ESPHome version (latest production, beta, dev branch)**

Version: 1.19.4

**Affected component:**

bme680_bsec
https://esphome.io/components/sensor/bme680_bsec.html

**Description of problem:**
When using the bme680_bsec component, the ESP crashes periodically and reboots.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  node_name: bme
  STANDARD_ALTITUDE: '47.85'
  STANDARD_SEA_LEVEL_PRESSURE: '1013.25'
esphome:
  build_path: build/bme
  name: bme
  platform: ESP32
  board: nodemcu-32s
  arduino_version: platformio/espressif32@3.2.0
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
wifi:
  ap:
    ssid: bme_AP
    password: !secret 'hotspot_password'
    ap_timeout: 1min
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  networks:
  - ssid: !secret 'wifi_name'
    password: !secret 'wifi_password'
    priority: 0.0
  use_address: bme.local
time:
- platform: homeassistant
  timezone: PST8PDT7,M3.2.0/2,M11.1.0/2
  update_interval: 15min
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  level: DEBUG
  logs: {}
api:
  password: !secret 'ota_password'
  reboot_timeout: 0s
  port: 6053
ota:
  password: !secret 'ota_password'
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
[... 235 lines cut ...]
[16:07:13]entry 0x400806a8
[16:07:13][I][logger:170]: Log initialized
[16:07:13][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[16:07:13][I][app:029]: Running through setup()...
[16:07:13][D][text_sensor:015]: 'bme_version': Sending state '1.19.4 Jul  8 2021, 15:07:02'
[16:07:13][C][bme680_bsec.sensor:016]: Setting up BME680 via BSEC...
[16:07:13][C][wifi:037]: Setting up WiFi...
[16:07:13][D][wifi:365]: Starting scan...
[16:07:13][D][sensor:117]: 'bme_uptime': Sending state 0.21900 s with 0 decimals of accuracy
[16:07:13][D][sensor:117]: 'Equivalent sea level pressure': Sending state nan hPa with 1 decimals of accuracy
[16:07:13][D][sensor:117]: 'Absolute Humidity': Sending state nan g/m³ with 2 decimals of accuracy
[16:07:13][D][sensor:117]: 'Dew Point': Sending state nan °C with 1 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 IAQ': Sending state 25.00000 IAQ with 0 decimals of accuracy
[16:07:14][D][text_sensor:015]: 'BME680 IAQ Accuracy': Sending state 'Stabilizing'
[16:07:14][D][sensor:117]: 'BME680 CO2 Equivalent': Sending state 500.00000 ppm with 1 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 Breath VOC Equivalent': Sending state 0.50000 ppm with 1 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 Pressure': Sending state 1001.72998 hPa with 1 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 Gas Resistance': Sending state 37666.00000 Ω with 0 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 Temperature': Sending state 22.26000 °C with 1 decimals of accuracy
[16:07:14][D][sensor:117]: 'BME680 Humidity': Sending state 52.74001 % with 1 decimals of accuracy
[16:07:16][D][wifi:380]: Found networks:
```

**Additional information and things you've tried:**
I don't have this issue using the bme680 component, but bme680_bsec seems more accurate and would prefer to use it.

lanrat, 2021-07-09:
Got a few more crashes today. They seem to happen every 5-20 minutes.

It seems to have something to do with publishing updated states from the sensor.

```log
11:03:53][D][sensor:117]: 'Equivalent sea level pressure': Sending state 1010.87177 hPa with 1 decimals of accuracy
[11:03:55][D][sensor:117]: 'Absolute Humidity': Sending state 10.27654 g/m³ with 2 decimals of accuracy
[11:03:57][D][sensor:117]: 'Dew Point': Sending state 12.00801 °C with 1 decimals of accuracy
[11:03:59][D][sensor:117]: 'bme_uptime': Sending state 1246.82495 s with 0 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 IAQ': Sending state 27.78821 IAQ with 0 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 CO2 Equivalent': Sending state 511.15283 ppm with 1 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 Breath VOC Equivalent': Sending state 0.52152 ppm with 1 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 Pressure': Sending state 1005.26001 hPa with 1 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 Gas Resistance': Sending state 49442.00000 Ω with 0 decimals of accuracy
[11:04:00][D][sensor:117]: 'BME680 Temperature': Sending state 22.51180 °C with 1 decimals of accuracy
[11:04:00][D][sensor:117]: 'Temp F': Sending state 72.52124 °F with 1 decimals of accuracy
[11:04:00]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[11:04:00]Core 1 register dump:
[11:04:00]PC      : 0x400ec51e  PS      : 0x00060e30  A0      : 0x800ec5be  A1      : 0x3ffb18e0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d build/bme -t idedata
WARNING Decoded 0x400ec51e: AsyncEventSourceClient::_runQueue() at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[11:04:02]A2      : 0x3ffd1f40  A3      : 0x00000001  A4      : 0x0000007f  A5      : 0x3ffb3512  
[11:04:02]A6      : 0x0000000a  A7      : 0x1b796361  A8      : 0x800ec52c  A9      : 0x3ffb18c0  
[11:04:02]A10     : 0x0000007f  A11     : 0x3ffba8b3  A12     : 0x0000007f  A13     : 0x3ffb2788  
[11:04:02]A14     : 0x00000000  A15     : 0xad567800  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[11:04:02]EXCVADDR: 0x00000001  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[11:04:02]
[11:04:02]ELF file SHA256: 0000000000000000
[11:04:02]
[11:04:02]Backtrace: 0x400ec51e:0x3ffb18e0 0x400ec5bb:0x3ffb1900 0x400ec5db:0x3ffb1920 0x400ec631:0x3ffb1940 0x400db996:0x3ffb1970 0x400d8865:0x3ffb1990 0x400d89ad:0x3ffb19c0 0x400e18dd:0x3ffb19e0 0x400e190a:0x3ffb1a10 0x400da9ad:0x3ffb1a70 0x400daa31:0x3ffb1ad0 0x400e2454:0x3ffb1b00 0x40172c27:0x3ffb1b30 0x40172c03:0x3ffb1b60 0x400da93d:0x3ffb1b80 0x400da9c9:0x3ffb1bb0 0x400d967e:0x3ffb1c10 0x400d96aa:0x3ffb1c30 0x400daa38:0x3ffb1c60 0x400d6f2a:0x3ffb1c90 0x400d70c5:0x3ffb1cc0 0x400d72a7:0x3ffb1d20 0x400d74fb:0x3ffb1ed0 0x40081156:0x3ffb1f10 0x400dff71:0x3ffb1f60 0x400e289a:0x3ffb1f90 0x400f5cc1:0x3ffb1fb0 0x4008a546:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400ec51e: AsyncEventSourceClient::_runQueue() at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400ec5bb: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400ec5db: AsyncEventSourceClient::write(char const*, unsigned int) at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400ec631: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400db996: std::_Function_handler<void (int, char const*, char const*), esphome::web_server::WebServer::setup()::{lambda(int, char const*, char const*)#2}>::_M_invoke(std::_Any_data const&, int&&, char const*&&, int&&) at /home/lanrat/code/esphome/build/bme/src/esphome/components/web_server/web_server.cpp:68
 (inlined by) _M_invoke at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d8865: std::function<void (int, char const*, char const*)>::operator()(int, char const*, char const*) const at /home/lanrat/code/esphome/build/bme/src/esphome/components/logger/logger.cpp:175
 (inlined by) esphome::CallbackManager<void (int, char const*, char const*)>::call(int, char const*, char const*) at /home/lanrat/code/esphome/build/bme/src/esphome/core/helpers.h:216
 (inlined by) esphome::logger::Logger::log_message_(int, char const*, int) at /home/lanrat/code/esphome/build/bme/src/esphome/components/logger/logger.cpp:113
WARNING Decoded 0x400d89ad: esphome::logger::Logger::log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/lanrat/code/esphome/build/bme/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400e18dd: esphome::esp_log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/lanrat/code/esphome/build/bme/src/esphome/core/log.cpp:32
WARNING Decoded 0x400e190a: esphome::esp_log_printf_(int, char const*, int, char const*, ...) at /home/lanrat/code/esphome/build/bme/src/esphome/core/log.cpp:14
[... cut ...]
[11:04:03]load:0x3fff001c,len:1044
[11:04:03]load:0x40078000,len:10124
[11:04:03]load:0x40080400,len:5828
[11:04:03]entry 0x400806a8
[11:04:03][I][logger:170]: Log initialized
[11:04:03][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[11:04:03][I][app:029]: Running through setup()...
[11:04:03][D][text_sensor:015]: 'bme_version': Sending state '1.19.4 Jul  9 2021, 10:01:31'
[11:04:03][C][bme680_bsec.sensor:016]: Setting up BME680 via BSEC...
[11:04:03][C][wifi:037]: Setting up WiFi...
```

hmax42, 2021-10-05:
I found out yesterday about this extended variant of the bme680 module.
my sensor node was previously built with esphome 1.13.0 ( from pip3) and consists of a ccs811, a bme680 and a dht22.
i was using esphome 1.18, when i built my variant with bme680_bsec.
i noticed today that shortly after i updated it yesterday (and went to bed) the sensor readings for bme and dht stagnated/stopped sending new values.
resetting the node did not change the behavior.
so i flashed the old version, but built with esphome 1.18

i cannot prove that my problem is the same as noted by the OP, but i searched for bme680 and only found this issue.

trvrnrth, 2021-10-16:
If this is still a problem with the latest version I'd probably try running without the web server and see if that resolves the issue.

lanrat, 2021-10-17:
It does seem to be a lot more stable with the web server disabled.

closed 2025-01-24 by lanrat