esphome/issues #2227: Custom preset added in custom climate component but not registered in HA

maretodoric, 2021-07-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, Supervisor

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

Custom Climate

**Description of problem:**
When using custom climate component with custom c++ code, custom_preset is not registered in home assistant and unable to be used. I got the standard things like hvac mode, fan speed and swing but i tried to add "boost" as a custom_preset and it's not appeared in HA. Also, supported_features reported are 41 (https://developers.home-assistant.io/docs/core/entity/climate/#supported-features).
I would like assistance at least on how to register this.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: mcu_dnevna_soba_transciever
  platform: ESP8266
  board: nodemcuv2
  includes: 
    - custom_components/gree/gree_ir.h
  libraries:
    - IRremoteESP8266

wifi:
  ssid: "WiFi"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "MCU Dnevna Soba Transciever"
    password: "password"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "password"

mqtt:
  broker: 192.168.0.201
  username: hamqtt
  password: password
  discovery: false
  discovery_retain: false

remote_transmitter:
  pin: 
    number: D0
    inverted: false
  carrier_duty_percent: 50%
  id: "ir_dsoba_transmitter"

climate:
- platform: custom
  lambda: |-
    auto dsoba_ac = new GreeAC();
    App.register_component(dsoba_ac);
    return {dsoba_ac};
  climates:
    - name: "Dnevna Soba AC"

binary_sensor:
- platform: gpio
  pin:
    #D7
    number: GPIO13
    inverted: true
  internal: true
  name: "ir_dsoba_button"
  id: "ir_dsoba_button"

- platform: custom
  lambda: |-
    auto gree_binary_sensor = new GreeSensor();
    App.register_component(gree_binary_sensor);
    return {gree_binary_sensor};
    
  binary_sensors:
    name: "Dnevna Soba Gree light"
```

**Logs (if applicable):**

```
N/A
```

**Additional information and things you've tried:**
Bellow is a code used:
https://pastebin.com/jfEugYwX

In short, I've registered custom_preset as:

```c++
  climate::ClimateTraits traits()
  {
    auto traits = climate::ClimateTraits();
 
    traits.set_supports_current_temperature(true);
    traits.set_supports_two_point_target_temperature(false);
    traits.set_visual_min_temperature(16);
    traits.set_visual_max_temperature(30);
    traits.set_visual_temperature_step(1.f);
 
    traits.set_supported_modes({
      climate::CLIMATE_MODE_OFF,
      climate::CLIMATE_MODE_HEAT_COOL,
      climate::CLIMATE_MODE_COOL,
      climate::CLIMATE_MODE_HEAT,
      climate::CLIMATE_MODE_FAN_ONLY,
      climate::CLIMATE_MODE_DRY,
    });
 
    traits.set_supported_fan_modes({
      climate::CLIMATE_FAN_AUTO,
      climate::CLIMATE_FAN_LOW,
      climate::CLIMATE_FAN_MEDIUM,
      climate::CLIMATE_FAN_HIGH,
    });
 
    traits.set_supported_swing_modes({
      climate::CLIMATE_SWING_OFF,
      climate::CLIMATE_SWING_VERTICAL,
    });
 
    traits.set_supported_presets({
      climate::CLIMATE_PRESET_BOOST,
    });
 
    return traits;
  }
```

OttoWinter, 2021-07-13:
> In short, I've registered custom_preset as:

The custom preset isn't registered anywhere in the block you posted (`set_supported_custom_presets`)

maretodoric, 2021-07-13:
Yeah, I've removed it and tried with regular preset only (set_supported_presets) it still wasn't detected (CLIMATE_PRESET_BOOST).
However, that night i updated Home Assistant Core from 2021.6.6 to 2021.7.1 and it gotten added. So maybe server restart was needed? Is this expected?
Later on, i added set_supported_custom_presets and they were added to HA too.

OttoWinter, 2021-07-13:
It is. Climate preset support was added to the HA side in 2021.7 - for versions before that the preset will just be ignored.

closed 2021-07-13 by OttoWinter