esphome/issues #2229: i2c bus fails after restart

LeoMaguire, 2021-07-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Host Operating System | Home Assistant OS 6.1
-- | --
Update Channel | stable
Supervisor Version | supervisor-2021.06.8
Docker Version | 20.10.6
Disk Total | 1833.1 GB
Disk Used | 5.9 GB
Healthy | true
Supported | true
Board | rpi4-64
Supervisor API | ok
Version API | ok
Installed Add-ons | Node-RED (9.1.4), MariaDB (2.4.0), Visual Studio Code (3.5.0), Mosquitto broker (6.0.1), Terminal & SSH (9.1.3), ESPHome (1.19.4)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Adafruit Huzzah ESP8266 attached to Atlas Scientific Hydroponics WiFi basic kit

**ESPHome version (latest production, beta, dev branch)**

ESPHome Current version: 1.19.4 
**Affected component:**

**Description of problem:**
Restarting ESP8266 causes i2c bvus failure.  Anytime power is removed from ESP8266 we are no longer able to enable the i2c bvus without pushing multiple sketches to change i2c pins to other pins and then back to the proper pins - thus triggering some sort of hard reset on the pins it seems.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: attic-sensors
  platform: ESP8266
  board: huzzah

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "**************************"

wifi:
  ssid: "*******************"
  password: "****************"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Attic-Sensors Fallback Hotspot"
    password: "*************"

captive_portal:

i2c:
  scan: True
#  sda: 13
#  scl: 14

sensor:
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    update_interval: 30s

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

repeats every time power is cycled on esp8266.  Steps to fix are:

Change:
#  sda: 13
#  scl: 14

to

  sda: GPIO13
  scl: GPIO14
  
  Push sketch, wait for boot.
  
  THEN
  
  Change
  sda: GPIO13
  scl: GPIO14

to

  sda:13
  scl: 14

 push sketch, wait for boot.
 
 
 THEN
 
 Change
  sda:13
  scl: 14

to

#  sda: 13
#  scl: 14

push sketch, wait for boot and it should be able to detect i2c devices once again until next reboot / power cycle.

LeoMaguire, 2021-07-11:
This behavior does not change based on connection type.

Exact same hardware does not have this issue when using Arduino IDE with Atlas Scientific provided code.

LeoMaguire, 2021-07-11:
Log when it works.....

[09:11:43][C][i2c:029]:   SDA Pin: GPIO4
[09:11:43][C][i2c:030]:   SCL Pin: GPIO5
[09:11:43][C][i2c:031]:   Frequency: 50000 Hz
[09:11:43][I][i2c:033]: Scanning i2c bus for active devices...
[09:11:43][I][i2c:040]: Found i2c device at address 0x66
[09:11:43][C][logger:189]: Logger:
[09:11:43][C][logger:190]:   Level: DEBUG
[09:11:43][C][logger:191]:   Log Baud Rate: 115200
[09:11:43][C][logger:192]:   Hardware UART: UART0
[09:11:43][D][api.connection:630]: Client 'Home Assistant 2021.7.1 XXXXXXXXXXX' connected successfully!
[09:11:43][C][ezo.sensor:014]: EZO 'RTD Temperature'
[09:11:43][C][ezo.sensor:014]:   State Class: ''
[09:11:43][C][ezo.sensor:014]:   Unit of Measurement: '°C'
[09:11:43][C][ezo.sensor:014]:   Accuracy Decimals: 2
[09:11:43][C][ezo.sensor:015]:   Address: 0x66
[09:11:43][C][ezo.sensor:018]:   Update Interval: 30.0s
[09:11:43][C][captive_portal:148]: Captive Portal:
[09:11:43][C][ota:029]: Over-The-Air Updates:
[09:11:43][C][ota:030]:   Address: XXXXXXXXXXX
[09:11:43][C][ota:032]:   Using Password.
[09:11:43][C][api:095]: API Server:
[09:11:43][C][api:096]:   Address: XXXXXXXXXXX
[09:12:03][D][sensor:117]: 'RTD Temperature': Sending state 22.72800 °C with 2 decimals of accuracy
[09:12:33][D][sensor:117]: 'RTD Temperature': Sending state 22.73900 °C with 2 decimals of accuracy

LeoMaguire, 2021-07-13:
****
Changed
scan: True 

To

scan: False

Pushed sketch, power cycle - lost the sensor board over i2c.

`[06:03:26][W][i2c:086]: Requesting 20 bytes from 0x66 failed!

[06:03:26][E][ezo.sensor:049]: read error`

I had not tested this change previously.

I had already tried altering the scan frequencies with no change in behavior. 

I also realize my sketch above is what I am entering into ESP Home - and now I see once I click "Validate" it will show me what it is really changing it all too - which I'm pasting below as it is injecting some additional options it seems.

`INFO Reading configuration /config/esphome/attic-sensors.yaml...
INFO Configuration is valid!
esphome:
  name: attic-sensors
  platform: ESP8266
  board: huzzah
  arduino_version: platformio/espressif8266@2.6.2
  build_path: attic-sensors
  platformio_options: {}
  esp8266_restore_from_flash: false
  board_flash_mode: dout
  includes: []
  libraries: []
  name_add_mac_suffix: false
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
ota:
  password: XXXXXXXXXXXXXXXX
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
wifi:
  ap:
    ssid: XXXXXXXXXXXXXXXX
    password: XXXXXXXXXXXXXXXX
    ap_timeout: 1min
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  output_power: 20.0
  networks:
  - ssid: XXXXXXXXXXXXXXXX
    password: XXXXXXXXXXXXXXXX
    priority: 0.0
  use_address: attic-sensors.local
captive_portal: {}
i2c:
- sda: 4
  scl: 5
  frequency: 50000.0
  scan: true
sensor:
- platform: ezo
  id: rtd_ezo
  name: RTD Temperature
  address: 0x66
  accuracy_decimals: 2
  unit_of_measurement: °C
  update_interval: 30s
  force_update: false`

agners, 2021-07-19:
It seems that this EZO module requires some twiggling to put it in I2C mode?

https://atlas-scientific.com/files/EZO_RTD_Datasheet.pdf (see page 42).

I am guessing that is what the multiple programming essentially achieves?

Maybe the original author of the component @ssieb has some thoughts?

LeoMaguire, 2021-07-19:
Device does require that one time - currently the device is in I2C mode and functions perfectly. 

Upon a reboot -  the ESP8266 loses what pins are what and you have to push multiple sketches to get them back to their previous state.

No changes are being made in any way to the sensors or sensor stamp.

agners, 2021-07-19:
Maybe power related? Are you powering it from ESP8266?

Another thing to check is the I2C bus. I guess the EZO RTD doesn't have pull-ups. Do you have external ones?

mmakaay, 2021-11-17:
Various changes has been done for I2C, including the implementation of an I2C bus restore operation at (re)boot.
When there are still issues with this against the latest ESPHome version, please create a new issue, and include logging output.

closed 2021-11-17 by mmakaay