esphome/issues #2230: ESP32-CAM not working with Home Assistant 2021.7.1

MeisterBob, 2021-07-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Homeassistant and ESPHome are installed via pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-CAM

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.18.0, 1.19.0, 1.19.4

**Affected component:**

Camera

**Description of problem:**
After updating hass to 2021.7.1 (skipped 2021.7.0 at first) and esphome to 1.19.4 lovelace stopped showing camera pictures. Shots taken with the camera.snapshot service are corrupted. I downgraded the ESP32-CAM to ESPHome 1.18.0 and 1.19.0, but lovelace still refused to show the camera picture. Next I tried hass 2021.7.0 and 2021.6.6 with different ESPHome versions. All ESPHome versions worked with 2021.6.6 and stopped working in 2021.7.0.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: k..._cam
  upper_devicename: K... Cam

esphome:
  name: $devicename
  platform: ESP32
  board: esp32cam
  build_path: build/$devicename

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.....5
    gateway: 192.168....1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${upper_devicename} Fallback Hotspot
    password: !secret wifi_password

web_server:

# Enable logging
logger:
  level: DEBUG
  logs:
    api: DEBUG
    api.connection: DEBUG
    sensor: WARN
    esp32_camera: WARN

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

sensor:
  - platform: wifi_signal
    name: ${devicename}_wifi
    update_interval: 60s
  - platform: uptime
    name: ${devicename}_uptime
    update_interval: 60s

text_sensor:
  - platform: version
    name: ${devicename}_version

esp32_camera:
  name: Hoftor
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  vertical_flip: false
  horizontal_mirror: false
  resolution: 640x480
  idle_framerate: 1 fps
  max_framerate: 5 fps
```

**Logs (if applicable):**

```
INFO Successfully connected to 192.168.....5
[17:57:21][I][app:105]: ESPHome version 1.19.4 compiled on Jul 11 2021, 17:54:27
[17:57:21][C][wifi:484]: WiFi:
[17:57:21][C][wifi:344]:   SSID: 'j...e'
[17:57:21][C][wifi:345]:   IP Address: 192.168.....5
[17:57:21][C][wifi:347]:   BSSID: 2C:....:08
[17:57:21][C][wifi:348]:   Hostname: 'k...._cam'
[17:57:21][C][wifi:352]:   Signal strength: -56 dB ▂▄▆█
[17:57:21][C][wifi:356]:   Channel: 10
[17:57:21][C][wifi:357]:   Subnet: 255.255.255.0
[17:57:21][C][wifi:358]:   Gateway: 192.168.16.1
[17:57:21][C][wifi:359]:   DNS1: 0.0.0.0
[17:57:21][C][wifi:360]:   DNS2: 0.0.0.0
[17:57:21][C][uptime.sensor:030]: Uptime Sensor 'k...._cam_uptime'
[17:57:21][C][uptime.sensor:030]:   State Class: ''
[17:57:21][C][uptime.sensor:030]:   Unit of Measurement: 's'
[17:57:21][C][uptime.sensor:030]:   Accuracy Decimals: 0
[17:57:21][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[17:57:21][C][logger:189]: Logger:
[17:57:21][C][logger:190]:   Level: DEBUG
[17:57:21][C][logger:191]:   Log Baud Rate: 115200
[17:57:21][C][logger:192]:   Hardware UART: UART0
[17:57:21][C][logger:194]:   Level for 'api': DEBUG
[17:57:21][C][logger:194]:   Level for 'api.connection': DEBUG
[17:57:21][C][logger:194]:   Level for 'sensor': WARN
[17:57:21][C][logger:194]:   Level for 'esp32_camera': DEBUG
[17:57:21][C][version.text_sensor:021]: Version Text Sensor 'k...._cam_version'
[17:57:21][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[17:57:21][C][esp32_camera:043]: ESP32 Camera:
[17:57:21][C][esp32_camera:044]:   Name: Hoftor
[17:57:21][C][esp32_camera:045]:   Board Has PSRAM: YES
[17:57:21][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[17:57:21][C][esp32_camera:048]:   VSYNC Pin: 25
[17:57:21][C][esp32_camera:049]:   HREF Pin: 23
[17:57:21][C][esp32_camera:050]:   Pixel Clock Pin: 22
[17:57:21][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[17:57:21][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[17:57:21][C][esp32_camera:053]:   Reset Pin: -1
[17:57:21][C][esp32_camera:071]:   Resolution: 640x480 (VGA)
[17:57:21][C][esp32_camera:096]:   JPEG Quality: 10
[17:57:21][C][esp32_camera:098]:   Contrast: 0
[17:57:21][C][esp32_camera:099]:   Brightness: 0
[17:57:21][C][esp32_camera:100]:   Saturation: 0
[17:57:21][C][esp32_camera:101]:   Vertical Flip: OFF
[17:57:21][C][esp32_camera:102]:   Horizontal Mirror: OFF
[17:57:21][C][esp32_camera:119]:   Test Pattern: NO
[17:57:21][C][web_server:136]: Web Server:
[17:57:21][C][web_server:137]:   Address: 192.168.....5:80
[17:57:21][C][ota:029]: Over-The-Air Updates:
[17:57:21][C][ota:030]:   Address: 192.168.....5:3232
[17:57:21][C][ota:032]:   Using Password.
[17:57:21][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
[17:57:21][C][api:095]: API Server:
[17:57:21][C][api:096]:   Address: 192.168.....5:6053
[17:57:21][C][wifi_signal.sensor:009]: WiFi Signal 'k...._cam_wifi'
[17:57:21][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[17:57:21][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[17:57:21][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[17:57:21][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[17:57:21][D][esp32_camera:156]: Got Image: len=14907
[17:57:22][D][esp32_camera:156]: Got Image: len=14933
[17:57:23][D][esp32_camera:156]: Got Image: len=14914
[17:57:24][D][esp32_camera:156]: Got Image: len=14932
[17:57:25][D][esp32_camera:156]: Got Image: len=14897
[17:57:26][D][esp32_camera:156]: Got Image: len=14964
[17:57:27][D][esp32_camera:156]: Got Image: len=14948
[17:57:28][D][esp32_camera:156]: Got Image: len=15360
[17:57:29][D][esp32_camera:156]: Got Image: len=14973
[17:57:30][D][esp32_camera:156]: Got Image: len=14948
[17:57:31][D][esp32_camera:156]: Got Image: len=14926
[17:57:32][D][esp32_camera:156]: Got Image: len=15360
[17:57:33][D][esp32_camera:156]: Got Image: len=14972
[17:57:34][D][esp32_camera:156]: Got Image: len=14934
[17:57:35][D][esp32_camera:156]: Got Image: len=14935
[17:57:36][D][esp32_camera:156]: Got Image: len=14919
[17:57:37][D][esp32_camera:156]: Got Image: len=14959
[17:57:38][D][esp32_camera:156]: Got Image: len=14982
[17:57:39][D][esp32_camera:156]: Got Image: len=14904
[17:57:40][D][esp32_camera:156]: Got Image: len=14920
[17:57:41][D][esp32_camera:156]: Got Image: len=15360
[17:57:42][D][esp32_camera:156]: Got Image: len=14872
[17:57:43][D][esp32_camera:156]: Got Image: len=14974
[17:57:44][D][esp32_camera:156]: Got Image: len=14956
[17:57:45][D][esp32_camera:156]: Got Image: len=14858
```

pat1300, 2021-07-11:
I have exactly same issue.
Problem is not related to ESPhome:( I tried with different versions)  but to core-2021.7.1 
If I revert  back to snapshot of core-2021.6.6 -> working fine
updating again to core-2021.7.1  -> no image , but a text "Preview of the entity-name camera"
[image]

pat1300, 2021-07-11:
see #https://github.com/home-assistant/core/pull/52783 : 
will be resolved in 2021.7.2

OttoWinter, 2021-07-13:
Closing per https://github.com/home-assistant/core/pull/52783

closed 2021-07-13 by OttoWinter