esphome/issues #2232: One ESPHome device disappear after other upgraded to 1.19 firmware

wigster, 2021-07-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.io (now 2021.7.1, but happened when I was still running 2021.6) running in VM on Windows Server 2012R2

**ESP (ESP32/ESP8266, Board/Sonoff):**

2x identical Esp32-wroom-32d which were running ESPHome 1.18 with nearly identical configurations
**ESPHome version (latest production, beta, dev branch)**

Occured when I upgraded 1.18 -> 1.19. Now 1.19.4
**Affected component:**

all.

**Description of problem:**
I have 2 ESPHome devices, which were running v1.18. I upgraded the HASS add-on to 1.19 and both the devices were still visible. I then upgraded the firmware on one of the ESP32's to 1.19. This went ok. However, the *other* ESP32, still running the 1.18 firmware disappeared from the ESPHome dashboard and the config is not in the .esphome folder anymore. Nonetheless, the 1.18 device is still connected to HASS and working correctly, the web server is woking ok. I just cannot see it in the ESPHome dashboard.

NB: The remote transmitter/rc_switch part of the code never properly, generating noise and nothing much else.

**Problem-relevant YAML-configuration entries:**
This is the config for the v1.19 esp32 that continues to be visible. The invisible one was the same, but I cannot find it anymore in the esphome folder.
```yaml
PASTE YAML FILE HERE
substitutions:
  device_name: esp32-salon
  device_name_up: "ESP32 Salon"
  domain_name: .lan
  ap_mac:   "DC:EF:09:F2:9B:C2" # router
  #ap_mac:   "B0:39:56:CC:F4:A7" # extender

esphome:
  name: $device_name
  platform: ESP32
  board: nodemcu-32s

wifi:
  fast_connect: true
  domain: $domain_name
  networks: 
    - ssid: !secret wifi24_ssid
      password: !secret wifi24_pwd
      bssid: $ap_mac

  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$device_name Fallback"
    password: !secret esphome_fallback_pwd

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_ota_pwd

ota:
  password: !secret esphome_ota_pwd

# Add webserver on port 80
web_server:
  port: 80
  auth:
    username: !secret esp_web_usr
    password: !secret esp_web_pwd

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

   
time:
  # sync time from HASS 
  - platform: homeassistant
    id: homeassistant_time
    timezone: "Europe/Prague"
    on_time_sync:
      then:
      - logger.log: "Synchronized system clock"
      

binary_sensor:
  # Show connection status of esp32
  - platform: status
    name: "$device_name_up Status"

sensor:

  # wifi signal strength
  - platform: wifi_signal
    name: "${device_name_up} Signal"
    update_interval: 420s      
  
  
  # count uptime and then convert to a human readable format  
  - platform: uptime
    name: "${device_name_up} Uptime"
    id: uptime_sensor
    internal: true
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

  # Xiaomi sensor nr 4: Living            
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:D8:C4:57"
    bindkey: "24d760afa2bd90354deb628ce8a39da0"
    temperature:
      name: "Xiaomi Salon Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi Salon Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi Salon Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery

  # Xiaomi sensor nr 6: Zimna Zahrada (ZZ)            
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:ED:0B:51"
    bindkey: "0b383ca8cee2938413aec09429e3363a"
    temperature:
      name: "Xiaomi ZZ Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi ZZ Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi ZZ Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery
              
text_sensor:
  - platform: template
    name: "${device_name_up} Uptime"
    id: uptime_human
    icon: mdi:clock-start

remote_transmitter:
  pin: GPIO12
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%

switch:
  - platform: template
    name: RF Power Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '100010000000000010111110'
          protocol: 2
```

**Logs (if applicable):**

There is nothing in the logs of either esp32 device. 
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

wigster, 2021-07-12:
I have just found the config file for the missing esp32 (esp32-master) in the .esphome/trash directory
```yaml
substitutions:
  device_name: esp32-master
  device_name_up: "ESP32 Master"
  domain_name: .lan
  #ap_mac:   "DC:EF:09:F2:9B:C2" # router
  ap_mac:   "B0:39:56:CC:F4:A7" # extender

esphome:
  name: $device_name
  platform: ESP32
  board: nodemcu-32s

wifi:
  fast_connect: true
  power_save_mode: none
  domain: $domain_name
  networks: 
    - ssid: !secret wifi24_ssid
      password: !secret wifi24_pwd
      bssid: $ap_mac

  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$device_name Fallback"
    password: !secret esphome_fallback_pwd

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_ota_pwd

ota:
  password: !secret esphome_ota_pwd

# Add webserver on port 80
web_server:
  port: 80
  auth:
    username: !secret esp_web_usr
    password: !secret esp_web_pwd

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

   
time:
  # sync time from HASS 
  - platform: homeassistant
    id: homeassistant_time
    timezone: "Europe/Prague"
    on_time_sync:
      then:
      - logger.log: "Synchronized system clock"
      

binary_sensor:
  # Show connection status of esp32
  - platform: status
    name: "$device_name_up Status"

sensor:

  # wifi signal strength
  - platform: wifi_signal
    name: "${device_name_up} Signal"
    update_interval: 420s
  
  
  # count uptime and then convert to a human readable format  
  - platform: uptime
    name: "${device_name_up} Uptime"
    id: uptime_sensor
    internal: true
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();
  
  # Xiaomi sensor nr 1: Dzieci   
  # need to rename this "Xiaomi Kids Temperature"
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:A0:52:AC"
    bindkey: "0dd0fd5655a5813acf09fd62b9b3c42e"
    temperature:
      name: "Xiaomi Kids Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi Kids Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi Kids Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery
      
  # Xiaomi sensor nr 2: Master            
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:C6:BE:EB"
    bindkey: "3ac8ec48a84b5de934586111f581a3cd"
    temperature:
      name: "Xiaomi Master Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi Master Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi Master Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery
      
   # Xiaomi sensor nr 3: Lazienka            
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:EE:F1:DF"
    bindkey: "6fae43e9586715b04dc0bba47af064d5"
    temperature:
      name: "Xiaomi Bath Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi Bath Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi Bath Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery     

   # Xiaomi sensor nr 6: Hall            
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:85:77:36"
    bindkey: "2354af86c47e89cf3f304d137a00cc99"
    temperature:
      name: "Xiaomi Hall Temperature"
      unit_of_measurement: "°C"
      accuracy_decimals: 1
      icon: mdi:thermometer
      
    humidity:
      name: "Xiaomi Hall Humidity"
      accuracy_decimals: 0
      unit_of_measurement: "% RH"
      icon: mdi:water-percent
      
    battery_level:
      name: "Xiaomi Hall Battery Level"
      accuracy_decimals: 0
      unit_of_measurement: "%"
      icon: mdi:battery   
text_sensor:
  - platform: template
    name: "${device_name_up} Uptime"
    id: uptime_human
    icon: mdi:clock-start

# RF 433 MHz receive/transmit for InterTechno switches
remote_receiver:
  pin: 
    number: GPIO27
    inverted: true
  dump:
    - rc_switch

  # Settings to optimize recognition of RF devices
  tolerance: 60%
  filter: 250us
  idle: 4ms
```

OttoWinter, 2021-07-13:
> in the .esphome/trash directory

That means the `Delete` button was pressed. Could it be that you accidentally pressed that?

wigster, 2021-07-13:
You mean the delete below the clean files options? Ugh. Could I have been that stupid? I was upgrading the firmwamre from the iPad, so it's possible in principle.

In any case, I ahve managed to restore the device from the yaml file in trash. So I will close this and recede in shame.

closed 2021-07-13 by wigster