esphome/issues #2235: 1.19.2 to 1.19.4 bang_bang reports gibberish state in HA 2021.7.2 thermostat UI

ptr727, 2021-07-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome: Docker, 1.19.4
HA: Docker, 2021.7.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff TH10

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

bang_bang

**Description of problem:**
Upgrading from 1.19.2 to 1.19.4 broke the UI text for the HA thermostat component.
1.19.2 work, 1.19.4 broke

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    name: ${device_name}_bangbang
    id: bangbang
    sensor: recirc_temp
    default_target_temperature_low: 30
    default_target_temperature_high: 40
    heat_action:
      - logger.log: "BangBang : Heat Action, Turning Relay On."
      - switch.turn_on: relay
    idle_action:
      - logger.log: "BangBang : Idle Action, Turning Relay Off."
      - switch.turn_off: relay
    visual:
      min_temperature: 20
      max_temperature: 50
      temperature_step: 0.5
```

**Logs (if applicable):**

```
[19:38:14][D][debug:023]: ESPHome version 1.19.4
[19:38:14][D][debug:025]: Free Heap Size: 17872 bytes
[19:38:14][D][debug:053]: Flash Chip: Size=1024kB Speed=40MHz Mode=DOUT
[19:38:14][D][debug:190]: Chip ID: 0x00006BAB
[19:38:14][D][debug:191]: SDK Version: 2.2.2-dev(38a443e)
[19:38:14][D][debug:192]: Core Version: 2_7_4
[19:38:14][D][debug:193]: Boot Version=31 Mode=1
[19:38:14][D][debug:194]: CPU Frequency: 80
[19:38:14][D][debug:195]: Flash Chip ID=0x0014605E
[19:38:14][D][debug:196]: Reset Reason: Software/System restart
[19:38:14][D][debug:197]: Reset Info: Software/System restart
[19:38:20][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_uptime': Sending state 8.79045 minutes with 0 decimals of accuracy
[19:38:21][D][dallas.sensor:153]: 'sonoff_th10_hot_recirc_pump_recirc_temp': Got Temperature=36.6°C
[19:38:21][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_recirc_temp': Sending state 36.83333 °C with 1 decimals of accuracy
[19:38:21][D][climate:369]: 'sonoff_th10_hot_recirc_pump_bangbang' - Sending state:
[19:38:21][D][climate:372]:   Mode: HEAT_COOL
[19:38:21][D][climate:374]:   Action: IDLE
[19:38:21][D][climate:392]:   Current Temperature: 36.83°C
[19:38:21][D][climate:396]:   Target Temperature: Low: -158456325028528675187087900672.00°C High: 0.00°C
[19:38:21][D][dallas.sensor:153]: 'sonoff_th10_hot_recirc_pump_hot_temp': Got Temperature=48.0°C
[19:38:21][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_hot_temp': Sending state 48.10417 °C with 1 decimals of accuracy
[19:38:21][D][dallas.sensor:153]: 'sonoff_th10_hot_recirc_pump_cold_temp': Got Temperature=25.8°C
[19:38:21][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_cold_temp': Sending state 25.72917 °C with 1 decimals of accuracy
[19:38:21][D][dallas.sensor:153]: 'sonoff_th10_hot_recirc_pump_ambient_temp': Got Temperature=31.1°C
[19:38:21][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_ambient_temp': Sending state 31.06250 °C with 1 decimals of accuracy
[19:38:28][D][sensor:117]: 'sonoff_th10_hot_recirc_pump_wifi_signal': Sending state -49.33333 dBm with 0 decimals of accuracy
```
[image]
[image]

**Additional information and things you've tried:**
After upgrading from 1.19.2 to 1.19.4 you have to go into the HA UI and drag the low and high temps to get the thermostat to behave correctly.

My money is on the HA API changes that broke backwards compatibility between ESPHome and HA for stored state information.

OttoWinter, 2021-07-13:
The temperature is already reported wrong on the device itself:
```
[19:38:21][D][climate:396]:   Target Temperature: Low: -158456325028528675187087900672.00°C High: 0.00°C
```

That would mean the HA side is not involved.

Does this issue still exist when you reboot the device on 1.19.4 - I think maybe the struct layout for state restoration changed, which would explain these numbers.

ptr727, 2021-07-13:
I know device was on 1.19.2 and HA reported correctly.
I upgraded the device to 1.19.4 and HA reported funky text and ESPLog reported funky ranges.

Since the YML has correct values, why would the device have different values if they come from YML?
I assumed HA restored the values as HA remembers them, and I suspected there is a binary incompatibility between HA and ESPHome when protobuf / API structs changed without being correctly versioned.

If HA did not set the state, and I do have `esp8266_restore_from_flash: false`, how did the device get the wrong configuration?

OttoWinter, 2021-07-13:
> I assumed HA restored the values as HA remembers them, and I suspected there is a binary incompatibility between HA and ESPHome when protobuf / API structs changed without being correctly versioned.

It does not. HA is not involved in any state restoration.

> If HA did not set the state, and I do have `esp8266_restore_from_flash: false`, how did the device get the wrong configuration?

There are two methods that restore happens. From RTC memory (preserved across reboots but not power loss) and flash (always preserved). With that option you only disable the latter.

Does it show funky values if you reboot the device?

ptr727, 2021-07-13:
> It does not. HA is not involved in any state restoration.
For HA and state, good, to know, the behavior always appeared confusing / non-deterministic to me, e.g. set state in HA, update device, where YML has different defaults, HA state preserves, maybe that is the RTC memory you are referring to?

> Does it show funky values if you reboot the device?
As soon as I changed the config from the HA UI the problem went away, so I don't know if a second reboot would have resolved the problem. I could test it by going back to 1.19.2 and then 1.19.4 again, but I don't think it is worth the effort.

Regardless, not expected behavior when doing an update, right?
I'd expect that either the in-memory structures need to be backwards compatible, or any persisted memory (RTC, Flash) needs to be wiped on firmware updates?

If HA does not restore state, then

OttoWinter, 2021-07-26:
> maybe that is the RTC memory you are referring to?

Correct.

Yeah it's not expected behavior. Yes the preference should be invalidated when the schema changes.

closed 2021-07-26 by OttoWinter