esphome/issues #2236: Instructions on "getting started command line" page are wrong - swapped configuration and action parametters

BugerDread, 2021-07-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3 on ubuntu 18.04 x64

**ESP (ESP32/ESP8266, Board/Sonoff):**

d1_mini

**ESPHome version (latest production, beta, dev branch)**

Version: 1.17.2

**Affected component:**

doc page itself - https://esphome.io/guides/getting_started_command_line.html

**Description of problem:**

Instructions on https://esphome.io/guides/getting_started_command_line.html#creating-a-project says:
```
esphome wizard livingroom.yaml
```

but this is wrong and should be:
```
esphome livingroom.yaml wizard 
```

the same problem here: https://esphome.io/guides/getting_started_command_line.html#first-uploading
```
esphome run livingroom.yaml
```
should be:
```
esphome livingroom.yaml run
```

and I also expect the docker variants of the commands should also have the action AFTER yaml name

**Problem-relevant YAML-configuration entries:**
```yaml
not relevant to yaml file

```
**Logs (if applicable):**

```
$ esphome run livingroom.yaml
usage: esphome [-h] [-v] [-q] [-s key value]
               [configuration [configuration ...]]
               {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard,vscode,update-all}
               ...
esphome: error: argument command: invalid choice: 'livingroom.yaml' (choose from 'config', 'compile', 'upload', 'logs', 'run', 'clean-mqtt', 'wizard', 'mqtt-fingerprint', 'version', 'clean', 'dashboard', 'vscode', 'update-all')
jdrozd@sdentbk:~/Projects/esphome-prednaska/esphome$ 

```

**Additional information and things you've tried:**

I looked on the output of the command and it was obvious for me, but can be very confusing for less experienced ppl when basic examples in "getting started" are wrong / not working

jesserockz, 2021-07-13:
The issue is that you are using an old version of esphome and the CLI argument order has been changed: esphome/esphome#1805

BugerDread, 2021-07-14:
Ok, but why I got old version of esphome? I installed it about hour before creating this issue as described at the top of the same page (https://esphome.io/guides/getting_started_command_line.html#installation):
`pip3 install esphome`
Thats why I created this issue - I was following steps described on "getting started command line" page and it did not worked.

jesserockz, 2021-07-14:
I am not sure why you got an old version if you just installed it, the latest is [1.19.4](https://pypi.org/project/esphome/)

```txt
➜  /tmp python -m virtualenv venv
created virtual environment CPython3.9.6.final.0-64 in 624ms
  creator CPython3Posix(dest=/tmp/venv, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/jesse/.local/share/virtualenv)
    added seed packages: pip==21.0.1, setuptools==57.0.0, wheel==0.36.2
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
➜  /tmp source venv/bin/activate
(venv) ➜  /tmp pip3 install esphome
Collecting esphome
  Downloading esphome-1.19.4-py2.py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 3.7 MB/s
```

BugerDread, 2021-07-14:
Yes, I installed just before creating this issue, on a fully updated ubuntu mate 18.04 x64. 
I also installed pip3 at the same time because it was not present on my system yet.
Tried to update esphome via pip now but still on 1.17.2, here is what I got:
```
$ sudo apt update
Hit:1 http://ppa.launchpad.net/yannubuntu/boot-repair/ubuntu bionic InRelease
Hit:2 https://download.virtualbox.org/virtualbox/debian bionic InRelease                                                                                                   
Hit:3 https://packages.microsoft.com/repos/ms-teams stable InRelease                                                                                                       
Hit:4 https://brave-browser-apt-release.s3.brave.com stable InRelease                                                                                                 
Hit:5 http://cz.archive.ubuntu.com/ubuntu bionic InRelease                                                                                      
Hit:6 http://security.ubuntu.com/ubuntu bionic-security InRelease                                                         
Hit:7 http://cz.archive.ubuntu.com/ubuntu bionic-updates InRelease                                                      
Hit:8 http://cz.archive.ubuntu.com/ubuntu bionic-backports InRelease                                
Hit:9 https://packagecloud.io/slacktechnologies/slack/debian jessie InRelease                     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```
```
$ pip3 install --upgrade esphome | tee pip3-esphome.log
Collecting esphome
  Using cached https://files.pythonhosted.org/packages/b7/6b/e6557976bc46900c3795c018441b459f5b8bf0867cd6e340cf477d0cbb3d/esphome-1.17.2-py2.py3-none-any.whl
Collecting paho-mqtt==1.5.1 (from esphome)
Collecting tornado==6.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/01/d1/8750ad20cbcefb499bb8b405e243f83c2c89f78d139e6f8c8d800640f554/tornado-6.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting esptool==2.8 (from esphome)
Collecting colorama==0.4.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/44/98/5b86278fbbf250d239ae0ecb724f8572af1c91f4a11edf4d36a206189440/colorama-0.4.4-py2.py3-none-any.whl
Collecting click==7.1.2 (from esphome)
  Using cached https://files.pythonhosted.org/packages/d2/3d/fa76db83bf75c4f8d338c2fd15c8d33fdd7ad23a9b5e57eb6c5de26b430e/click-7.1.2-py2.py3-none-any.whl
Collecting tzlocal==2.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/5d/94/d47b0fd5988e6b7059de05720a646a2930920fff247a826f61674d436ba4/tzlocal-2.1-py2.py3-none-any.whl
Collecting protobuf==3.15.6 (from esphome)
  Using cached https://files.pythonhosted.org/packages/b1/32/ddc383342be0316f4792d485522b976514e2606dafd84de1fe90931ffe5d/protobuf-3.15.6-cp36-cp36m-manylinux1_x86_64.whl
Collecting ifaddr==0.1.7 (from esphome)
  Using cached https://files.pythonhosted.org/packages/92/0f/a577a724c03982b800232713874e805c8fcc14f4a2c3060902ed20b50da8/ifaddr-0.1.7-py2.py3-none-any.whl
Collecting voluptuous==0.12.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/9c/b9/84f5bd190a7dc234cea2ca1da02c8900ace982e0ad4ae494660585844f04/voluptuous-0.12.1-py3-none-any.whl
Collecting pytz==2021.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/70/94/784178ca5dd892a98f113cdd923372024dc04b8d40abe77ca76b5fb90ca6/pytz-2021.1-py2.py3-none-any.whl
Collecting pyserial==3.5 (from esphome)
  Using cached https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl
Collecting PyYAML==5.4.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/7a/5b/bc0b5ab38247bba158504a410112b6c03f153c652734ece1849749e5f518/PyYAML-5.4.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting platformio==5.1.1 (from esphome)
Collecting pyaes (from esptool==2.8->esphome)
Collecting ecdsa (from esptool==2.8->esphome)
  Using cached https://files.pythonhosted.org/packages/4a/b6/b678b080967b2696e9a201c096dc076ad756fb35c87dca4e1d1a13496ff7/ecdsa-0.17.0-py2.py3-none-any.whl
Collecting six>=1.9 (from protobuf==3.15.6->esphome)
  Using cached https://files.pythonhosted.org/packages/d9/5a/e7c31adbe875f2abbb91bd84cf2dc52d792b5a01506781dbcf25c91daf11/six-1.16.0-py2.py3-none-any.whl
Collecting pyelftools<1,>=0.27 (from platformio==5.1.1->esphome)
  Using cached https://files.pythonhosted.org/packages/6f/50/3d7729d64bb23393aa4c166af250a6e6f9def40c90bf0e9af3c5ad25b6f7/pyelftools-0.27-py2.py3-none-any.whl
Collecting tabulate==0.8.* (from platformio==5.1.1->esphome)
  Using cached https://files.pythonhosted.org/packages/ca/80/7c0cad11bd99985cfe7c09427ee0b4f9bd6b048bd13d4ffb32c6db237dfb/tabulate-0.8.9-py3-none-any.whl
Collecting uvicorn==0.13.* (from platformio==5.1.1->esphome)
  Using cached https://files.pythonhosted.org/packages/c8/de/953f0289508b1b92debdf0a6822d9b88ffb0c6ad471d709cf639a2c8a176/uvicorn-0.13.4-py3-none-any.whl
Collecting bottle==0.12.* (from platformio==5.1.1->esphome)
  Using cached https://files.pythonhosted.org/packages/bf/44/aeafdd6ca05a8e1c3f91eeeb272a202d5cb1b3b23730a5ca686a81c48d24/bottle-0.12.19-py3-none-any.whl
Collecting ajsonrpc==1.1.* (from platformio==5.1.1->esphome)
  Using cached https://files.pythonhosted.org/packages/e8/e2/fdbac8c6bd28a7f22018a830043a2f02f8d26a149751f1b2d0b078d41bfe/ajsonrpc-1.1.0-py3-none-any.whl
[... cut ...] aho-mqtt-1.5.1 platformio-5.1.1 protobuf-3.15.6 pyaes-1.6.1 pyelftools-0.27 pyserial-3.5 pytz-2021.1 requests-2.26.0 semantic-version-2.8.5 six-1.16.0 starlette-0.14.2 tabulate-0.8.9 tornado-6.1 typing-extensions-3.10.0.0 tzlocal-2.1 urllib3-1.26.6 uvicorn-0.13.4 voluptuous-0.12.1 wsproto-1.0.0 zeroconf-0.28.8
```
```
$ esphome version
Version: 1.17.2
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```

jesserockz, 2021-07-14:
Maybe try `pip3 install esphome==1.19.4` ?

BugerDread, 2021-07-14:
```
$ pip3 install esphome==1.19.4
Collecting esphome==1.19.4
  Could not find a version that satisfies the requirement esphome==1.19.4 (from versions: 1.10.1, 1.11.0b1, 1.11.0b2, 1.11.0b3, 1.11.0, 1.11.1, 1.11.2, 1.12.0b1, 1.12.0b2, 1.12.0b3, 1.12.0b4, 1.12.0, 1.12.1, 1.12.2, 1.13.0b1, 1.13.0b2, 1.13.0b3, 1.13.0b4, 1.13.0b5, 1.13.0b6, 1.13.0b7, 1.13.0, 1.13.1, 1.13.2, 1.13.3, 1.13.4, 1.13.5, 1.13.6, 1.14.0b1, 1.14.0b2, 1.14.0b3, 1.14.0b4, 1.14.0b5, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.15.0b1, 1.15.0b2, 1.15.0b3, 1.15.0b4, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.16.0b1, 1.16.0b2, 1.16.0b3, 1.16.0b4, 1.16.0b5, 1.16.0b6, 1.16.0b7, 1.16.0b8, 1.16.0, 1.16.1, 1.16.2, 1.17.0b1, 1.17.0, 1.17.1, 1.17.2)
No matching distribution found for esphome==1.19.4
```

BugerDread, 2021-07-14:
from my previous log:
`  Using cached https://files.pythonhosted.org/packages/05/1b/0a0dece0e8aa492a6ec9e4ad2fe366b511558cdc73fd3abc82ba7348e875/certifi-2021.5.30-py2.py3-none-any.whl`

Isnt it possible that files.pythonhosted.org dont have newer version? Sorry if this is silly question, Im not a python guy...

jesserockz, 2021-07-14:
No, pythonhosted is the source for pypi

What python version are you using?

```txt
  Found link https://files.pythonhosted.org/packages/87/b2/82a1da103d994dc64c95cd3b99262874da9ffdab103846971d21063cfe7f/esphome-1.19.0-py2.py3-none-any.whl#sha256=d9f89335f698bcbdfd65cb662bb295ede9561e0937b6c090e3fa89e0e887d321 (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.0
  Found link https://files.pythonhosted.org/packages/aa/ce/e7e369985c3e79360160df2b626de5f15c8e2430bcc234ce4b13e6bb2589/esphome-1.19.0.tar.gz#sha256=377b01f2f3b8fb9c61a9ca4f3e1c811065ba7526c59bee25b9b85462384a093f (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.0
  Found link https://files.pythonhosted.org/packages/04/da/c488ad10dce6c54b1d2bed9c4aa9e1b2b62340d34016c7910dbb3e4c3c36/esphome-1.19.1-py2.py3-none-any.whl#sha256=44c7c9411acc82bda2a85205c684f10373e1fcf910fd1367f6f33d26dafcaa5b (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.1
  Found link https://files.pythonhosted.org/packages/48/31/8da0f58a175990fa8ed78cb9f89c44d852095da7b55b6032481249dee278/esphome-1.19.1.tar.gz#sha256=40711aa5587eb607b840cd1c6bc8a0376739b5af76327c1ba5b910b77eb7f3bd (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.1
  Found link https://files.pythonhosted.org/packages/19/64/98aa4304c867c723337c3f35ee37f741cd466b522a8b80a96b3ae09e8dea/esphome-1.19.2-py2.py3-none-any.whl#sha256=ee6ee4f79f713edd68304fdba84f1ced29357c4c66d51cf5963c06c256ed74c3 (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.2
  Found link https://files.pythonhosted.org/packages/2c/70/24850a27a4e6c3029c1f7a24025363edc0d32aeda6771a1ac5405c2f5815/esphome-1.19.2.tar.gz#sha256=780f367f945e0b43d88a2bfb2ac64fa9695cec7fda21aa0321a0294d1bd0d6e6 (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.2
  Found link https://files.pythonhosted.org/packages/ff/3b/c267d5093a6e71dca29495ff4108df3d31e7375f3a1ad2d50160232f0474/esphome-1.19.3-py2.py3-none-any.whl#sha256=6ee99e0bd23258d1d97f43386727d7ceaf0f0bf997685f8bbb28dd78f9ea64dd (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.3
  Found link https://files.pythonhosted.org/packages/8c/8d/818b1bf09cacc781b9f1d5519b97dec2bd4d8b766fde017dabc76b48c63c/esphome-1.19.3.tar.gz#sha256=5d5d3ab1f37f831b77ba84bbc2e2e8b7fb2b5ff573c62caccec4028688aaad0e (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.3
  Found link https://files.pythonhosted.org/packages/59/f0/94d6277c2582b86ea81e501fe0c374c9cd052b2a8ca43e5c4f7e04839647/esphome-1.19.4-py2.py3-none-any.whl#sha256=153a4651e1558fedd7f1943b57c5da2b348ed250d677b6ba2cbb2db81308b243 (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.4
  Found link https://files.pythonhosted.org/packages/06/00/1a80fcd4f24476e69273d6ecbcaea2be61fd706b53727ff43ca108e9fef5/esphome-1.19.4.tar.gz#sha256=6c879c780138e6bf7aa37160c5f2bbc5740e7684bdb563e2d93b00d529a97c32 (from https://pypi.org/simple/esphome/) (requires-python:>=3.7,<4.0), version: 1.19.4

```

jesserockz, 2021-07-14:
I am guessing python 3.6?

BugerDread, 2021-07-14:
```
$ python --version
Python 2.7.17
$ python3 --version
Python 3.6.9
```
```
$ apt list --installed | grep -i python

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

dh-python/bionic,bionic,now 3.20180325ubuntu2 all [installed,automatic]
glade2script-python3/bionic,bionic,now 3.2.4~ppa23 all [installed,auto-removable]
libpeas-1.0-python2loader/bionic,now 1.22.0-2 amd64 [installed]
libpython-all-dev/bionic,now 2.7.15~rc1-1 amd64 [installed,automatic]
libpython-dev/bionic,now 2.7.15~rc1-1 amd64 [installed,automatic]
libpython-stdlib/bionic,now 2.7.15~rc1-1 amd64 [installed]
libpython2.7/bionic-updates,bionic-security,now 2.7.17-1~18.04ubuntu1.6 amd64 [installed]
libpython2.7-dev/bionic-updates,bionic-security,now 2.7.17-1~18.04ubuntu1.6 amd64 [installed,automatic]
libpython2.7-minimal/bionic-updates,bionic-security,now 2.7.17-1~18.04ubuntu1.6 amd64 [installed]
libpython2.7-stdlib/bionic-updates,bionic-security,now 2.7.17-1~18.04ubuntu1.6 amd64 [installed]
libpython3-dev/bionic-updates,now 3.6.7-1~18.04 amd64 [installed,automatic]
libpython3-stdlib/bionic-updates,now 3.6.7-1~18.04 amd64 [installed]
libpython3.6/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed]
libpython3.6-dev/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed,automatic]
libpython3.6-minimal/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed]
libpython3.6-stdlib/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed]
python/bionic,now 2.7.15~rc1-1 amd64 [installed]
python-all/bionic,now 2.7.15~rc1-1 amd64 [installed,automatic]
python-all-dev/bionic,now 2.7.15~rc1-1 amd64 [installed,automatic]
python-apt-common/bionic-updates,bionic-updates,now 1.6.5ubuntu0.6 all [installed]
python-asn1crypto/bionic,bionic,now 0.24.0-1 all [installed,automatic]
python-bs4/bionic,bionic,now 4.6.0-1 all [installed,automatic]
python-cairo/bionic,now 1.16.2-1 amd64 [installed,automatic]
python-caja/bionic,now 1.20.0-1 amd64 [installed]
python-caja-common/bionic,bionic,now 1.20.0-1 all [installed]
python-cffi-backend/bionic,now 1.11.5-1 amd64 [installed,automatic]
python-chardet/bionic,bionic,now 3.0.4-1 all [installed,automatic]
python-configobj/bionic,bionic,now 5.0.6-2 all [installed]
python-crypto/bionic,now 2.6.1-8ubuntu2 amd64 [installed]
python-cryptography/bionic-updates,bionic-security,now 2.1.4-1ubuntu1.4 amd64 [installed,automatic]
python-dbus/bionic,now 1.2.6-1 amd64 [installed,automatic]
python-dev/bionic,now 2.7.15~rc1-1 amd64 [installed,automatic]
python-enum34/bionic,bionic,now 1.1.6-2 all [installed,automatic]
python-fasteners/bionic,bionic,now 0.12.0-3 all [installed]
python-gi/bionic-updates,now 3.26.1-2ubuntu1 amd64 [installed]
python-gobject-2/bionic,now 2.28.6-12ubuntu3 amd64 [installed,automatic]
python-gtk2/bionic,now 2.24.0-5.1ubuntu2 amd64 [installed,automatic]
python-html5lib/bionic,bionic,now 0.999999999-1 all [installed,automatic]
python-idna/bionic,bionic,now 2.6-1 all [installed,automatic]
python-ipaddress/bionic,bionic,now 1.0.17-1 all [installed,automatic]
python-keyring/bionic,bionic,now 10.6.0-1 all [installed,automatic]
python-keyrings.alt/bionic,bionic,now 3.0-1 all [installed,automatic]
python-ldb/bionic-updates,bionic-security,now 2:1.2.3-1ubuntu0.2 amd64 [installed]
python-lockfile/bionic,bionic,now 1:0.12.2-2 all [installed]
python-lxml/bionic-updates,bionic-security,now 4.2.1-1ubuntu0.4 amd64 [installed,automatic]
python-mate-menu/bionic,now 1.20.0-2 amd64 [installed]
python-minimal/bionic,now 2.7.15~rc1-1 amd64 [installed]
python-monotonic/bionic,bionic,now 1.1-2 all [installed]
python-numpy/bionic,now 1:1.13.3-2ubuntu1 amd64 [installed,automatic]
python-pip/bionic-updates,bionic-updates,now 9.0.1-2.3~ubuntu1.18.04.5 all [installed]
python-pip-whl/bionic-updates,bionic-updates,now 9.0.1-2.3~ubuntu1.18.04.5 all [installed,automatic]
python-pkg-resources/bionic,bionic,now 39.0.1-2 all [installed]
python-samba/bionic-updates,bionic-security,now 2:4.7.6+dfsg~ubuntu-0ubuntu2.23 amd64 [installed]
python-scour/bionic,bionic,now 0.36-2 all [installed,automatic]
python-secretstorage/bionic,bionic,now 2.3.1-2 all [installed,automatic]
python-serial/bionic,bionic,now 3.4-2 all [installed]
python-setproctitle/bionic,now 1.1.10-1build2 amd64 [installed]
python-setuptools/bionic,bionic,now 39.0.1-2 all [installed]
python-six/bionic,bionic,now 1.11.0-2 all [installed]
[... 85 lines cut ...]
python3-simplejson/bionic,now 3.13.2-1 amd64 [installed]
python3-sip/bionic-updates,now 4.19.7+dfsg-1ubuntu0.1 amd64 [installed,automatic]
python3-six/bionic,bionic,now 1.11.0-2 all [installed]
python3-smbc/bionic,now 1.0.15.6-1build2 amd64 [installed]
python3-software-properties/bionic-updates,bionic-updates,bionic-security,bionic-security,now 0.96.24.32.14 all [installed]
python3-speechd/bionic,bionic,now 0.8.8-1ubuntu1 all [installed]
python3-systemd/bionic,now 234-1build1 amd64 [installed]
python3-twisted/bionic-updates,bionic-updates,bionic-security,bionic-security,now 17.9.0-2ubuntu0.1 all [installed]
python3-twisted-bin/bionic-updates,bionic-security,now 17.9.0-2ubuntu0.1 amd64 [installed]
python3-uno/bionic-updates,bionic-security,now 1:6.0.7-0ubuntu0.18.04.10 amd64 [installed]
python3-update-manager/bionic-updates,bionic-updates,now 1:18.04.11.13 all [installed]
python3-urllib3/bionic-updates,bionic-updates,bionic-security,bionic-security,now 1.22-1ubuntu0.18.04.2 all [installed]
[... cut ...]
python3-xkit/bionic,bionic,now 0.5.0ubuntu2 all [installed]
python3-xlib/bionic,bionic,now 0.20-3 all [installed]
python3-yaml/bionic,now 3.12-1build2 amd64 [installed]
python3-zope.interface/bionic,now 4.3.2-1build2 amd64 [installed]
python3.6/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed]
python3.6-dev/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed,automatic]
python3.6-minimal/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1.4 amd64 [installed]

```

jesserockz, 2021-07-14:
Yeah so esphome 1.18.0 and above requires python 3.7 or above, that's why it wont install anything higher.

BugerDread, 2021-07-14:
ok, tried to use docker version (bcs I dont want to poke with python versions on this machine) and yes - got 1.19.4 and yes - it works as mentioned in getting started page.

Anyway maybe its worth to mention this in the installation instructions (that you need at leas python 3.7 to get latest version)

jesserockz, 2021-07-14:
Good point, will alter this issue for that

closed 2021-07-14 by frenck

BugerDread, 2021-07-14:
yes, thank you to make this clear