esphome/issues #2243: Sonoff Basic not booting

KC-inDomus, 2021-07-16:
### The problem

Sonoff Basic with latest ESPHOME gives this log after 5 boot attempts:

> WARNING Exception type: InstructionFetchError: Processor internal physical address or data error during instruction fetch
[12:40:29]epc1=0x3fff1f9c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff1f9c depc=0x00000000
[12:40:29]
[12:40:29]ctx: cont 
[12:40:29]sp: 3ffffa10 end: 3fffffd0 offset: 01a0
[12:40:29]
[12:40:29]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[12:40:29]3ffffbb0:  3ffffc23 3ffea286 00302073 00000003  
[12:40:29]3ffffbc0:  3ffea289 00000003 3ffffbf0 3ffea286  
[12:40:29]3ffffbd0:  00000000 3ffe8304 3ffffc80 4021ee74  

### What is version of ESPHome has the issue?

1.19.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Sonoff Basic

j-k-olofsson, 2021-07-16:
Same as #2198 and #2205

jesserockz, 2021-07-19:
Duplicate of #2198

closed 2021-07-19 by jesserockz