esphome/issues #2244: Dashboard not usable with Kiwi browser on Android

derandiunddasbo, 2021-07-17:
### The problem

I've recently updated from 1.16.2 to 1.19.4 and noticed, that the new popups for Validate, Install and Logs don't appear when using Kiwi browser on Android. Only the edit popup is still working. I suspect, the new popups trigger some unwanted popup prevention code in Kiwi, as the dashboard is still running fine in Google Chrome.

### What is version of ESPHome has the issue?

1.19.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Additional information

I know I can always use another browser, i.e. Chrome, to mitigate this problem and that's exactly what I'm doing ATM.

But as the Home Assistant / ESPHome ecosystem is especially focused on user privacy, at least I wanted to let you know, that ESPHome isn't working properly anymore with one of the more common privacy focused Android browsers. Maybe you'll have a look at it.

gabipintea, 2021-07-17:
I experience the same behavior in Chrome on desktop device. Having a look at the console shows that t.filter is not a function in a .js file looking for OTA.
`c.2472ed8e.js:formatted:3 Uncaught (in promise) TypeError: t.filter is not a function
    at VM190 c.2472ed8e.js:1`
`const h = ()=>t("./serial-ports").then((t=>t.filter((t=>"OTA" !== t.port))));`

derandiunddasbo, 2021-08-11:
Some days ago, suddenly everything was working again in Kiwi with the same ESPHome version. 

After experiencing similar problems in Chrome on Windows 10 today, but this time only the Log popup refused to open, this behavior started downright yelling _**cache problem !!!11!!1!**_ right into my face. And yes, after clearing the cache, everything was working again in chrome as well.

So, nevermind. I'll close this issue, as I think, there's not much that can be done about random browser cache problems from ESPHome's side.

closed 2021-08-11 by derandiunddasbo