esphome/issues #2245: SHT3X-D wemos shield not working

raidnet-ms, 2021-07-18:
### The problem

Hello,
I'm having issues to make this sensor work. Just want to be sure if wemos shield eith SHT-30 is supposed to work with sht3xd platform.

### What is version of ESPHome has the issue?

V1.19.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

D1mini

### Component causing the issue

Sht3xd sensor

### Example YAML snippet

```yaml
i2c:
  sda: D2
  scl: D1
# frequency: 1000 Hz
  scan: false

sensor:
  - platform: sht3xd
    address: 0x44 #also tryied 0x45
    Temperature: ...
    Humidity ...

I tryeid reducing i2c frequency to 1000 but i always get nack value for the sensor.
Setting scan to true will cause device timeout.
```

### Anything in the logs that might be useful for us?

```txt
Coming soon just please tell me if wemos sht30 is supposed to work.
```

intermittech, 2021-07-18:
Try and enable the I2C scan and see if it at least sees something on the bus? Also maybe try to use the GPIO numbers, not the "D" numbers stuff.

raidnet-ms, 2021-07-19:
Hi, @intermittech if I enable scanning the device get stuck on "scanning". I can see that trough the serial log because in wifi won't connect if scan is enabled.

Already tried with gpio4 and 5.

So you can confirm that sht30 v1.0 shield should work platform sht3xd?

agners, 2021-07-19:
I do have SHT3x sensors working with ESPHome v1.19. It should also work in that particular combination, also reported by others:
https://community.home-assistant.io/t/esphome-wemos-d1-mini-pro-v2-sht30-working-hp303b-pressure-shield-not-working/166711/8

Maybe v1.0 has no pull-ups or something? It seems that v2.1 is current?
https://www.wemos.cc/en/latest/d1_mini_shield/sht30.html

raidnet-ms, 2021-07-21:
I had a doubt related to the relay shield that was connected upon the sensor sharing the D1 pin (of course I had it disabled).
Btw I soldered pins to the sensor now so I could connect directly on the d1mini with no other interference but I still get  [E][sht3xd:037]: Communication with sht3xd failed!

The i2c scan now is not getting stuck anymore but it does not find any device.
I also tryed with 0x44 and 0x45 address with no success.
Maybe the v1.0 of sjt30 shield is not supported?

raidnet-ms, 2021-07-21:
> I do have SHT3x sensors working with ESPHome v1.19. It should also work in that particular combination, also reported by others:
> h

> I do have SHT3x sensors working with ESPHome v1.19. It should also work in that particular combination, also reported by others:
> https://community.home-assistant.io/t/esphome-wemos-d1-mini-pro-v2-sht30-working-hp303b-pressure-shield-not-working/166711/8
> 
Yup I already trued this way.

raidnet-ms, 2021-07-25:
Solved with esphome 1.20 🤷

closed 2021-07-25 by raidnet-ms