esphome/issues #2246: White channel on RGBW strip is only set correctly during transition period of light state change

WeekendWarrior1, 2021-07-20:
### The problem

Regardless of what level I have set the white channel, the white LED only lights for the length of the transition, then it turns off. 
I set the white channel to max, and it lights at the maximum intensity for the default transition length (1.0s), before turning off.

Brief discussion:
- https://github.com/esphome/issues/issues/2115#issuecomment-883098741
- https://github.com/esphome/issues/issues/2115#issuecomment-883327674

This is the offending line (reverted it and white channel worked as expected):
esphome/esphome@f979782#diff-3b81c3b82af5aae01e880f1f5b17d55754abcc67dd56e012dd47955284cb91e9R32

Though I then discovered a second interesting bug (which I don't believe was due to the reverted line, though not certain):
When I am using the white channel at 50%, then turn off the light like so:
[image]
The white channel stays lit. - Thinking about it more, I do believe this is due to the reverted line.

### What is version of ESPHome has the issue?

v.1.20-dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Example YAML snippet

```yaml
light:
  - platform: neopixelbus
    type: RGBW
    pin: GPIO17
    variant: SK6812
    num_leds: 3
    name: "Test AC LED Strip"
    gamma_correct: 0
```

oxan, 2021-07-20:
> The reason I thought this may have been an intermediate/interim change is because I thought it might end up being split into two tabs like discussed above: [#2115 (comment)](https://github.com/esphome/issues/issues/2115#issuecomment-871145791)

That's only for interlocked lights, which can't enable color and white at the same time. The interface for non-interlocked RGBW lights will stay the same.

> Though when I say "works as expected", it works how the RGBW control used to, which didn't really make that much sense:
> 
>   * The `Brightness` slider controls the overall intensity of the RGB channels
> 
>   * The `Color brightness` slider does nothing, and if you move it, it will reset to max next time you change the light state in any way
> 
>   * The `White brightness` controls the white channel's intensity.

So this changed in v1.20, the brightness slider now is a master brightness that applies to both the RGB channel and the white channel.

> Though I then discovered a second interesting bug (which I don't believe was due to the reverted line, though not certain):
> When I am using the white channel at 50%, then turn off the light like so: [...]

Yes, this is due to your revertion.

As for your problem, I still can't reproduce this. I don't have a physical NeoPixelBus RGBW strip to test with, so I inserted this in neopixelbus_light.h:89 to log the input ESPHome sends to NeoPixelBus:
```cpp
    ESP_LOGD("neopixelbus", "writing addressable buffer");
    for (int i = 0; i < this->size(); i++) {
      auto color = this->get(i);
      ESP_LOGD("neopixelbus", "  [%2d] red=%3u green=%3u blue=%3u white=%3u", i, color.get_red_raw(), color.get_green_raw(),
                color.get_blue_raw(), color.get_white_raw());
    }
```
Whatever I do, the logged white value is never zero. I can reproduce that the intensity goes up during the transition and then drops down, but not to zero.

I did notice in your screenshot in the other issue that you set the brightness to a very low level. This will now also dim the white channel, and as I said some strips don't support very low light levels. What happens if you set the brightness slider to a higher value?

WeekendWarrior1, 2021-07-20:
Firstly, thanks for your patience.
> So this changed in v1.20, the brightness slider now is a master brightness that applies to both the RGB channel and the white channel.

If this is the case then it is likely that the issue I am seeing it just a residual bug that you are fixing here: https://github.com/esphome/esphome/pull/2045

I uploaded a version to my LED Strip built against ``origin/dev`` again, to triple check I hadn't missed something (since I have been basically ignoring the ``Color brightness`` slider since in the past it has been useless), but even with dropping that I still saw the same functionality: The ``Color brightness`` slider will jump back to max after any other light state change.

Am I correct in assuming that with v1.20, the correct way one would turn on **JUST** the white channel of a RGBW strip would be:
- ``Brightness`` slider: MAX
-  ``Color brightness`` slider: 0
- ``White brightness`` slider: MAX
If so, this does currently not work.

> I did notice in your screenshot in the other issue that you set the brightness to a very low level. This will now also dim the white channel, and as I said some strips don't support very low light levels. What happens if you set the brightness slider to a higher value?

White channel turns off after transition regardless of brightness settings.

WeekendWarrior1, 2021-07-20:
So the issue actually seems to be:
### "Color brightness slider is ignored"
[image]
Results in: 
```
[13:31:37][D][neopixelbus:090]: writing addressable buffer
[13:31:37][D][neopixelbus:094]:   [ 0] red=255 green=255 blue=255 white=255
```

WeekendWarrior1, 2021-07-20:
I thought I was seeing that the white channel level wasn't properly being decoupled from the RGB channel levels, but:
[image]
Results in:
```
[13:36:28][D][neopixelbus:090]: writing addressable buffer
[13:36:28][D][neopixelbus:094]:   [ 0] red=255 green=255 blue=255 white= 64
```
Which is expected.

oxan, 2021-07-20:
> So the issue actually seems to be:
> ### "Color brightness slider is ignored"

Yes, that's known and pending changes in HA, but it shouldn't have any impact on the white channel.

WeekendWarrior1, 2021-07-20:
> Yes, that's known and pending changes in HA, but it shouldn't have any impact on the white channel.

With your assistance, https://github.com/esphome/esphome/pull/2045 and adding the debug code you suggested, I must apologise for wasting your time, since the issue was not directly to do with the white channel after all, but the color slider.

WeekendWarrior1, 2021-07-20:
Though thinking about it, until those pending changes are made in HA, people with RGBW strips are going to lose the ability of correctly controlling their RGBW strips. Is this the 1.20 release being held off on until those changes are made? If not, then a regression has been introduced.

oxan, 2021-07-20:
> Though thinking about it, until those pending changes are made in HA, people with RGBW strips are going to lose the ability of correctly controlling their RGBW strips. 

That depends on your definition of correctly. The pre-v1.20 behaviour isn't compatible with what HA expects, how non-addressable RGBW lights work and how effects work.

> Is the 1.20 release being held off on until those changes are made? 

No. The color brightness slider fix might hit HA 2021.8, or it might end up bundled with color model support in HA 2021.9.

> If not, then a regression has been introduced, since in v1.19 a user could solely turn on the white channel of their RGBW strips, and that is currently no longer possible.

You can still do that from YAML and the API.

oxan, 2021-08-03:
The fix has landed in HA 2021.8.

closed 2021-08-03 by oxan