esphome/issues #2248: Double action on_time_sync

radio-m, 2021-07-21:
### The problem

Hi! I am trying to restrict the thermostat from turning on when the dishwasher is turned on. I created a binary sensor on the platform: homeassistant. After turning on the device and synchronizing the time, I check the time period and the state of the binary sensor. Depending on this data, I turn on or off the thermostat. But I have a wrong job. The thermostat turns on and then turns off.

### What is version of ESPHome has the issue?

1.19.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

BlitzWolf SHP8 esp8285 based

### Component causing the issue

time

### Example YAML snippet

```yaml
substitutions:
  device_name: BW-SHP8
  node_number: "21"
  current_res: "0.00221"
  voltage_div: "820"

  dallas_pin: GPIO2
  button_pin: GPIO3
  led_pin: GPIO13
  relay_pin: GPIO14
  sel_pin: GPIO12
  cf_pin: GPIO04
  cf1_pin: GPIO05

esphome:
  name: blitzshp8
  platform: ESP8266
  board: esp01_1m

packages:
  device_base: !include common/device_base.yaml

# web_server:
#   port: 80
  
time:
  platform: homeassistant
  id: ha_time
  on_time:
    - seconds: 0
      minutes: 59
      hours: 6
      then:
        - climate.control:
            id: boiler_thermostat
            mode: 'OFF'
    - seconds: 0
      minutes: 01
      hours: 23
      then:
        - if:
            condition:
              - binary_sensor.is_off: dishwasher
[... 304 lines cut ...]
[10:41:38][D][climate:398]:   Target Temperature: 70.00°C
[10:41:38][D][time:040]: Synchronized time: Wed Jul 21 10:41:37 2021
[10:41:38][D][homeassistant.binary_sensor:025]: 'binary_sensor.bme280': Got state ON
[10:41:38][D][binary_sensor:034]: 'dishwasher': Sending initial state ON
[10:41:40][W][main:687]: Synchronized
[10:41:41][D][hlw8012:071]: Got power=1377.5W, voltage=201.1V
[10:41:41][D][sensor:117]: 'BW-SHP8_voltage': Sending state 201.07770 V with 0 decimals of accuracy
[10:41:41][D][sensor:117]: 'BW-SHP8_power': Sending state 1377.46533 W with 1 decimals of accuracy
[10:41:45][W][main:693]: State 1
[10:41:45][D][climate:009]: 'Boiler Thermostat' - Setting
[10:41:45][D][climate:013]:   Mode: OFF
[10:41:45][D][switch:025]: 'BW-SHP8_relay' Turning OFF.
[10:41:45][D][switch:045]: 'BW-SHP8_relay': Sending state OFF
[10:41:45][D][light:275]: 'BW-SHP8 Blue LED' Setting:
[10:41:45][D][light:284]:   State: ON
[10:41:45][D][light:314]:   Transition Length: 1.0s
[10:41:45][D][climate:369]: 'Boiler Thermostat' - Sending state:
[10:41:45][D][climate:372]:   Mode: OFF
[10:41:45][D][climate:374]:   Action: OFF
[10:41:45][D][climate:392]:   Current Temperature: 67.56°C
[10:41:45][D][climate:398]:   Target Temperature: 70.00°C
[10:41:46][D][hlw8012:071]: Got power=1139.1W, voltage=203.2V
[10:41:46][D][sensor:117]: 'BW-SHP8_voltage': Sending state 203.15858 V with 0 decimals of accuracy
[10:41:46][D][sensor:117]: 'BW-SHP8_power': Sending state 1139.12708 W with 1 decimals of accuracy
```

OttoWinter, 2021-07-26:
Can you highlight what is supposed to happen and what actually happens in the log? It's hard to figure out with that big config/log dump.

radio-m, 2021-07-26:
> Can you highlight what is supposed to happen and what actually happens in the log? It's hard to figure out with that big config/log dump.

```
[10:41:32][W][main:687]: Synchronized
[10:41:38][W][main:693]: State 0
[10:41:38][D][climate:009]: 'Boiler Thermostat' - Setting
[10:41:38][D][climate:013]:   Mode: HEAT

[10:41:40][W][main:687]: Synchronized
[10:41:45][W][main:693]: State 1
[10:41:45][D][climate:009]: 'Boiler Thermostat' - Setting
[10:41:45][D][climate:013]:   Mode: OFF
```

OttoWinter, 2021-07-26:
Ah ok so that is kind of expected.

The time is synchronized every time a client connects to the API server. In this case any time the HA instance _and_ any time you view the logs (because that client also transmits the time).

radio-m, 2021-07-26:
But if I put the SNTP time server, then everything works correctly

OttoWinter, 2021-07-26:
Yes, because that synchronization works a very different way. Anyway, you shouldn't rely on `on_time_sync`'s timing - even with SNTP we don't control when time is synchronized, it just happens to sync time less often.

closed 2021-07-26 by OttoWinter