esphome/issues #2250: Anova Cooker not connecting

jmarshak, 2021-07-22:
### The problem

Turned off BT on my phone.  I found my anvoa cooker's mac address.  ESP32 tries to connect but fails.  I use this nodemcu to connect with xiami temp+humidity trackers, so I know the BT is setup correctly.  The cooker works well with Anova app.  Using Anova Nano model.  Model: AN400-US00

Appreciate your work, ESPHome is so awesome.  I would love to help debug, but I don't know much about how BLE works.  Thanks!

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

Anova nano

### Example YAML snippet

```yaml
esphome:
  name: ble_base
  platform: ESP32
  board: esp-wrover-kit

logger:

esp32_ble_tracker:

switch:
  - platform: restart
    name: "BLE_base1-restart"
    id: restart_switch
  - platform: ble_client
    ble_client_id: my_anova
    name: "Enable Anova sous vide cooker"

ble_client:
  - mac_address: 00:81:F9:B9:E0:F2
    id: my_anova
    
climate:
  - platform: anova
    name: "Anova sous vide cooker"
    ble_client_id: my_anova
```

### Anything in the logs that might be useful for us?

```txt
19:02:24][I][ble_client:083]: Attempting BLE connection to 00:81:f9:b9:e0:f2
[19:02:25][I][ble_client:159]: Service UUID: 0x1800
[19:02:25][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[19:02:25][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0x2
[19:02:25][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[19:02:25][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[19:02:25][I][ble_client:159]: Service UUID: 0x1801
[19:02:25][I][ble_client:160]:   start_handle: 0x8  end_handle: 0x8
[19:02:25][I][ble_client:159]: Service UUID: 0E1400000AF145-82A2-4277-3E63-054C68
[19:02:25][I][ble_client:160]:   start_handle: 0x9  end_handle: 0xffff
[19:02:25][I][ble_client:339]:  characteristic 0E1400010AF145-82A2-4277-3E63-054C68, handle 0xb, properties 0x8
[19:02:25][I][ble_client:339]:  characteristic 0E1400020AF145-82A2-4277-3E63-054C68, handle 0xe, properties 0x10
[19:02:25][I][ble_client:339]:  characteristic 0E1400030AF145-82A2-4277-3E63-054C68, handle 0x12, properties 0x10
[19:02:25][W][anova:062]: [Anova sous vide cooker] No control service found at device, not an Anova..?
```

### Additional information

 Anova Nano model.  Model: AN400-US00

jmarshak, 2021-07-22:
@buxtronix

buxtronix, 2021-07-23:
It looks like the Nano uses a different API over BLE, as seen in this pyanova thread:

https://github.com/c3V6a2Vy/pyanova/issues/10

Until this gets worked out, I don't think they can be supported now. And I don't have a Nano to test with. Will need to update the docs here to note this.

jmarshak, 2021-07-24:
> It looks like the Nano uses a different API over BLE, as seen in this pyanova thread:
> 
> [c3V6a2Vy/pyanova#10](https://github.com/c3V6a2Vy/pyanova/issues/10)
> 
> Until this gets worked out, I don't think they can be supported now. And I don't have a Nano to test with. Will need to update the docs here to note this.

I see, bummer.  Well thanks for following up and updating the docs.  Hopefully they well make progress eventually.

closed 2021-12-05 by github-actions[bot]