esphome/issues #2254: Can't connect on 1.20.0

OptimusGREEN, 2021-07-22:
### The problem

Having updated my esp to 1.20.0 I can no longer connect and get the following error.

```WARNING Choked at offset 33075 while unpacking b'\x00\x00\x00\x00\x00\x04\x00\x00\x00\x07\x00\x005homeassistant45009 [080a1e15d66b44b3bf58067cc44a01af]\x0c_workstation\x04_tcp\x05local\x00\x00\xff\x00\x015homeassistant45013 [080a1e15d66b44b3bf58067cc44a01af]\xc0B\x00\xff\x00\x01\x12homeassistant45013\xc0T\x00\xff\x00\x015homeassistant45007 [080a1e15d66b44b3bf58067cc44a01af]\xc0B\x00\xff\x00\x015homeassistant44983 [080a1e15d66b44b3bf58067cc44a01af]\xc0B\x00\xff\x00\x01\xc0\x9b\x00\x01\x80\x01\x00\x00\x00x\x00\x04\nO\x00d\xc0\xb4\x00\x10\x80\x01\x00\x00\x00x\x00\x01\x00\xc0\xb4\x00!\x80\x01\x00\x00\x00x\x00 \x00\x00\x00\x00\x00\x00\x12homeassistant45007\x05local\x00\xc0\x0c\x00\x10\x80\x01\x00\x00\x00x\x00\x01\x00\xc0\x0c\x00!\x80\x01\x00\x00\x00x\x00 \x00\x00\x00\x00\x00\x00\x12homeassistant45009\x05local\x00\xc0_\x00\x10\x80\x01\x00\x00\x00x\x00\x01\x00\xc0_\x00!\x80\x01\x00\x00\x00x\x00 \x00\x00\x00\x00\x00\x00\x12homeassistant45013\x05local\x00'
WARNING Exception occurred:
Traceback (most recent call last):
  File "/opt/esphome/esphome/zeroconf.py", line 203, in __init__
    self.read_others()
  File "/opt/esphome/esphome/zeroconf.py", line 258, in read_others
    domain = self.read_name()
  File "/opt/esphome/esphome/zeroconf.py", line 297, in read_name
    length = self.data[off]
IndexError: index out of range```

and then...

```WARNING Error resolving IP address of kitchen_touch_switch.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 1 seconds```

If I use a previous version, everything works ok.

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

zeroconf?

### Example YAML snippet

```yaml
no
```

### Anything in the logs that might be useful for us?

```txt
see above
```

### Additional information

this is an esp controlling a nextion screen.

raidnet-ms, 2021-07-23:
did you check nextion changelog?

OptimusGREEN, 2021-07-23:
Yes had a look, have I missed something?

raidnet-ms, 2021-07-23:
I only told you because 1.20 had lot of chenges for that display that may affect your configuration.

OptimusGREEN, 2021-07-23:
I've had another look at the changelog and the only nextion breaking change isn't relevant to my case as far as i can tell.
Everything compiles and uploads, but once uploaded, cant connect when clicking logs.

OptimusGREEN, 2021-07-23:
ok, having done a port scan on the device, port 6053 isn't open. This seems to be the issue but no idea what would cause that.

OptimusGREEN, 2021-07-24:
@SenexCrenshaw maybe you could help?

OptimusGREEN, 2021-07-24:
I've narrowed the issue down to being introduced in this [merge](https://github.com/esphome/esphome/pull/1464) as 1.20.0b1 works ok but breaks in 1.20.0b2

ssieb, 2021-07-24:
I don't see how that change would affect mdns at all and this isn't breaking anything for other people.  Does it work without any nextion bits in the config?  Can you get serial logs (maybe use the secondary uart tx pin if you're using the hardware uart for the nextion)?

OptimusGREEN, 2021-07-24:
I don't understand it either, but works fine in 1.20.0b1 and not in any version afterwards. 
My config is quite sectioned and compartmentalised so removing the nextion bits would effectively be removing it all. I do believe its something in the nextion changes but without logs, I'm stuck.
Is there a guide to doing the serial logging thing you mention?

ssieb, 2021-07-24:
Can you join the esphome Discord server?  It will be a lot easier that way.

OptimusGREEN, 2021-07-25:
Yeah I think I am already joined. I feel questions get lost in the waves in there sometimes. I will pop in later then.

OptimusGREEN, 2021-07-25:
managed to get some logs while connected physically to computer.

seems to be boot looping...

I hope someone can make some sense of this...

`Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.wchusbserialfa1240 with baud rate 115200
[14:42:21];ld\x9c\x9e|\x84l\xe0<

                                \x84
                                    d\xecc|\x8e\xc3\xe4\x9br\x9b#\x84
                                                                     #\xc4\xf3og\x9flng\x9c\xe3\xe4cp\xc4\xc7ds$sdp\xfb'\xe0
                                                                                                                            \x83
                                                                                                                                $
                                                                                                                                 \x84\x9c
                                                                                                                                         c
                                                                                                                                          o\xe2<\xecd\x87\x87#\x8c\xf3og\xe7lćd`\x98o
                                                                                                                                                                                     $`'{\x87\x93\xdb'
                                                                                                                                                                                                      cd`;\x93\xdb'
                                                                                                                                                                                                                   c$`\x9cb
                                                                                                                                                                                                                           \x8e\x93$s\x9b`\xfc\x83o\xdc[I][logger:170]: Log initialized
[14:42:21][C][ota:378]: There have been 0 suspected unsuccessful boot attempts.
[14:42:21][I][app:029]: Running through setup()...
[14:42:21][C][uart_esp8266:046]: Setting up UART bus...
[14:42:21][D][text_sensor:015]: 'esp_mac': Sending state '8C:CE:4E:E1:E5:E9'
[14:42:21][C][wifi:037]: Setting up WiFi...
[14:42:21][I][wifi:250]: WiFi Connecting to 'IoT'...
[14:42:22][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:22][W][nextion:072]: Nextion is not connected! 
[14:42:23][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:23][W][nextion:072]: Nextion is not connected! 
[14:42:23][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:23][W][nextion:072]: Nextion is not connected! 
[14:42:24][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:24][W][nextion:072]: Nextion is not connected! 
[14:42:25][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:25][W][nextion:072]: Nextion is not connected! 
[14:42:26][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:26][W][nextion:072]: Nextion is not connected! 
[14:42:26][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[14:42:26][W][nextion:072]: Nextion is not connected! 
[14:42:27][I][wifi:513]: WiFi Connected!
[14:42:27][C][wifi:359]:   SSID: 'IoT'
[14:42:27][C][wifi:360]:   IP Address: 10.79.8.9
[14:42:27][C][wifi:362]:   BSSID: 86:AC:B9:C1:E4:F5
[14:42:27][C][wifi:363]:   Hostname: 'kitchen_touch_switch_backup'
[14:42:27][C][wifi:367]:   Signal strength: -35 dB ▂▄▆█
[14:42:27][C][wifi:371]:   Channel: 11
[14:42:27][C][wifi:372]:   Subnet: 255.255.255.0
[14:42:27][C][wifi:373]:   Gateway: 10.79.8.1
[14:42:27][C][wifi:374]:   DNS1: 10.79.8.1
[14:42:27][C][wifi:375]:   DNS2: (IP unset)
[14:42:27][D][wifi:522]: Disabling AP...
[14:42:27][C][ota:029]: Over-The-Air Updates:
[14:42:27][C][ota:030]:   Address: kitchen_touch_switch_backup.local:8266
[14:42:27][C][api:022]: Setting up Home Assistant API server...
[14:42:27][C][sntp:017]: Setting up SNTP...
[14:42:27][D][TimeoutTime:035]: Timeout set to: 30000 ms
[14:42:27][I][app:059]: setup() finished successfully!
[14:42:27]
[14:42:27]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[14:42:27]
[14:42:27]Exception (29):
WARNING Exception type: Access to invalid address: STORE (wild pointer?)
[14:42:27]epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d kitchen_touch_switch_backup -t idedata
[14:42:28]
[... 114 lines cut ...]
[14:42:28]3ffff340:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff350:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff360:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff370:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff380:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff390:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff3a0:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff3b0:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff3c0:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff3d0:  feefeffe feefeffe feefeffe feefeffe  
[14:42:28]3ffff3e0:  feefeffe feefeffe feefeffe feefeffe  0000000 3fff293c 402150b4  
WARNING Decoded 0x402150b4: loop
[... cut ...]
[14:42:28]--------------- CUT HERE FOR EXCEPTI[I][logger:170]: Log initialized
[14:42:28][C][ota:378]: There have been 1 suspected unsuccessful boot attempts.
[14:42:28][I][app:029]: Running through setup()...
[14:42:28][C][uart_esp8266:046]: Setting up UART bus...
[14:42:28][D][text_sensor:015]: 'esp_mac': Sending state '8C:CE:4E:E1:E5:E9'
[14:42:28][C][wifi:037]: Setting up WiFi...
[14:42:28][I][wifi:250]: WiFi Connecting to 'IoT'...
[14:42:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
`

mihaiblaga89, 2021-07-26:
@OptimusGREEN had the same issues, without the connection error. I had to plug nextion into the Editor (with the Editor updated to the latest version) and reupload the screens with the mentioned 
```
baud=115200  
bkcmd=0   
```   
_to the program.s source file (in the Nextion Editor) before the page line_.  

After that I connected the display to the ESP and changed the baudrate to 115200 and everything worked perfectly.

OptimusGREEN, 2021-07-26:
@mihaiblaga89 Hi, thanks for this. Unfortunately I already have this and I'm having the same issues.

29axe, 2021-07-30:
Hello,
I also noticed connection issues lately on the addon for home assistant.
It seems that the mdns IP resolution does not work anymore. Btw I can toggle all sonoff devices on/off through home assistant, so they are working great. I just can't update or read the logs through esphome plugin.

All my sonoffs basic appear offline and when I try to install or view logs I get the following error:

```
INFO Successfully compiled program.
INFO Resolving IP address of sonoff_basic_r3.local
ERROR Error resolving IP address of sonoff_basic_r3.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```

I can get the the devices online sometimes by pinging the devices using their .local url but it does not stay online long enough to update.

OptimusGREEN, 2021-07-31:
ok so i tried on multiple d1 mini's with same outcome. I then tried on an esp32 and it was ok until I added more sensors then started to get similar results until i went into platformio and incresed the stack size as seen in another post. 

Its looking like the new nextion additions are using up lots more of the stack which in turn is tipping it over the edge causing a crash and constant boot loop in configs that are a bit more involved than most.

Having an option to set stack size in the config could possibly resolve this.

OptimusGREEN, 2021-08-01:
looks like this [PR](https://github.com/esphome/esphome/pull/1651) might be the solution to this.

closed 2021-12-26 by github-actions[bot]

paul-at-pebbletech, 2022-01-30:
@OptimusGREEN did you find the [https://github.com/esphome/esphome/pull/1651](url) fixed the issue with the displays? If so what is the manual change to the stack size for esp8266. Thanks

OptimusGREEN, 2022-01-31:
Hi,

No it didn't fix it. I had to switch to esp32 to continue using nextion and all my functions.

paul-at-pebbletech, 2022-01-31:
Ok thanks at the moment I have found that if I set the debug to very verbose for the nextion device the my codebase works, I guess this just slows down the whole process or maybe it there is some other mechanism at work. Iam just hoping that it continues to work as I add nextion pages so I dont have to move device