esphome/issues #2256: Globals String array not returning valid values

soloam, 2021-07-23:
### The problem

I tried to declare a globals String array, and the return is not what I expected:

```yaml
globals:
  - id: allowed_list
    type: String[2]
    initial_value: '{"1234567890-100-1","0987654321-100-1"}'
```

When I check it on the lamda:

```c++
  //Inside for loop
  uuid_t = id(allowed_list)[x];
  uuid_t.replace("-","");
  uuid_t.toLowerCase();
  
  ESP_LOGD("ble_adv", "Testing %s", uuid_t.c_str());
```
uuid_t is emty, no value is printed

But if I do all in the lamda

```c++
//Declaration
String allowed_list[2] = {"1234567890-100-1","0987654321-100-1"};
```
...

```c++
  //Inside for loop
  uuid_t = allowed_list[x];
  uuid_t.replace("-","");
  uuid_t.toLowerCase();
  
  ESP_LOGD("ble_adv", "Testing %s", uuid_t.c_str());
```
This works Ok

### What is version of ESPHome has the issue?

v1.19.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

globals

### Example YAML snippet

```yaml
globals:
  - id: allowed_list
    type: String[2]
    initial_value: '{"1234567890-100-1","0987654321-100-1"}'
```

oxan, 2021-07-23:
Your YAML fragment actually makes my compiler crash :scream: 

```
Compiling .pioenvs/lambdatest/src/main.cpp.o
In file included from src/esphome.h:15:0,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h: In instantiation of 'esphome::globals::GlobalsComponent<T>::GlobalsComponent(std::array<typename std::remove_extent<_Up>::type, std::extent<_Tp>::value>) [with T = String [2]; typename std::remove_extent<_Up>::type = String]':
lambdatest.yaml:67:98:   required from here
src/esphome/components/globals/globals_component.h:15:116: internal compiler error: Segmentation fault
   explicit GlobalsComponent(std::array<typename std::remove_extent<T>::type, std::extent<T>::value> initial_value) {
                                                                                                                    ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/lambdatest/src/main.cpp.o] Error 1
```

What happens if you use `type: std::string` (so without the array length)?

soloam, 2021-07-23:
That was the first type that I tried! The result was the same! Strange that it breaks your compiler, my runs okay, but the arrays items all look empty, I print them and nothing, and never matches!

soloam, 2021-07-27:
I tried again with "std::string[2]" and does the same! Can this be a bug with the globals plugin?

oxan, 2021-07-29:
Judging from the code, the globals integration doesn't support C-style arrays (and I don't think that can easily be added). You should use `std::array` instead: `type: std::array<std::string, 2>`.

soloam, 2021-11-27:
Any news about this?

soloam, 2022-01-21:
Any one knows if this will be possible?

Thank you

jesserockz, 2022-01-21:
I personally use a `std::vector<std::string>>` type with no issues

soloam, 2022-01-27:
This now works for me! Tks for the help

closed 2022-01-27 by soloam