esphome/issues #2258: Sonoff Dual R3 ESPHOME

interkom, 2021-07-23:
### The problem

Good afternoon. Help me to understand.
Bought and reflashed Sonoff Dual R3 on ESPHOME
Everything works, except for energy monitoring.

Did so
Created a program in ESPHOME, saved as file.bin
Then I connected to Sonoff Dual R3, cleaned it completely, and flashed it as follows.
bootloader_dout_40m.bin - 0x1000
partitions.bin - 0x8000
boot_app0.bin - 0xe000
file.bin - 0x10000
The files bootloader_dout_40m.bin, partitions.bin, boot_app0.bin are taken from the Tasmota firmware along the path Tasmota-firmware \ static \ esp32 \ .... since I don’t know where to get these files for ESPHOME

here is my firmware code and logs
https://pastebin.com/raDPrfNq

changed the UART GPIO settings to 26 - RX, 25 - TX as in Tasmota, but did not bring any result.

What am I doing wrong ? Or ESPHOME for ESP32 doesn't support cse7766 yet?

On Tasmota 9.5.0 by Theo Arends, the relay is working fine and power monitoring is working

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Sonoff Dual R3

### Component causing the issue

cse7766

### Example YAML snippet

```yaml
esphome:
  name: esp32_lip_bas_sonoff_dual_r3
  platform: ESP32
  board: esp32dev

wifi:
  networks:
  - ssid: "INTERCOTTAGE-L"
    password: "secret"
  - ssid: "INTERCOTTAGE-F"
    password: "secret"
  ap:
    ssid: "esp32_lip_bas_sonoff_dual_r3"
    password: "secret"
  manual_ip:
    static_ip: 192.168.5.241
    gateway: 192.168.5.1
    subnet: 255.255.255.0

captive_portal:

logger:
#  baud_rate: 0
[... 173 lines cut ...]
[22:04:51][C][ota:030]:   Address: 192.168.5.241:3232
[22:04:51][C][ota:032]:   Using Password.
[22:04:51][C][api:095]: API Server:
[22:04:51][C][api:096]:   Address: 192.168.5.241:6053
[22:04:51][C][wifi_info:009]: WifiInfo IPAddress 'esp32_lip_bas_sonoff_dual_r3_ip'
[22:04:51][C][wifi_signal.sensor:009]: WiFi Signal 'esp32_lip_bas_sonoff_dual_r3_wi-fi'
[22:04:51][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[22:04:51][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[22:04:51][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[22:04:51][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[22:04:51][D][cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:04:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[22:04:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_current': Sending state 0.00000 A with 2 decimals of accuracy
[22:04:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_power': Sending state 0.00000 W with 1 decimals of accuracy
[22:04:57][D][switch:021]: 'esp32_lip_bas_sonoff_dual_r3_relay_1' Turning ON.
[22:04:57][D][switch:045]: 'esp32_lip_bas_sonoff_dual_r3_relay_1': Sending state ON
[22:04:57][D][switch:021]: 'esp32_lip_bas_sonoff_dual_r3_relay_2' Turning ON.
[22:04:57][D][switch:045]: 'esp32_lip_bas_sonoff_dual_r3_relay_2': Sending state ON
[22:05:01][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_uptime': Sending state 93760.47656 s with 0 decimals of accuracy
[22:05:51][D][cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:05:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[22:05:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_current': Sending state 0.00000 A with 2 decimals of accuracy
[22:05:51][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_power': Sending state 0.00000 W with 1 decimals of accuracy
[22:05:53][D][sensor:127]: 'esp32_lip_bas_sonoff_dual_r3_wi-fi': Sending state -14.00000 dBm with 0 decimals of accuracy
```

cdheiser, 2021-07-24:
The Dual R3 uses the CSE7761 which isn't supported yet:  https://github.com/esphome/feature-requests/issues/1151

interkom, 2021-07-24:
Thank you very much

closed 2021-07-24 by interkom