esphome/issues #2259: 1.20 breaks manual control of bang_bang from AUTO/HEAT_COOL to HEAT and does not honor mode

ptr727, 2021-07-24:
### The problem

v1.20 changed the climate component behavior, again, and broke my bang_bang controller, [again](https://github.com/esphome/issues/issues/2154) :(

My config uses `on_boot` to turn the bang_bang to AUTO/HEAT_COOL.
A [new](https://github.com/esphome/issues/issues/2233) `mode` option was added to climate that is supposed to alleviate the need for using `on_boot` to turn the thermo on.
But this does not work:
Yaml:
```
# https://esphome.io/components/climate/index.html
climate:

  # https://esphome.io/components/climate/bang_bang.html
  - platform: bang_bang
    name: ${device_name}_bangbang
    id: bangbang
    sensor: recirc_temp
    mode: HEAT_COOL
    default_target_temperature_low: 30 # ~85F
    default_target_temperature_high: 40 # ~105F
    heat_action:
      - logger.log: "BangBang : Heat Action, Turning Relay On."
      - switch.turn_on: relay
    idle_action:
      - logger.log: "BangBang : Idle Action, Turning Relay Off."
      - switch.turn_off: relay
    visual:
      min_temperature: 20
      max_temperature: 50
      temperature_step: 0.5
```

Validate:
```
INFO Reading configuration /config/sonoff_th10_hot_recirc_pump.yaml...
INFO Detected timezone 'PST' with UTC offset -8 and daylight saving time from 13 March 02:00:00 to 06 November 02:00:00
Failed config

climate.bang_bang: [source /config/sonoff_th10_hot_recirc_pump.yaml:149]
  platform: bang_bang
  name: sonoff_th10_hot_recirc_pump_bangbang
  id: bangbang
  sensor: recirc_temp
  
  [mode] is an invalid option for [climate.bang_bang]. Please check the indentation.
  mode: HEAT_COOL [source /config/sonoff_th10_hot_recirc_pump.yaml:153]
  default_target_temperature_low: 30
  default_target_temperature_high: 40
  heat_action: 
    - logger.log: BangBang : Heat Action, Turning Relay On.
    - switch.turn_on: relay
  idle_action: 
    - logger.log: BangBang : Idle Action, Turning Relay Off.
    - switch.turn_off: relay
  visual: 
    min_temperature: 20
    max_temperature: 50
    temperature_step: 0.5
```

Trying with `HEAT`:
```
INFO Reading configuration /config/sonoff_th10_hot_recirc_pump.yaml...
INFO Detected timezone 'PST' with UTC offset -8 and daylight saving time from 13 March 02:00:00 to 06 November 02:00:00
Failed config

climate.bang_bang: [source /config/sonoff_th10_hot_recirc_pump.yaml:149]
  platform: bang_bang
  name: sonoff_th10_hot_recirc_pump_bangbang
  id: bangbang
  sensor: recirc_temp
  
  [mode] is an invalid option for [climate.bang_bang]. Please check the indentation.
  mode: HEAT [source /config/sonoff_th10_hot_recirc_pump.yaml:153]
  default_target_temperature_low: 30
  default_target_temperature_high: 40
  heat_action: 
    - logger.log: BangBang : Heat Action, Turning Relay On.
    - switch.turn_on: relay
  idle_action: 
    - logger.log: BangBang : Idle Action, Turning Relay Off.
    - switch.turn_off: relay
  visual: 
    min_temperature: 20
    max_temperature: 50
    temperature_step: 0.5
```

It seems it is also now no longer possible to manually turn the heat on, i.e. turn the mode from AUTO/HEAT_COOL to HEAT, as HEAT is now the only allowed mode.

Yaml:
```
  # Periodic recirc interval
  - interval: 30min
    then:
      # When ambient temperature is higher than recirc low temp threshold, recirc never turns on.
      # Periodically circulate hot water even when recirc is not calling for heat, making sure water in lines are never cold.
      - logger.log: "RecircTimer : Setting Pump to Heat."
      - climate.control: 
          id: bangbang
          mode: HEAT
      - logger.log: "RecircTimer : Delay Start."
      # Periodic recirc runtime
      - delay: 2min 
      - logger.log: "RecircTimer : Delay End."
      - logger.log: "RecircTimer : Setting Pump to Auto."
      - climate.control: 
          id: bangbang
          mode: HEAT_COOL
```

Log:
```
[22:41:21][D][main:201]: RecircTimer : Delay End.
[22:41:21][D][main:204]: RecircTimer : Setting Pump to Auto.
[22:41:21][D][climate:009]: 'sonoff_th10_hot_recirc_pump_bangbang' - Setting
[22:41:21][W][climate:053]:   Mode HEAT_COOL is not supported by this device!
```

My conclusion is that 1.19.2 works, and 1.20.0 fails.
1.20 has no way to manually turn the controller from auto to heat.
1.20 does not honor the `mode` [docs](https://esphome.io/components/climate/index.html) 

For a bang_bang:
1) How am I supposed to set it to default to AUTO mode on startup?
2) How am I supposed to manually turn the controller to HEAT and to AUTO?

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Sonoff TH10

### Component causing the issue

bang_bang

### Example YAML snippet

```yaml
# https://esphome.io/components/climate/index.html
climate:

  # https://esphome.io/components/climate/bang_bang.html
  - platform: bang_bang
    name: ${device_name}_bangbang
    id: bangbang
    sensor: recirc_temp
    mode: HEAT_COOL
    default_target_temperature_low: 30 # ~85F
    default_target_temperature_high: 40 # ~105F
    heat_action:
      - logger.log: "BangBang : Heat Action, Turning Relay On."
      - switch.turn_on: relay
    idle_action:
      - logger.log: "BangBang : Idle Action, Turning Relay Off."
      - switch.turn_off: relay
    visual:
      min_temperature: 20
      max_temperature: 50
      temperature_step: 0.5
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/sonoff_th10_hot_recirc_pump.yaml...
INFO Detected timezone 'PST' with UTC offset -8 and daylight saving time from 13 March 02:00:00 to 06 November 02:00:00
Failed config

climate.bang_bang: [source /config/sonoff_th10_hot_recirc_pump.yaml:149]
  platform: bang_bang
  name: sonoff_th10_hot_recirc_pump_bangbang
  id: bangbang
  sensor: recirc_temp
  
  [mode] is an invalid option for [climate.bang_bang]. Please check the indentation.
  mode: HEAT [source /config/sonoff_th10_hot_recirc_pump.yaml:153]
  default_target_temperature_low: 30
  default_target_temperature_high: 40
  heat_action: 
    - logger.log: BangBang : Heat Action, Turning Relay On.
    - switch.turn_on: relay
  idle_action: 
    - logger.log: BangBang : Idle Action, Turning Relay Off.
    - switch.turn_off: relay
  visual: 
    min_temperature: 20
    max_temperature: 50
    temperature_step: 0.5
```

### Additional information

https://github.com/esphome/issues/issues/2233
https://github.com/esphome/esphome/pull/1994
https://github.com/esphome/esphome/pull/2053
https://github.com/esphome/esphome/pull/1942
https://github.com/esphome/esphome/pull/1941
https://github.com/esphome/issues/issues/2154

OttoWinter, 2021-07-26:
> A new mode option was added to climate that is supposed to alleviate

1. that option is called `default_mode`
2. it was added to the thermostat climate platform, not `bang_bang`

> It seems it is also now no longer possible to manually turn the heat on, i.e. turn the mode from AUTO/HEAT_COOL to HEAT, as HEAT is now the only allowed mode.

Correct, that was a breaking change (and was made to keep it consistent with other climate devices and what HA specifies. Unfortunately it seems I forgot to tag it as a breaking change. Will fix that.

closed 2021-07-26 by frenck

ptr727, 2021-07-26:
I tried `default_mode` on a thermostat and bang_bang component, does not validate:
`[default_mode] is an invalid option for [climate.bang_bang]. Please check the indentation.` 
`[default_mode] is an invalid option for [climate.thermostat]. Did you mean [heat_mode], [auto_mode], [fan_only_mode]?`
Did the change not make it into 1.20, or am I doing something wrong?

Removing the ability to manually turn the device to heat or cool is a major degradation in functionality, in my opinion.
How should I manually turn a thermostat to heat or cool ignoring the temperature now that this ability has been removed?

When I the code below I was advised by discord/dev to change the climate state and not to not change the underlying relay state, but this functionality is now gone.
```
  # Periodic recirc interval
  - interval: 30min
    then:
      # When ambient temperature is higher than recirc low temp threshold, recirc never turns on.
      # Periodically circulate hot water even when recirc is not calling for heat, making sure water in lines are never cold.
      - logger.log: "RecircTimer : Setting Pump to manual Heat."
      - climate.control: 
          id: bangbang
          mode: HEAT
      - logger.log: "RecircTimer : Delay Start."
      # Periodic recirc runtime
      - delay: 2min 
      - logger.log: "RecircTimer : Delay End."
      - logger.log: "RecircTimer : Setting Pump to Auto."
      - climate.control: 
          id: bangbang
          mode: HEAT_COOL
```

ptr727, 2021-07-30:
Still [hoping](https://github.com/esphome/issues/issues/2270) that `default_mode` will be moved to climate so that thermostat, bang_bang_ and pid can get the benefit.

As for manual control, found a workaround, posting it here in case it helps somebody else:
```
  # Periodic recirc interval
  # When ambient temperature is higher than recirc low temp threshold, recirc never turns on.
  # Periodically circulate hot water even when recirc is not calling for heat, making sure water in lines are never cold.
  - interval: ${recirc_interval}
    then:
      # Disable thermostat before modifying underlying relay state
      - logger.log: "RecircTimer : Setting Thermostat to Off."
      - climate.control: 
          id: bangbang
          mode: "OFF"
      # Turn relay on
      - logger.log: "Recirctimer : Turning Relay On."
      - switch.turn_on: relay
      - logger.log: "RecircTimer : Delay Start."
      # Sleep
      - delay: ${recirc_duration}
      - logger.log: "RecircTimer : Delay End."
      # Turn relay off
      - logger.log: "RecircTimer : Turning Relay Off."
      - switch.turn_off: relay
      # Turn thermostat back on for normal operation
      - logger.log: "RecircTimer : Setting Thermostat to Heat."
      - climate.control: 
          id: bangbang
          mode: HEAT
```