esphome/issues #2261: Nextion is not connected.

g-enzo, 2021-07-24:
### The problem

I had a complex setup with Nextion working and running fine. I made some changes to the order and names of some pages on the Nextion tft file and started getting "Nextion is not connected" error in the logs and obviously it all stopped working. Very odd.

As part of my troubleshooting and because the changes I made to the Nextion tft file had no reference in the esphome yaml I reduced the esphome yaml and the Nextion tft file to bare minimium and am still getting the error.

What am I missing, please?

Cheers People

Gavin

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Adafruit HUZZAH32

### Component causing the issue

Nextion

### Example YAML snippet

```yaml
esphome:
  name: thermostat_dashboard
  platform: ESP32
  board: featheresp32
 
wifi:
  ssid: "name"
  password: "pass"

api:
  password: "pass"

ota:
  password: "pass"
  
logger:
  baud_rate: 0
  level: VERBOSE

uart:
  id: serial
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600

display:
  - platform: nextion
    id: nextion_display
    uart_id: serial
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/thermostat_dashboard.yaml...
INFO Starting log output from thermostat_dashboard.local using esphome API
INFO Connecting to thermostat_dashboard.local:6053 (192.168.0.47)
INFO Successfully connected to thermostat_dashboard.local
[15:49:04][I][app:105]: ESPHome version 1.20.0 compiled on Jul 24 2021, 15:46:24
[15:49:04][C][wifi:499]: WiFi:
[15:49:04][C][wifi:359]:   SSID: [redacted]
[15:49:04][C][wifi:360]:   IP Address: 192.168.0.47
[15:49:04][C][wifi:362]:   BSSID: [redacted]
[15:49:04][C][wifi:363]:   Hostname: 'thermostat_dashboard'
[15:49:04][C][wifi:367]:   Signal strength: -71 dB ▂▄▆█
[15:49:04][V][wifi:369]:   Priority: 0.0
[15:49:04][C][wifi:371]:   Channel: 11
[15:49:04][C][wifi:372]:   Subnet: 255.255.255.0
[15:49:04][C][wifi:373]:   Gateway: 192.168.0.1
[15:49:04][C][wifi:374]:   DNS1: 194.168.4.100
[15:49:04][C][wifi:375]:   DNS2: 194.168.8.100
[15:49:04][C][uart_esp32:088]: UART Bus:
[15:49:04][C][uart_esp32:090]:   TX Pin: GPIO17
[15:49:04][C][uart_esp32:093]:   RX Pin: GPIO16
[15:49:04][C][uart_esp32:094]:   RX Buffer Size: 256
[15:49:04][C][uart_esp32:096]:   Baud Rate: 9600 baud
[15:49:04][C][uart_esp32:097]:   Data Bits: 8
[15:49:04][C][uart_esp32:098]:   Parity: NONE
[15:49:04][C][uart_esp32:099]:   Stop bits: 1
[15:49:04][C][logger:189]: Logger:
[15:49:04][C][logger:190]:   Level: VERBOSE
[15:49:04][C][logger:191]:   Log Baud Rate: 0
[15:49:04][C][logger:192]:   Hardware UART: UART0
[15:49:04][C][nextion:117]: Nextion:
[15:49:04][C][nextion:118]:   Device Model:     
[15:49:04][C][nextion:119]:   Firmware Version: 
[15:49:04][C][nextion:120]:   Serial Number:    
[15:49:04][C][nextion:121]:   Flash Size:       
[15:49:04][C][nextion:122]:   Wake On Touch:    True
[15:49:04][C][ota:029]: Over-The-Air Updates:
[15:49:04][C][ota:030]:   Address: thermostat_dashboard.local:3232
[15:49:04][C][ota:032]:   Using Password.
[15:49:04][W][nextion:072]: Nextion is not connected! 
[15:49:06][C][api:095]: API Server:
[15:49:06][C][api:096]:   Address: thermostat_dashboard.local:6053
[15:49:06][W][nextion:072]: Nextion is not connected! 
[15:49:06][W][nextion:072]: Nextion is not connected!
```

g-enzo, 2021-07-24:
Thinking this is NOT an esphome issue. I have just found out I can also get this error by disabling a single timer in the Nextion tft.

Gavin

OptimusGREEN, 2021-07-24:
its defo an esphome issue as if you roll back it will work. I'm having the same issue. there is a problem within the nextion component as far as i can tell, doesn't seem to open the appropriate port for connection.

OptimusGREEN, 2021-07-24:
my nextion no longer works using 1.20 either.

g-enzo, 2021-07-24:
Yeah there are some breaking changes for Nextion in 1.20.0. I got them out of the way, now this issue.

Dunno if related but getting this sometimes...

[18:29:00][W][nextion:072]: Nextion is not connected! 
[18:29:00][V][app:081]: A component took a long time in a loop() cycle (1.01 s).
[18:29:00][V][app:082]: Components should block for at most 20-30ms in loop().
[18:29:01][E][uart_esp32:147]: Reading from UART timed out at byte 0!

g-enzo, 2021-07-24:
From the docs...

"bkcmd=0       // Tells the Nextion to not send responses on commands. This is the current default but can be set just in case"

This is important.

Adding this line to the program.s section of  the Nextion tft seems to have solved this issue.

Sorry for wasting  your time.

OptimusGREEN, 2021-07-24:
It says this is default. I'm surprised that made a difference. I actually can't connect to my esp since updating to 1.20

Electronlibre2012, 2021-07-25:
hello @g-enzo and @OptimusGREEN 

first i just remove the old command in esphome that was replaced by "bkcmd=0",

it works after OTA but if i unplug power and plug again, the Wemos D1 mini connect to HA but the nextion 2.4" doesnt refresh anymore.

Same with a Wemos D1 pro and a nextion 3.5", no more refresh after putting back power, just after a new OTA flash...lol

So i test the "baud=115200   and bkcmd=0" and it works nice, after change 
`uart:
  id: uart_2
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200` 

happy to do not have to change everything in my nextion setup...it has take lots of hours of tweaking before and dont want to spend hours again...

Ili9341 is a better, simple way now... ;)

OptimusGREEN, 2021-07-28:
@Electronlibre2012 Hi, what do you mean 'remove the old command in esphome'?

mavrick01, 2021-07-29:
I did some testing and found the slowest I could go was 38400, any slower and it did not like talking to the system. 

Also be wary of power issues, powering the screen from a Wemos Mini 5V Pin was causing havoc with the serial communication (even though the screen works). Putting it on its own 5V circuit took all that frustration way.

Electronlibre2012, 2021-07-29:
> 
> 
> @Electronlibre2012 Hi, what do you mean 'remove the old command in esphome'?

hello @OptimusGREEN 

sorry i dont remeber the command and google it 👍 the old command is `it.set_wait_for_ack(false);` and need to be removed...replace by `bkcmd=0 ` in `program.s` under nextion editor

cheers

Electronlibre2012, 2021-07-29:
> I did some testing and found the slowest I could go was 38400, any slower and it did not like talking to the system.
>
> Also be wary of power issues, powering the screen from a Wemos Mini 5V Pin was [...]

hello @mavrick01 

i powered all my nextion (2.4", 3.5" and 7" from the 5V on the wemos D1, exept for the 7" its an ESP32, otherwise it doesnt work at all) and have any issue...

perhaps you need a better filtered power supply...not the chin... cheap ones , who have very bad filtering quality...just a guess...

cheers

OptimusGREEN, 2021-07-29:
Thanks. That didn't work for me. I'm starting to think it's crashing due to all the new nextion stuff taking up to much memory or something as when I downgrade, all works well.

mavrick01, 2021-07-29:
Hi @Oli,

You were right, I changed out the Wemos and success!  Bad components = inconsistency.

It even solved the issue I was having when my update intervals were below 5s and now runs smoothly at 115200.

Electronlibre2012, 2021-07-30:
nice! happy for you @mavrick01

Giuseppe-P, 2021-08-01:
i have a thermostat based on nextion and wemos mini.  yesterday i upgraded from version 1.19.4 to version 1.20.3 and it stopped working.  the wemos restarts continuously and the nextion from the logs is not connected.  i used the lines in the program.s file, but nothing.

Electronlibre2012, 2021-08-01:
try desactivate all sensor first,

then when the nextion will work again, activate one by one to find the "faulty" one, there is a lot of breaking change with the last EspHome for the Nextion display component.

OptimusGREEN, 2021-08-01:
@Giuseppe-P Yeah thats my exact issue. Its looking like the new nextion stuff is much larger and taking up a lot of the stack, If your config is quite big then this fills it and causes a stack overflow and reboot loop. Increasing the size of the stack appears to be the solution.

There is a PR for this [here](https://github.com/esphome/esphome/pull/1651) but I have unfortunately been unable to get this working on first attempt.

Here is a discussion about increasing on esp32, i'm assuming there is a similar methon for 8266.

@Electronlibre2012 this will work until the sensor that fills the stack, If it was working on 1.19.4 but not 1.20.x then there won't be a faulty sensor.

Giuseppe-P, 2021-08-01:
i think my config is valid because as you said since it works with 1.19.4 i don't see why it shouldn't work with the latest version.  the problem is that the log indicates that the nextion is not connected.  from the example on esphome I see that it uses the uart2 I think through the softwarserial library and not the native TX RX hardware PINs.  since i have wemos mini i cannot change the communication PINs because they are not on this board.  therefore since I use docker on raspbian I have two images of esphome, and then for the nextion I compile with version 1.19.4 and for the other boards like luci etc. I compile with the latest version.  this is my nextion thermostat project. 
https://github.com/Giuseppe-P/Thermostat_Esphome-Home-Assistant_Nextion-3.2-inches_ON-OFF-MANUAL

Giuseppe-P, 2021-08-01:
> @Giuseppe-P Sì, questo è il mio problema esatto. Sembra che la nuova roba di nextion sia molto più grande e occupi molto dello stack, se la tua configurazione è abbastanza grande, questa la riempie e provoca un overflow dello stack e un ciclo di riavvio. L'aumento delle dimensioni dello stack sembra essere la soluzione.
> 
> C'è un PR per questo [qui,](https://github.com/esphome/esphome/pull/1651) ma sfortunatamente non sono stato in grado di farlo funzionare al primo tentativo.
> 
> [Ecco](https://discord.com/channels/429907082951524364/524177279270780928/822457121941225473) una discussione sull'aumento su esp32 (questo ha funzionato per me), presumo che ci sia un metodo simile per 8266.
> 
> @Electronlibre2012 funzionerà fino al sensore che riempie lo stack, se funzionava su 1.19.4 ma non su 1.20.x, non ci sarà un sensore difettoso.

i think my config is valid because as you said since it works with 1.19.4 i don't see why it shouldn't work with the latest version.  the problem is that the log indicates that the nextion is not connected.  from the example on esphome I see that it uses the uart2 I think through the softwarserial library and not the native TX RX hardware PINs.  since i have wemos mini i cannot change the communication PINs because they are not on this board.  therefore since I use docker on raspbian I have two images of esphome, and then for the nextion I compile with version 1.19.4 and for the other boards like luci etc. I compile with the latest version.  this is my nextion thermostat project.  https://github.com/Giuseppe-P/Thermostat_Esphome-Home-Assistant_Nextion-3.2-inches_ON-OFF-MANUAL

Giuseppe-P, 2021-08-01:
I believe that the communication PINs are not the problem anyway.  but surely as you said many things have been added which flooding the chip.  why make these changes when a component works well?  I don't understand why updates instead of improving spoil

closed 2021-10-01 by g-enzo