esphome/issues #2263: Esp32 remote receiver behaving differently than esp8266

sourabhjaiswal, 2021-07-24:
### The problem

Hi,

I am seeing different reading from both ESP32 and ESP8266 while using IR receiver with same configuration on the boards.
My setup is I have used 1 IR sensor and connected it to both ESP32 and ESP8266.
If I press any key on an IR remote I get different results from both ESP32 and ESP8266.

My configs:-

Esp32 Config:-
```
esphome:
  name: remote-test2
  platform: ESP32
  board: esp32doit-devkit-v1

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxxxx"
  manual_ip:
    static_ip: 192.168.0.106
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.201
  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Remote-Test2 Fallback Hotspot"
#    password: "OWbu8IC1yG9A"

captive_portal:
remote_receiver:
  id: rcvr
  pin:
    number: GPIO5
    inverted: True
    mode: INPUT_PULLUP
  dump: [ sony, lg, jvc, nec, raw ]
  # high 55% tolerance is recommended for some remote control units
  tolerance: 40%
```
ESP8266 Config:-
```
esphome:
  name: remote-test
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:
#  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxxxx"
  password: "xxxxxxxx"

  manual_ip:
    static_ip: 192.168.0.104
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.201
  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Remote-Test Fallback Hotspot"
#    password: "R3Mbt7urp7Zp"

captive_portal:

remote_receiver:
  id: rcvr
  pin:
    number: GPIO14
    inverted: True
    mode: INPUT_PULLUP
  dump: [ sony, lg, jvc, nec, raw ]
  # high 55% tolerance is recommended for some remote control units
  tolerance: 40%
``` 

Logs from ESP32 and ESP8266 from same IR sensor:-

ESP32 logs:-

```
INFO Reading configuration /config/esphome/remote-test2.yaml...
INFO Starting log output from 192.168.0.106 using esphome API
INFO Connecting to 192.168.0.106:6053 (192.168.0.106)
INFO Successfully connected to 192.168.0.106
[22:21:48][I][app:105]: ESPHome version 1.20.0 compiled on Jul 24 2021, 22:15:32
[22:21:48][C][wifi:499]: WiFi:
[22:21:48][C][wifi:359]:   SSID: [redacted]
[22:21:48][C][wifi:360]:   IP Address: 192.168.0.106
[22:21:48][C][wifi:362]:   BSSID: [redacted]
[22:21:48][C][wifi:363]:   Hostname: 'remote-test2'
[22:21:48][C][wifi:367]:   Signal strength: -58 dB ▂▄▆█
[22:21:48][C][wifi:371]:   Channel: 11
[22:21:48][C][wifi:372]:   Subnet: 255.255.255.0
[22:21:48][C][wifi:373]:   Gateway: 192.168.0.1
[22:21:48][C][wifi:374]:   DNS1: 192.168.0.201
[22:21:48][C][wifi:375]:   DNS2: 0.0.0.0
[22:21:48][C][logger:189]: Logger:
[22:21:48][C][logger:190]:   Level: DEBUG
[22:21:48][C][logger:191]:   Log Baud Rate: 115200
[22:21:48][C][logger:192]:   Hardware UART: UART0
[22:21:48][C][remote_receiver.esp32:054]: Remote Receiver:
[22:21:48][C][remote_receiver.esp32:055]:   Pin: GPIO5 (Mode: INPUT_PULLUP, INVERTED)
[22:21:48][C][remote_receiver.esp32:060]:   Channel: 0
[22:21:48][C][remote_receiver.esp32:061]:   RMT memory blocks: 3
[22:21:48][C][remote_receiver.esp32:062]:   Clock divider: 80
[22:21:48][C][remote_receiver.esp32:063]:   Tolerance: 40%
[22:21:48][C][remote_receiver.esp32:064]:   Filter out pulses shorter than: 50 us
[22:21:48][C][remote_receiver.esp32:065]:   Signal is done after 10000 us of no changes
[22:21:48][C][captive_portal:148]: Captive Portal:
[22:21:48][C][ota:029]: Over-The-Air Updates:
[22:21:48][C][ota:030]:   Address: 192.168.0.106:3232
[22:21:48][C][ota:032]:   Using Password.
[22:21:48][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[22:21:48][C][api:095]: API Server:
[22:21:48][C][api:096]:   Address: 192.168.0.106:6053
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:21:58][D][remote.raw:041]: Received Raw: 2397, -602, 1187, -596, 592, -601, 1187, -596, 592, -602, 1186, -596, 592, -601, 593, -602, 591, -602, 592, -602, 591, -602, 592, -601, 1186, -597, 1186, -596, 592, -603
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.raw:041]: Received Raw: 2397, -601, 1187, -596, 594, -600, 1187, -595, 593, -601, 1187, -595, 593, -601, 593, -600, 594, -600, 592, -601, 593, -601, 593, -601, 1185, -597, 1186, -596, 593, -601
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
```

ESP8266 Logs:-

```
INFO Reading configuration /config/esphome/remote-test.yaml...
INFO Starting log output from 192.168.0.104 using esphome API
INFO Connecting to 192.168.0.104:6053 (192.168.0.104)
INFO Successfully connected to 192.168.0.104
[22:21:55][I][app:105]: ESPHome version 1.20.0 compiled on Jul 24 2021, 21:42:37
[22:21:55][C][wifi:499]: WiFi:
[22:21:55][C][wifi:359]:   SSID: [redacted]
[22:21:55][C][wifi:360]:   IP Address: 192.168.0.104
[22:21:55][C][wifi:362]:   BSSID: [redacted]
[22:21:55][C][wifi:363]:   Hostname: 'remote-test'
[22:21:55][C][wifi:367]:   Signal strength: -50 dB ▂▄▆█
[22:21:55][C][wifi:371]:   Channel: 11
[22:21:55][C][wifi:372]:   Subnet: 255.255.255.0
[22:21:55][C][wifi:373]:   Gateway: 192.168.0.1
[22:21:55][C][wifi:374]:   DNS1: 192.168.0.201
[22:21:55][C][wifi:375]:   DNS2: (IP unset)
[22:21:55][C][template.switch:058]: Template Switch 'Raw Code Power Button'
[22:21:55][C][template.switch:059]:   Restore State: NO
[22:21:55][C][template.switch:060]:   Optimistic: NO
[22:21:55][C][logger:189]: Logger:
[22:21:55][C][logger:190]:   Level: DEBUG
[22:21:55][C][logger:191]:   Log Baud Rate: 115200
[22:21:55][C][logger:192]:   Hardware UART: UART0
[22:21:55][C][remote_transmitter:018]: Remote Transmitter...
[22:21:55][C][remote_transmitter:019]:   Carrier Duty: 50%
[22:21:55][C][remote_transmitter:020]:   Pin: GPIO16 (Mode: OUTPUT)
[22:21:55][C][remote_receiver.esp8266:059]: Remote Receiver:
[22:21:55][C][remote_receiver.esp8266:060]:   Pin: GPIO14 (Mode: INPUT_PULLUP, INVERTED)
[22:21:55][C][remote_receiver.esp8266:065]:   Buffer Size: 1000
[22:21:55][C][remote_receiver.esp8266:066]:   Tolerance: 40%
[22:21:55][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 50 us
[22:21:55][C][remote_receiver.esp8266:068]:   Signal is done after 10000 us of no changes
[22:21:55][C][captive_portal:148]: Captive Portal:
[22:21:55][C][ota:029]: Over-The-Air Updates:
[22:21:55][C][ota:030]:   Address: 192.168.0.104:8266
[22:21:55][C][ota:032]:   Using Password.
[22:21:55][C][api:095]: API Server:
[22:21:55][C][api:096]:   Address: 192.168.0.104:6053
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:21:58][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[22:22:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
```

Please confirm if any other information is needed.

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

remote_receiver

sourabhjaiswal, 2021-07-26:
Actually from my execution what I have observed that its happening quite frequently.
In fact there are some instances where ESP32 is only reading raw values where as ESP8266 is decoding same key press properly.

ESP8266 key logs:-
```
[19:46:13][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:13][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:14][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:23][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:24][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:24][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:24][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:34][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:34][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:34][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:37][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:37][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:37][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:37][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:38][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:38][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:38][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:38][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:42][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:42][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:42][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:42][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:44][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:44][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:44][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:45][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:45][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:45][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:47][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:47][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:47][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:51][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:51][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:51][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:00][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:00][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:00][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:00][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:00][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:02][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:02][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:47:02][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
```

ESP32 keylogs:-
```
[19:46:13][D][remote.raw:041]: Received Raw: 2491, -485, 1267, -516, 643, -558, 1229, -565, 597, -598, 1213, -571, 617, -577, 617, -576, 618, -576, 617, -576, 618, -577, 617, -576, 1211, -572, 1210, -573, 617, -577
[19:46:13][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:14][D][remote.raw:041]: Received Raw: 2401, -577, 1211, -572, 617, -577, 1210, -573, 615, -578, 1211, -572, 615, -579, 616, -577, 617, -578, 615, -578, 616, -577, 616, -579, 1209, -574, 1209, -573, 615, -578, 3000
[19:46:23][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:24][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:24][D][remote.raw:041]: Received Raw: 2376, -601, 1188, -595, 593, -601, 1187, -595, 594, -600, 1188, -595, 610, -583, 596, -598, 594, -600, 593, -601, 593, -600, 611, -584, 1196, -586, 1187, -596, 593, -600, 5646, -1224
[19:46:24][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:33][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[19:46:34][D][remote.raw:041]: Received Raw: 2488, -488, 1241, -541, 644, -550, 1213, -589, 597, -598, 1189, -595, 594, -600, 593, -601, 594, -600, 592, -601, 593, -602, 593, -600, 1187, -595, 1208, -575, 593, -601
[19:46:34][D][remote.raw:041]: Received Raw: 2376, -602, 1187, -596, 593, -600, 1210, -573, 594, -599, 1188, -595, 594, -600, 593, -601, 592, -602, 592, -601, 593, -601, 592, -601, 1188, -596, 1186, -596, 593, -601
[19:46:34][D][remote.raw:041]: Received Raw: 2376, -601, 1187, -596, 594, -599, 1188, -595, 594, -600, 1187, -595, 603, -591, 593, -601, 616, -578, 610, -583, 594, -600, 616, -578, 1186, -596, 1210, -573, 616, -578, 4941
[19:46:37][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[... cut ...]
[19:47:02][D][remote.raw:041]: Received Raw: 2377, -601, 1187, -596, 593, -600, 1188, -595, 593, -600, 1188, -595, 594, -600, 593, -601, 593, -601, 593, -600, 593, -602, 591, -602, 1186, -596, 1187, -596, 593, -601
[19:47:02][D][remote.raw:041]: Received Raw: 2377, -601, 1187, -596, 593, -601, 1187, -595, 593, -601, 1187, -596, 593, -600, 594, -601, 592, -602, 592, -602, 591, -602, 592, -602, 1185, -597, 1187, -596, 591, -603
```
if you see the iterations on ```19:46:51``` and ```19:47:02``` its observed that ESP32 is only reading raw where as ESP8266 decoded properly.

glmnet, 2021-07-26:
can you disable the sony receiver on the esp8266 **only** and run the same test again *just one key press* so we can see what timings the esp32 is liking and which aren't. And how different the perceived timings are.

IMO this is very chaotic response from the ESP32 vs the ESP8266, so I'm pretty sure something is not working as expected. Unfortunately debugging this requires also setting up an environment where you can also measure the timings with a logic analyzer.

WeekendWarrior1, 2021-07-27:
What happens when you play with the tolerance %?
Or maybe reduce(set) the ``idle:`` to 4ms?

sourabhjaiswal, 2021-07-27:
Please find the logs below as requested:-
```
INFO Reading configuration /config/esphome/remote-test2.yaml...
INFO Starting log output from 192.168.0.106 using esphome API
INFO Connecting to 192.168.0.106:6053 (192.168.0.106)
INFO Successfully connected to 192.168.0.106
[20:56:45][I][app:105]: ESPHome version 1.20.0 compiled on Jul 24 2021, 22:15:32
[20:56:46][C][wifi:499]: WiFi:
[20:56:46][C][wifi:359]:   SSID: [redacted]
[20:56:46][C][wifi:360]:   IP Address: 192.168.0.106
[20:56:46][C][wifi:362]:   BSSID: [redacted]
[20:56:46][C][wifi:363]:   Hostname: 'remote-test2'
[20:56:46][C][wifi:367]:   Signal strength: -42 dB ▂▄▆█
[20:56:46][C][wifi:371]:   Channel: 2
[20:56:46][C][wifi:372]:   Subnet: 255.255.255.0
[20:56:46][C][wifi:373]:   Gateway: 192.168.0.1
[20:56:46][C][wifi:374]:   DNS1: 192.168.0.201
[20:56:46][C][wifi:375]:   DNS2: 0.0.0.0
[20:56:46][C][logger:189]: Logger:
[20:56:46][C][logger:190]:   Level: DEBUG
[20:56:46][C][logger:191]:   Log Baud Rate: 115200
[20:56:46][C][logger:192]:   Hardware UART: UART0
[20:56:46][C][remote_receiver.esp32:054]: Remote Receiver:
[20:56:46][C][remote_receiver.esp32:055]:   Pin: GPIO5 (Mode: INPUT_PULLUP, INVERTED)
[20:56:46][C][remote_receiver.esp32:060]:   Channel: 0
[20:56:46][C][remote_receiver.esp32:061]:   RMT memory blocks: 3
[20:56:46][C][remote_receiver.esp32:062]:   Clock divider: 80
[20:56:46][C][remote_receiver.esp32:063]:   Tolerance: 40%
[20:56:46][C][remote_receiver.esp32:064]:   Filter out pulses shorter than: 50 us
[20:56:46][C][remote_receiver.esp32:065]:   Signal is done after 10000 us of no changes
[20:56:46][C][captive_portal:148]: Captive Portal:
[20:56:46][C][ota:029]: Over-The-Air Updates:
[20:56:46][C][ota:030]:   Address: 192.168.0.106:3232
[20:56:46][C][ota:032]:   Using Password.
[20:56:46][C][api:095]: API Server:
[20:56:46][C][api:096]:   Address: 192.168.0.106:6053
[20:59:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[20:59:06][D][remote.raw:041]: Received Raw: 2398, -601, 1188, -595, 594, -600, 1188, -595, 593, -601, 1187, -595, 594, -600, 593, -601, 593, -601, 593, -601, 592, -601, 594, -600, 1187, -596, 1187, -595, 593, -601
[20:59:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
[20:59:06][D][remote.sony:064]: Received Sony: data=0x0000540C, nbits=15
```
Unfortunately I dont have a Logic analyzer so can help in that aspect.
Any other logs or scenarios from ESP itself I can execute.

vietlehung, 2021-07-31:
Hi, I am facing same problem with sony ir code. As I remember, from ver 1.16.0 (earlier not yet tested) I can't use ESP32 for sony ir remote code. Same setting on esp8266 works without problem but I prefer more GPIO pins of ESP32. I tried to play with some settings like:   increase buffer_size:  up to 50kb, reduce idle: to 4ms ,... but can't help.
With default setting, and a fresh start of ESP32, it can recognize 1 code (1st press) correctly only then only raw code received. There are things go wrong inside with 1st process of sony code.

closed 2021-12-15 by github-actions[bot]