esphome/issues #2264: Climate PID appears to be broken, worked in 1.18

MattW2, 2021-07-25:
### The problem

I use ESPHome running a PID loop to control the speed of several fans. The PID loop runs in cool only mode. This code had previously worked in an older version of ESPHome but now causes the ESP8266 to bootloop 10x times before ending in safe mode. Removing the climate/pid section of the YAML fixes the boot loop but leaves me without a controller. 

I notice that when I compile using the climate/pid in the YAML I get the below warning in the compiler output. I haven't seen this before on older versions. I don't use the parameter "away" in my configuration. 

```
Compiling /data/server_rack/.pioenvs/server_rack/src/esphome/components/climate/climate.cpp.o
src/esphome/components/climate/climate.cpp: In constructor 'esphome::climate::Climate::Climate(const string&)':
src/esphome/components/climate/climate.cpp:431:58: warning: 'esphome::climate::Climate::away' is deprecated (declared at src/esphome/components/climate/climate.h:196): away is deprecated, use preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : Nameable(name) {}
                                                          ^
```

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1_mini_lite

### Component causing the issue

climate/pid

### Example YAML snippet

```yaml
esphome:
  name: server_rack
  platform: ESP8266
  board: d1_mini_lite

sensor:
  - platform: bme280
    temperature:
      name: "Server Rack Temperature"
      id: server_rack_temperature
      oversampling: 16x

climate:
  - platform: pid
    id: server_rack_pid_temperature_controller
    name: "Server Rack PID Temperature Controller"
    sensor: server_rack_temperature
    default_target_temperature: 32 °C
    cool_output: fan_pwm_output_d6
    control_parameters:
      kp: 0.45
      ki: 0.0001
      kd: 50.0
    visual:
      min_temperature: 20 °C
      max_temperature: 40 °C
      temperature_step: 1.0 °C
```

### Anything in the logs that might be useful for us?

```txt
[22:12:40]User exception (panic/abort/assert)
[22:12:40]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[22:12:40]
[22:12:40]Panic pid_climate.cpp:107 void esphome::pid::PIDClimate::handle_non_auto_mode_(): Assertion 'false' failed.
[22:12:40]
[22:12:40]>>>stack>>>
[22:12:40]
[22:12:40]ctx: cont
[22:12:40]sp: 3ffffd60 end: 3fffffc0 offset: 0000
[22:12:40]3ffffd60:  00000000 3ffffd5c 3ffffd5c 00000014  
[22:12:40]3ffffd70:  000000fe 00000000 00000000 00000000  
[22:12:40]3ffffd80:  00000000 00000000 00000000 4025fa9c  
[22:12:40]3ffffd90:  3fff115c 00000053 3ffffd8c 00000014  
[22:12:40]3ffffda0:  3fff2e34 3ffffe00 3fff2dc4 4022250e  
[22:12:40]3ffffdb0:  3ffffdd0 3ffffdc0 00000010 40222542  
[22:12:40]3ffffdc0:  3f800000 00000000 00000000 4020c663  
[22:12:40]3ffffdd0:  00000000 40400000 41a00000 4020c6b6  
[22:12:40]3ffffde0:  3fff2e34 3ffffe00 3ffe9fca 40209100  
[22:12:40]3ffffdf0:  00000002 3fff2864 3fff2dc4 4020c49a  
[22:12:40]3ffffe00:  3fff2dc4 00000201 00000001 42000000  
[22:12:40]3ffffe10:  00000000 3ffffe50 3fff0c00 00000000  
[22:12:40]3ffffe20:  00000000 3fff2900 3fff13e0 40212700  
[22:12:40]3ffffe30:  00000000 3fff13e0 00000201 00000000  
[22:12:40]3ffffe40:  00420000 3f000000 3fff2434 402128ea  
[22:12:40]3ffffe50:  00000000 00000005 3fff0c98 4020cd24  
[22:12:40]3ffffe60:  00000000 00000000 3fff0c98 40224f84  
[22:12:40]3ffffe70:  3fff2ed4 00000004 3fff0c98 40225005  
[22:12:40]3ffffe80:  3fff2b0c 00000004 3fff0c98 40211635  
[22:12:40]3ffffe90:  3fff2ed4 0000000b 3fff2ee0 3fff13e0  
[22:12:40]3ffffea0:  3f800000 3fff0d24 3fff2dc4 4022e9d6  
[22:12:40]3ffffeb0:  3fffff7c 3fff0d24 3fff2dc4 3fff0d20  
[22:12:40]3ffffec0:  3f800000 3fff0d24 3fff0d3c 4021380e  
[22:12:40]3ffffed0:  3fff26a8 00000000 00000000 3fff26d8  
[22:12:40]3ffffee0:  00000000 00000000 00000000 402269c8  
[22:12:40]3ffffef0:  00000000 402269c8 00000000 402269c8  
[22:12:40]3fffff00:  00000000 402269c8 00000000 402269c8  
[22:12:40]3fffff10:  00000000 00000000 3fff2618 00000000  
[22:12:40]3fffff20:  00000000 3fff2638 00000000 00000000  
[22:12:40]3fffff30:  00000000 402269c8 00000000 402269c8  
[22:12:40]3fffff40:  00000000 402269c8 00000000 402269c8  
[22:12:40]3fffff50:  00000000 402269c8 00000000 00000000  
[22:12:40]3fffff60:  3fff2dc4 3fff2ccc 3fff2bc4 3fff2a74  
[22:12:40]3fffff70:  3fff290c 3fff2864 3fff2ea8 feefeffe  
[22:12:40]3fffff80:  3fff2dc4 3fff0c98 00000000 feefeffe  
[22:12:40]3fffff90:  feefeffe feefeffe feefeffe 3fff111c  
[22:12:40]3fffffa0:  3fffdad0 00000000 3fff10dc 402220c8  
[22:12:40]3fffffb0:  feefeffe feefeffe 3ffe84f8 40100a21  
[22:12:40]<<<stack<<<
```

OttoWinter, 2021-07-26:
Hmm, not sure what that could be. The PID climate did work for me here in my testing of the release.

ptr727, 2021-07-27:
```
void PIDClimate::handle_non_auto_mode_() {
  // in non-auto mode, switch directly to appropriate action
  //  - OFF mode -> Output at 0%
  if (this->mode == climate::CLIMATE_MODE_OFF) {
    this->write_output_(0.0);
  } else {
    assert(false);
  }
}
```

My guess; mode was restored from RTC, mode meanings changed due to many climate mode changes, numeric value of restored mode was no longer a valid mode?

OttoWinter, 2021-07-28:
True, that makes perfect sense. Thanks!

closed 2021-07-28 by OttoWinter