esphome/issues #2268: logs won't display

finity69x2, 2021-07-26:
### The problem

I just updated from v1.16.x.dev version to v1.20.0. Now when I click on the "logs" button nothing happens. 

as far as I know all of the other buttons seem to work ok. It's just the "logs" button is non-functional.

I tried two different browsers with the same result.

### What is version of ESPHome has the issue?

v1.20.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

all

### Anything in the logs that might be useful for us?

```txt
nothing in the ESPHome Docker logs
```

vazquezjm, 2021-07-26:
This happened to me as well on the mobile app. I noticed that since the last few versions (not sure exactly which one), when clicking the logs button, it shows a modal window to confirm if you want to connect wirelessly, when that's the only option and it fails to show on mobile. This does not happen on the web version (at least for me). 

Using Win10 and Chrome

[image]

finity69x2, 2021-07-26:
that seems related but my issue is different. I don't get that pop-up window.

The devices are showing as "online" and connected and otherwise work properly.

I just can't open any logs at all with any device.

That's going to be a problem when I have to re-calibrate my scale that I built.

pmannk, 2021-07-27:
Interesting, I have the exact same problem after updating to 1.20.0. I'm running within HA (2021.7.4).
Can you try opening it in an incognito browser window? I noticed that works around the problem for me.

I tried clearing the site related cookies/cache however that hasn't helped and I haven't had any more time to tinker with it.

finity69x2, 2021-07-27:
I just tried it using an incognito window and I get a different result.

it does pop up with the same window as  @vazquezjm got above and when I click the button it actually opens the logs. Even tho it it shows online prior to that:

[image]

But if I try to use it in a HA iframe or in a standard  standalone browser (non-incognito) tab the button does absolutely nothing.

[image]

pmannk, 2021-07-27:
Yep, same behaviour I'm seeing - in the HA iframe or standalone in Chrome it won't work.
It also works fine in Edge (chromium) as an iframe.

CoyoteSequoia, 2021-08-18:
I'm having the same problem with the same conditions listed above but on a mac.  It works okay in Safari though.

omegacore, 2021-08-21:
Also happening for me, just upgraded today to 2021.8.0.

 Add-on version: 2021.8.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 5.13  (armv7 / raspberrypi4)
 Home Assistant Core: 2021.6.6
 Home Assistant Supervisor: 2021.06.8

closed 2022-01-03 by github-actions[bot]