esphome/issues #2269: DHT22 value read error

vazquezjm, 2021-07-26:
### The problem

I'm getting this warning in the logs for a DHT22 sensor:

```
[12:24:08][W][dht:167]: Falling edge for bit 40 failed!
[12:24:13][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

This is sporadic, but once it starts happening I need to reboot the Wemos D1 Mini, and eventually, it'll work again for a period of time, then, again with the same warning and no readings.

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

dht

### Example YAML snippet

```yaml
- platform: dht
    pin: 14
    model: DHT22
    temperature:
      name: "Temperatura huerta"
      accuracy_decimals: 1
      filters:
        - offset: 0.5
        - median:
            window_size: 5
            send_every: 3
            send_first_at: 1
    humidity:
      name: "Humedad huerta"
      accuracy_decimals: 0
      filters:
        - offset: 3.0
```

### Anything in the logs that might be useful for us?

```txt
[12:24:08][W][dht:167]: Falling edge for bit 40 failed!
[12:24:13][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

IWIK-Apps, 2021-09-12:
Any update on this? I also see this error

vazquezjm, 2021-09-12:
I replaced de pull-up resistor with a 2k2 (4k7 also works) and powered the sensor with 5v instead of 3v3. I don't get this error anymore.

closed 2021-09-12 by vazquezjm