esphome/issues #2271: Cover/daikin not syncing state when "forwarding" data from remote_receiver to remote_transmitter

muxa, 2021-07-27:
### The problem

I'm integrating ESP8266 into  Daikin Air Conditioner by adding ESP8266 module as a "man-in-the-middle". I've disconnected the original IR receiver from the AC, connected it to the ESP as input and connected the output from the ESP to the AC. All IR signals are "forwarded" to the AC via ESP,  and the `cover` state is kept in sync based on the IR commands. 

I've connected everything and when I control the `cover` from Home Assistant, it sends correct signals to the AC and controls it. 

However when use the IR remote, it never actually "forwards" the commands to the AC, and the state of the `cover` is not always changing (it worked a few times, and then stopped reacting).

When I remove the "forwarding" code, the `cover` starts to react to IR remote commands.

### What is version of ESPHome has the issue?

1.20.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

climate/daikin

### Example YAML snippet

```yaml
substitutions:
  device_name: Lab D1 Mini
  
esphome:
  name: lab_d1_mini
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

logger:
  level: DEBUG

api:
  password: !secret api_password

ota:
  password: !secret ota_password

remote_receiver:
  id: ir_receiver
  pin: 
    number: D7
    inverted: True
    # mode: INPUT_PULLUP
  #dump: panasonic
  #dump: all
  # high 55% tolerance is recommended for some remote control units
  tolerance: 55%
  # the below stops cover responding to IR remote
  # and does not actually have any effect on the AC
  on_raw:
    - remote_transmitter.transmit_raw:
        code: !lambda 'return x;'
        repeat:
          times: 5

remote_transmitter:
  id: ir_transmitter
  pin: 
    number: D8
    inverted: True
  carrier_duty_percent: 100%

climate:
  - platform: daikin
    name: "Daikin AC"
    receiver_id: ir_receiver
    transmitter_id: ir_transmitter
```

### Additional information

I've also tried sending `panasonic` commands (this is what `daikin` uses) to no avail:

```yaml
  on_panasonic:
    - remote_transmitter.transmit_panasonic:
        address: !lambda 'return x.address;'
        command: !lambda 'return x.command;'
        repeat:
          times: 5
```

When I've tested the circuit on it's own (with IR module attached to the ESP) it did seem to "forward" the IR signal (I've used an oscilloscope to observe the output pin)

WeekendWarrior1, 2021-07-27:
Hi muxa,

I also am using an ESP as a MITM controller for my air conditioner (not coolix though), and found the same problem.

If you're willing to build esphome from source, at:
https://github.com/esphome/esphome/blob/b0a38914987b4df5c45901c6b29c844046d96c8b/esphome/components/coolix/coolix.cpp#L226
```c++
  this->publish_state();

  return true;
}
```
change to this:
```c++
  this->publish_state();
  this->transmit_state();

  return true;
}
```
Since I'm not the only one using an ESP in this fashion, I guess I should create a merge request with a configurable option to allow this.

muxa, 2021-07-28:
I've updated description replacing `cover` with `climate` (I was also working on a cover integration, and when submitting this issue late at night mixed up the terms)

muxa, 2021-07-28:
@WeekendWarrior1 thanks for your input! I agree that ability to "forward" data from receiver to transmitter by the climate control might be a good addition to ESPHome. 

Since there are many different climate integrations, it would make sense to add this option to the base climate control. 

But since there are other platforms that have support for IR, would it be worth having the forwarding functionality outside of those platforms, and close to the actual transmitter and receiver components? I don't know ESPHome inner workings well enough, but seems like it should be possible. Something like this perhaps?

```yaml
remote_receiver:
  id: ir_receiver
  pin: D7
  forward_to: ir_transmitter # this is the new options

remote_transmitter:
  id: ir_transmitter
  pin: D8
```

WeekendWarrior1, 2021-07-29:
When I was answering your question I also got mixed up, though you were using ``coolix`` instead of ``daikan`` for some reason

Slapped something together here: https://github.com/WeekendWarrior1/esphome/tree/ir_forward_packet
I used your "forward_to" syntax so you can literally copy+paste your example yaml.

I have not tested it in the slightest yet, probably will tomorrow. I am a little unsure if it will *just work*, mainly concerning carrier frequency.

muxa, 2021-07-29:
@WeekendWarrior1 this is awesome, thank you! 

I have been testing a [different approach](https://github.com/esphome/esphome/discussions/2078) today, with the receiver and transmitter connected to the same pin, and it works! I like that approach since it will let me control the AC with the remote even if ESP crashes or not working properly.

muxa, 2021-08-02:
I've been using the ["1 pin for both receiver and transmitter" approach](https://github.com/esphome/esphome/discussions/2078) for a few days now, and it's working with no problems. So don't need to "forward" the data in code now.

closed 2021-08-02 by muxa

WeekendWarrior1, 2021-08-03:
@jesserockz @OttoWinter 

Do you think this functionality would be useful in esphome?
If so I can clean it up, properly test it and submit a merge request.