esphome/issues #2273: Invalid preamble for SM300D2 response after ~5 mins

Plawasan, 2021-07-28:
### The problem

I have a brand new SM300D2 7-in-1 sensor that I was hoping to use with ESP32:

- Initial connection looks OK, I get the correct response from the sensor:

[W][sm300d2:039]: Successfully read SM300D2 data
[D][sm300d2:049]: Received CO₂: 412 ppm
[D][sensor:127]: 'Air Quality CO₂': Sending state 412.00000 ppm with 0 decimals of accuracy
[D][sm300d2:053]: Received Formaldehyde: 1 µg/m³
[D][sensor:127]: 'Air Quality Formaldehyde': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:057]: Received TVOC: 8 µg/m³
[D][sm300d2:057]: Received TVOC: 8 µg/m³
[D][sensor:127]: 'Air Quality TVOC': Sending state 8.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:061]: Received PM2.5: 39 µg/m³
[D][sensor:127]: 'Air Quality PM2.5': Sending state 39.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:065]: Received pm_10_0: 70 µg/m³
[D][sensor:127]: 'Air Quality PM10': Sending state 70.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:069]: Received Temperature: 28.90 °C
[D][sensor:127]: 'Air Quality Temperature': Sending state 28.90000 °C with 1 decimals of accuracy
[D][sm300d2:073]: Received Humidity: 57.10 percent
[D][sensor:127]: 'Air Quality Humidity': Sending state 57.10000 % with 0 decimals of accuracy

However typically only after a couple minutes the sensor stops responding either completely or starts sending stale data and then dies.

[W][sm300d2:024]: Invalid preamble for SM300D2 response!

What I have already tried
- 3 different ESP32
- powering the board either via USB or via 5V power source
- disabling logger in case there were any UART conflicts
- for what it's worth there's a blue LED blinking at ~1.5Hz on the sensor

Nothing helped.. I cannot obviously tell if this is a faulty sensor (came from Aliexpress..) however I'm raising this more to see if anyone had a similar experience (and hopefully a fix) or if the issue can possibly be on the ESPHome side.

Thanks!

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

SM300D2

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO25
  baud_rate: 9600

sensor:
  - platform: sm300d2
    id: sm300d2id
    co2:
      name: "Air Quality CO₂"
    formaldehyde:
      name: "Air Quality Formaldehyde"
    tvoc:
      name: "Air Quality TVOC"
    pm_2_5:
      name: "Air Quality PM2.5"
    pm_10_0:
      name: "Air Quality PM10"
    temperature:
      name: "Air Quality Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Air Quality Humidity"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[W][sm300d2:024]: Invalid preamble for SM300D2 response!
```

Plawasan, 2021-07-29:
I made a quick update to the source to get a bit more info about what is happening:

 if (response[0] != 0x3C || response[1] != 0x02) {
    ESP_LOGW(TAG, "Invalid preamble for SM300D2 response: Ox%02X - 0x%02X", response[0], response[1]);

plus the same in the message reporting the successful read, here's what I get now:

[W][sm300d2:039]: Successfully read SM300D2 data: Ox3C - 0x02
[D][sm300d2:049]: Received CO₂: 0 ppm
[D][sensor:127]: 'Air Quality CO₂': Sending state 0.00000 ppm with 0 decimals of accuracy
[D][sm300d2:053]: Received Formaldehyde: 0 µg/m³
[D][sensor:127]: 'Air Quality Formaldehyde': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:057]: Received TVOC: 0 µg/m³
[D][sensor:127]: 'Air Quality TVOC': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:061]: Received PM2.5: 27 µg/m³
[D][sensor:127]: 'Air Quality PM2.5': Sending state 27.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:065]: Received pm_10_0: 48 µg/m³
[D][sensor:127]: 'Air Quality PM10': Sending state 48.00000 µg/m³ with 0 decimals of accuracy
[D][sm300d2:069]: Received Temperature: 27.20 °C
[D][sensor:127]: 'Air Quality Temperature': Sending state 27.20000 °C with 1 decimals of accuracy
[D][sensor:127]: 'Air Quality Temperature': Sending state 27.20000 °C with 1 decimals of accuracy
[D][sm300d2:073]: Received Humidity: 56.10 percent
[D][sensor:127]: 'Air Quality Humidity': Sending state 56.10000 % with 0 decimals of accuracy
[I][ota:046]: Boot seems successful, resetting boot loop counter.

And then when it stops working:

[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox3C - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxDD - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxD0 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxCC - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxD4 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxE7 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: OxCD - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox6E - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox6F - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox73 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox66 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox7C - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox6D - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox60 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox62 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox62 - 0x3C
[W][sm300d2:024]: Invalid preamble for SM300D2 response: Ox68 - 0x3C

I can't really draw any conclusions from it, just adding more data...

wuast94, 2021-08-01:
i have the same issue, anything we can do? provide more information for example ?

Plawasan, 2021-08-02:
I'm increasingly convinced this is a ESPHome issue rather than a sensor issue, consistent with the report above (2 faulty sensors are still a possibility..). 

I've added a restart switch to my yaml so that I can restart the ESP independently of the sensor itself:

switch:
  - platform: restart
    id: restart_switch
    name: "ESP Test Restart"

So when the communication dies:
[16:27:08][W][sm300d2:024]: Invalid preamble for SM300D2 response!

I can restart the ESP from HA:
[16:27:16][D][switch:021]: 'ESP Test Restart' Turning ON.
[16:27:16][D][switch:045]: 'ESP Test Restart': Sending state OFF
[16:27:16][I][restart:015]: Restarting device...
[16:27:16][I][app:145]: Rebooting safely...
INFO Connecting to 192.168.1.103:6053 (192.168.1.103)
INFO Successfully connected to 192.168.1.103

and then it works again:
[16:28:17][W][sm300d2:039]: Successfully read SM300D2 data

The sensor is powered from a breadboard, parallel to the ESP so restarting the ESP (or even physically cutting power to it) does NOT restart the sensor (at least not completely).

In the end I implemented an automatic restart of the ESP, not even that is bulletproof however the communication does come back on the next reboot..

interval:
  - interval: 65s
    then:
      - component.update: sm300d2id
      - delay: 60s
      - switch.turn_on: restart_switch

**[16:50:39][W][sm300d2:039]: Successfully read SM300D2 data**

WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer
INFO Successfully connected to 192.168.1.103
**[16:51:17][W][sm300d2:024]: Invalid preamble for SM300D2 response!**
[16:51:41][D][switch:021]: 'ESP Test Restart' Turning ON.
[16:51:41][D][switch:045]: 'ESP Test Restart': Sending state OFF
[16:51:41][I][restart:015]: Restarting device...
**[16:51:41][I][app:145]: Rebooting safely...**

INFO Successfully connected to 192.168.1.103
**[16:52:18][W][sm300d2:039]: Successfully read SM300D2 data**

I've had the ESP running like this for several hours and I'm still getting updated values - not something I've been able to achieve before without the resets but still far from ideal (e.g. I cannot use moving averages to smooth out the values, the entity history graph has gaps etc.)

@wuast94 - can you test the same? Full config is here - https://gist.github.com/Plawasan/225a95506cf537a180f32ad2e46ac10d

wuast94, 2021-08-03:
test is running and i will update later if it works, but im pretty sure that it will work :) 
I have 3 of this sensors (one broken by myself) and all of them have the same problem.

So i guess its as you sad not the sensors and the esp/esphome is the problem here :/ on one sensor im using a esp32 and o the other a wemos d1 mini. 

i would guess that there is a timing issue or so

wuast94, 2021-08-03:
The workaround works for me too but that's just a dirty workaround :/ 

The device is not available every time the reboot happened (quite often)

Plawasan, 2021-08-03:
Right, I don't see this being a solution but rather just a troubleshooting data point to see where the issue is...

Now we can only hope somebody smarter than us looks into it :)

jamesbraid, 2021-08-15:
Came across the same problem, the sensor component was just doing a 17 byte read and hoping the read aligned with the sensor output. Wrote a PR to fix this, please test it. Makes the SM300D2 perfectly reliable in my testing (24h @ 10 second interval stress testing so far)

Plawasan, 2021-08-22:
I only got home today after seeing this update, I see you've made some changes to make the fix even cleaner - what's the best way for me to test this and help confirm it's working as expected?

Thanks!

Plawasan, 2021-08-23:
I've managed to update my local src files with your changes, for the last 16h I'm getting consistent data from the sensor.. great! I've only had an issue that after the first restart the CO2 ppm value was zero, it corrected itself on the next reboot.

Looking very promising so far!

Plawasan, 2021-08-28:
I've had the sensor running with the updated code for almost a week now, it does return data but not still 100%, I get good data maybe on 50% of the sensor reads.
Here's what the logs look like:

[07:55:21][W][sm300d2:034]: SM300D2 Checksum doesn't match: 0x00!=0xFB
[07:56:21][W][sm300d2:034]: SM300D2 Checksum doesn't match: 0x31!=0x61
[07:57:21][W][sm300d2:026]: Invalid preamble for SM300D2 response!
[07:58:21][W][sm300d2:041]: Successfully read SM300D2 data
[07:58:21][D][sm300d2:051]: Received CO₂: 862 ppm
...
[07:59:22][W][sm300d2:026]: Invalid preamble for SM300D2 response!
[08:00:22][W][sm300d2:041]: Successfully read SM300D2 data
[08:00:22][D][sm300d2:051]: Received CO₂: 841 ppm
...
[08:01:22][W][sm300d2:041]: Successfully read SM300D2 data
[08:01:22][D][sm300d2:051]: Received CO₂: 863 ppm
[08:02:22][W][sm300d2:034]: SM300D2 Checksum doesn't match: 0x32!=0x77

I'm also seeing weird data showing up in HA but I suspect this is not a sensor issue as the spike in the graph (and the underlying data) is too sharp given that I have a moving average set for all values (window_size=10, update every 60s) so even if the sensor did return complete bogus values the graph should go up gradually...

[image]

DB data:
2021-08-27 21:28:23.101096|7
2021-08-27 21:16:23.105131|8
2021-08-27 21:04:23.118459|1544
2021-08-27 20:57:23.133008|1543
2021-08-27 20:51:23.126931|1542
2021-08-27 20:41:23.143723|7
2021-08-27 20:34:23.135622|8

yaml:
```
    pm_2_5:
      name: "Air Quality PM2.5"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_first_at: 3
            send_every: 3     
```

lux4rd0, 2021-09-02:
This was my first ESP32 and HA project and I've run into the exact situation. Happy to help test along the way as well.

lux4rd0, 2021-09-08:
I grabbed the files from the PR and have been running it for the last hour @ 10 second pulls. So far it's looking good. Will keep an eye on it for the rest of the day and see if I'm missing any polling entries.

[image]

Plawasan, 2021-09-09:
I made a mistake of updating esphome and then reinstalling the firmware of the node that has the sm300d2 connected... is there a more official way of merging the changes into my own build other than just copying the contents of the updated files to my local src?

closed 2021-09-13 by OttoWinter

bakerkj, 2021-10-05:
Thanks for this change!

Back in 1.27.2 and 1.18.0 I was able to read my sm300d2 air quality sensor. However when I updated to 1.19.4 it was no longer able to read the sm300d2 when I had blue tooth enabled to read a xiaomi plant sensor). At the time it returned `Invalid preamble for SM300D2 response!`. If turned the bluetooth configuration off it worked fine.

@jesserockz suggested I try setting `esphome: arduino_version: 1.0.4` this worked and I have been running in that configuration since July.

However I recently saw this change and wanted to give it a try under 2021.9.2. I pulled in this patch: https://github.com/esphome/esphome/commit/c6109024aaff83c9a44ccd579f5994fafe4a3078

I gave it a try and it appears to have resolved this issue.

Thanks!