esphome/issues #2275: Nextion Docs: Breaking Changes

OptimusGREEN, 2021-07-28:
### The problem

My nextion device no longer works with 1.20.0.

I believe not all breaking changes were listed in the changelog. I have posted another [issue](https://github.com/esphome/issues/issues/2254) a few days ago with logs etc but on further investigation it seems to boot loop.

It would be great if any other breaking changes (as I know it was a huge nextion change) were documented.

This issue isn't a duplicate of my issue above, this issue is purely that all breaking changes haven't been listed in changelog etc...

### What is version of ESPHome has the issue?

1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

Nextion

OptimusGREEN, 2021-07-31:
It appears the new nextion additions are using up much more of the stack than previously and an already large config will crash when going from < 1.20.0 to 1.20.x

This should maybe be mentioned in the docs somewhere? or if possible an option to increase stack size?

Susan-Beijing-DWIN, 2021-08-02:
Hello manager, I'm Susan from Beijing DWIN technology, nice to see you, we are a Chinese manufacturer produce smart touch lcd screen and industrial hmi with our self-developed chip and open source software,  the display is based on UART Serial port communication. TTL/RS232/RS485 is available,
please refer to this link for more information about DWIN https://www.linkedin.com/pulse/faq-susan-van-/?trackingId=Sa8S8%2Bn6RE%2B9K%2B2V%2BuBR%2BQ%3D%3D

closed 2021-12-17 by github-actions[bot]