esphome/issues #2279: MPU6050 component error "Communication with MPU6050 failed!"

CarlosGS, 2021-07-29:
### The problem

There is a bug in the MPU6050 component that prevents using _some_ boards (see https://community.home-assistant.io/t/mpu6050-communication-with-mpu6050-failed/268114 and https://community.home-assistant.io/t/communication-with-mpu6050-failed/270449).

However, using the following snippet works OK for the problematic boards:

customGyro.h
-
```C
#include "esphome.h"

class MPU6050 : public PollingComponent {

public:

Sensor *temperature = new Sensor();
Sensor *gyroscope = new Sensor();
Sensor *accelerometer = new Sensor();

float accelgyro_temperature = 0, accelgyro_accel = 0, accelgyro_gyro = 0, accelgyro_accel_max = 0, accelgyro_gyro_max = 0;
const uint8_t MPU_addr = 0x68; // I2C address of the MPU-6050

MPU6050() : PollingComponent(500) { }

void setup() override {
  Wire.begin();

  // default at power-up:
  //    Gyro at 250 degrees second
  //    Acceleration at 2g
  //    Clock source at internal 8MHz
  Wire.beginTransmission(MPU_addr);
  Wire.write(0x6B);  // PWR_MGMT_1 register
  Wire.write(0);     // set to zero (wakes up the MPU-6050)
  Wire.endTransmission(true);
}

void update() override {

// This is the actual sensor reading logic.

  int16_t AcX, AcY, AcZ, Tmp, GyX, GyY, GyZ;
  Wire.beginTransmission(MPU_addr);
  Wire.write(0x3B);  // starting with register 0x3B (ACCEL_XOUT_H)
  Wire.endTransmission(false);
  Wire.requestFrom(MPU_addr, 14, true); // request a total of 14 registers
  AcX = Wire.read() << 8 | Wire.read(); // 0x3B (ACCEL_XOUT_H) & 0x3C (ACCEL_XOUT_L)
  AcY = Wire.read() << 8 | Wire.read(); // 0x3D (ACCEL_YOUT_H) & 0x3E (ACCEL_YOUT_L)
  AcZ = Wire.read() << 8 | Wire.read(); // 0x3F (ACCEL_ZOUT_H) & 0x40 (ACCEL_ZOUT_L)
  Tmp = Wire.read() << 8 | Wire.read(); // 0x41 (TEMP_OUT_H) & 0x42 (TEMP_OUT_L)
  GyX = Wire.read() << 8 | Wire.read(); // 0x43 (GYRO_XOUT_H) & 0x44 (GYRO_XOUT_L)
  GyY = Wire.read() << 8 | Wire.read(); // 0x45 (GYRO_YOUT_H) & 0x46 (GYRO_YOUT_L)
  GyZ = Wire.read() << 8 | Wire.read(); // 0x47 (GYRO_ZOUT_H) & 0x48 (GYRO_ZOUT_L)
  
  accelgyro_accel = (sqrt((float)AcX * AcX + (float)AcY * AcY + (float)AcZ * AcZ) / 32768.) * 2.; // [G]
  accelgyro_gyro = (sqrt((float)GyX * GyX + (float)GyY * GyY + (float)GyZ * GyZ) / 32768.) * 250.; // [deg/s]
  accelgyro_temperature = Tmp / 340. + 36.53; //equation for temperature in degrees C from datasheet

  temperature->publish_state(accelgyro_temperature);
  gyroscope->publish_state(accelgyro_gyro);
  accelerometer->publish_state(accelgyro_accel);

}

};
```

YAML
-
```
esphome:
  includes:
    - customGyro.h

sensor:
  - platform: custom
    lambda: |-
      auto mlx = new MPU6050();
      App.register_component(mlx);
      return {mlx->temperature, mlx->gyroscope, mlx->accelerometer};
    sensors:
      - name: "IMU temperature"
        unit_of_measurement: °C
        accuracy_decimals: 2
      - name: "IMU gyroscope"
        unit_of_measurement: "deg/s"
        accuracy_decimals: 4
      - name: "IMU accelerometer"
        unit_of_measurement: "G"
        accuracy_decimals: 4
```

### What is version of ESPHome has the issue?

v1.20.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

mpu6050

### Additional information

The affected component is https://github.com/esphome/esphome/blob/dev/esphome/components/mpu6050/mpu6050.cpp

I haven't been able to find the bug, posting in case someone has the time to research in more detail :)

CarlosGS, 2021-07-29:
The affected component is https://github.com/esphome/esphome/blob/dev/esphome/components/mpu6050/mpu6050.cpp

I haven't been able to find the bug, posting in case someone has the time to research in more detail :angel:

oxan, 2021-07-29:
It seems [this](https://community.home-assistant.io/t/mpu6050-communication-with-mpu6050-failed/268114) failing device returns `0x0E` in the identity register instead of the expected `0x68`. If you comment out line 26, does it work as expected?

CarlosGS, 2021-07-29:
Haven't been able to try it, but it's probably that. Well spotted! :D
Maybe this is related to the I2C bypass mode? The failing devices might have it enabled by default.

OttoWinter, 2021-08-01:
> Maybe this is related to the I2C bypass mode? The failing devices might have it enabled by default.

More likely this is caused by this being another chip (or knock off). Often cheap knock-off chips will be sold online under the original name, but these registers are changed. If you can confirm it works by uncommenting those lines we can make `0x0E` another valid value for that check.

d-rez, 2022-09-05:
Hey folks, would there be any progress/consensus on how this should be handled? It's been over a year and I've just tried connecting an GY-521 module with a MPU6050 on it, and it's unable to get initialised.

Commenting out the `who_am_i` checks section in `MPU6050Component::setup()` in source makes it work  just fine so I'm assuming mine is also a clone.

Maybe some kind of a config toggle to ignore this particular check?

CarlosGS, 2022-09-05:
Sorry, forgot about this! Please enable logger `level: VERY_VERBOSE` and see if you also get `Requesting 1 bytes from 0x68: / Received 0b00001110 (0x0E)` as Oxan pointed out in https://github.com/esphome/issues/issues/2279#issuecomment-889356861. If you also see 0x0E then we can make it valid as Otto proposed :)

d-rez, 2022-09-05:
I have a feeling the sensor either just returns garbage at 0x75 or it's using it for something else. From multiple reboots it appears that the value returned keeps changing:

```
[01:27:46][C][mpu6050:024]: Setting up MPU6050...
[01:27:46][VV][i2c.arduino:126]: 0x68 TX 75
[01:27:46][VV][i2c.arduino:102]: 0x68 RX A9
[01:27:46][E][component:112]: Component mpu6050.sensor was marked as failed.
```

```
[01:27:41][C][mpu6050:024]: Setting up MPU6050...
[01:27:41][VV][i2c.arduino:126]: 0x68 TX 75
[01:27:41][VV][i2c.arduino:102]: 0x68 RX 85
[01:27:41][E][component:112]: Component mpu6050.sensor was marked as failed.
```

```
[01:31:06][C][mpu6050:024]: Setting up MPU6050...
[01:31:06][VV][i2c.arduino:126]: 0x68 TX 75
[01:31:06][VV][i2c.arduino:102]: 0x68 RX 98
[01:31:06][E][component:112]: Component mpu6050.sensor was marked as failed.
```

CarlosGS, 2022-09-06:
Thank you for looking into it! I think the 0x0E from https://community.home-assistant.io/t/mpu6050-communication-with-mpu6050-failed/268114 might have been noise, like it has happened to you in the first tests.

I've made a PR adding the valid address (see also https://github.com/ElectronicCats/mpu6050/issues/21#issuecomment-825702249). It should work, but if you can please verify with:

```yaml
external_components:
  - source: github://carlosgs/esphome@patch-39
    components: [mpu6050]
```

d-rez, 2022-09-08:
Hey again, thanks for creating the pull request. Assuming it actually used the external component (and not my already-modified source file), it appears to be working well :) 

Thanks!

closed 2022-09-11 by jesserockz