esphome/issues #2283: Unable to link light effects

tungmeister, 2021-07-30:
### The problem

I have a 3 button light switch with controllable LEDs behind each button. I have a mode I've setup whereby long pressing the middle button activates a mode in which I can control the speed of a ceiling fan. This mode uses the pulse effect on the led behind button 1 and 3 to indicate they can be pressed to adjust the speed. Once the lowest/highest speed is reached the effect stops on the corresponding button. The problem arises when the speed is increased/decreased re-enabling the button  LED effect. At this point the pulse effect will be out of sync between the 2 buttons used to adjust it and from the documentation I can't identify any way to link the effects between the 2 LEDs so that when one is re-enabled it matches the pulse of the other one. see the attached code for further explanation. 

### What is version of ESPHome has the issue?

1.20 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8266

### Component causing the issue

light

### Example YAML snippet

```yaml
- platform: homeassistant
    name: fan speed
    id: fanspeed
    internal: true
    entity_id: input_number.fan_current_speed
    on_value:
      then:
        - if:
            condition:
              and:
              - lambda: 'return id(speedcontrol) == true;'
              - lambda: 'return id(fanspeed).state == 1;'
            then:
              - light.turn_off: button_led_1
        - if:
            condition:
              and:
              - lambda: 'return id(speedcontrol) == true;'
              - lambda: 'return id(fanspeed).state == 6;'
            then:
              - light.turn_off: button_led_3
        - if:
            condition:
              and:
              - lambda: 'return id(speedcontrol) == true;'
              - lambda: 'return id(fanspeed).state > 1;'
            then:
              - light.turn_on:
                  id: button_led_1
                  effect: pulsing
        - if:
            condition:
              and:
              - lambda: 'return id(speedcontrol) == true;'
              - lambda: 'return id(fanspeed).state < 6;'
            then:
              - light.turn_on:
                  id: button_led_3
                  effect: pulsing

...

light:
  - platform: monochromatic
    name: "Blind LED"
    output: led1
    internal: true
    id: button_led_1
    default_transition_length: 0.5s
    effects:
      - pulse:
          name: "pulsing"
          transition_length: 0.5s
          update_interval: 0.5s
  - platform: monochromatic
    name: "Fan LED"
    output: led2
    internal: true
    id: button_led_2
    default_transition_length: 0.5s
  - platform: monochromatic
    name: "Light LED"
    output: led3
    internal: true
    id: button_led_3
    default_transition_length: 0.5s
    effects:
      - pulse:
          name: "pulsing"
          transition_length: 0.5s
          update_interval: 0.5s
```

OttoWinter, 2021-08-01:
> to try and sync when the LED gets turned back on but this doesn't appear to be actually reporting the brightness (did some logging and it looked like it was always 1.0?)

The brightness field represents the brightness only in the on state (so that when you turn the light back on, we can set back the previous brightness). You can use `get_state` to check the ON/OFF state

tungmeister, 2021-08-01:
@OttoWinter should the brightness value constantly update during the pulse effect? if so I would have expected using ` < 0.1` to return true at some point but this didn't seem to be the case? also would the state return as off at some point during the pulse effect when the brightness is visibly 0?

oxan, 2021-08-10:
It might be that the pulse effect just switches the state on/off and leaves the brightness alone (and behaviour might've changed with the recent changes). Try to check the state value as well.

OttoWinter, 2021-08-10:
Yeah, I'd recommend logging the brightness on/off state to see what's going on. You can do that with the `logger.log` action

closed 2021-12-23 by github-actions[bot]