esphome/issues #2285: Build failks with latest docker image (esphome/esphome:dev)

vijaygill, 2021-07-31:
### The problem

Compiling code with latest docker image fails with following error:

`

Compiling .pioenvs/distancemeter-esp32/src/esphome/components/mqtt/mqtt_light.cpp.o
src/esphome/components/mqtt/mqtt_light.cpp:2:56: fatal error: esphome/components/light/light_json_schema.h: No such file or directory
compilation terminated.
*** [.pioenvs/distancemeter-esp32/src/esphome/components/mqtt/mqtt_light.cpp.o] Error 1

`

### What is version of ESPHome has the issue?

latest dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

D1 Mini

### Component causing the issue

src/esphome/components/mqtt/mqtt_light.cpp

paulmonigatti, 2021-07-31:
It appears that the MQTT Light component is being (incorrectly?) included when no lights have been defined in the config file.

As a temporary workaround, adding an empty `light:` into the config will force the missing files to be included.

vijaygill, 2021-08-01:
> It appears that the MQTT Light component is being (incorrectly?) included when no lights have been defined in the config file.
> 
> As a temporary workaround, adding an empty `light:` into the config will force the missing files to be included.

thanks @paulmonigatti that worked!

closed 2021-08-01 by OttoWinter