esphome/issues #2287: Anova Cooker Component Assumes Temperature from Device is in Celsius despite Unit being set to Fahrenheit

DatCFC, 2021-08-03:
### The problem

The Anova component reads the current temperature and target temperature in Celsius even though the cooker itself is displaying in Fahrenheit. 
```[23:22:48][D][climate:369]: 'The Anova' - Sending state:
[23:22:48][D][climate:372]:   Mode: OFF
[23:22:48][D][climate:392]:   Current Temperature: 77.70°C
[23:22:48][D][climate:398]:   Target Temperature: 85.00°C
```
This results in Home Assistant showing the temperature as such:
[image]

### Which version of ESPHome has the issue?

1.20

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

wemos

### Component causing the issue

Anova

buxtronix, 2021-08-03:
Is this purely about what is displayed in the logs?

It seems that the Anova itself is displaying F, HA is showing F, but the logs are in C. Is this right?

DatCFC, 2021-08-03:
> 
> 
> Is this purely about what is displayed in the logs?
> 
> It seems that the Anova itself is displaying F, HA is showing F, but the logs are in C. Is this right?

Not exactly.
- Anova displays 77.7 **F**
- ESPHome reads it as 77.70 **C** (see the log output in first message)
- HA converts the **C** reading from ESPHome to F, so in turn displays it as 9/5*77.70+32 = 172 **F**

closed 2021-08-08 by jesserockz