esphome/issues #2288: CustomAPIDevice broken

wburgers, 2021-08-03:
### The problem

I want to compile a Custom component which extends the CustomAPIDevice class.
Even the examples on the custom component page are not working (https://esphome.io/custom/custom_component.html).

When compiling, the following error message is shown:
```
In file included from src/main.cpp:10:0:
src/test.h:3:66: error: expected class-name before '{' token
 class CustomNativeAPI : public Component, public CustomAPIDevice {
                                                                  ^
```

I also tried the test case in custom.h (https://github.com/esphome/esphome/blob/dev/tests/custom.h) which also errors.

### Which version of ESPHome has the issue?

1.20.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

CustomAPIDevice

### Example YAML snippet

```yaml
esphome:
  name: test
  platform: ESP8266
  board: d1_mini
  includes:
    - test.h # test.h contains the CustomNativeAPI copied from https://github.com/esphome/esphome/blob/dev/tests/custom.h

custom_component:
  - lambda: |-
      auto testcomponent = new CustomNativeAPI();
      return {testcomponent};
```

oxan, 2021-08-03:
I think the examples are missing an `using esphome::api;` -- can you check if it works when you add that?

wburgers, 2021-08-03:
@oxan
That does not work either unfortunately.
```
In file included from src/main.cpp:10:0:
src/test.h:4:16: error: 'esphome::api' has not been declared
 using esphome::api;
                ^
```
Maybe it needs an extra `#include`?

oxan, 2021-08-03:
@wburgers Ah, my bad, that should be `using namespace esphome::api;`.

wburgers, 2021-08-03:
🙈 I don't use C++ daily, so I didn't catch that either.
Sadly that also errors:
```
In file included from src/main.cpp:10:0:
src/test.h:4:26: error: 'api' is not a namespace-name
 using namespace esphome::api;
                          ^
```

oxan, 2021-08-03:
Did you add the API integration to your YAML?

wburgers, 2021-08-03:
Ah I didn't know the yaml had any influence on the included parts of the library, but that does fix the issue!
Thank you @oxan!

I'll see if I can get around to include that in the documentation.

closed 2021-08-03 by wburgers