esphome/issues #2289: Running esphome fails with KeyError: 'framework-stm32cube'

ulido, 2021-08-03:
### The problem

Running esphome fails with a platformio crash. This happens also on a clean new virtual environment with a fresh esphome install via pip. The exact config does not matter, but I still attached an example yaml below.

### Which version of ESPHome has the issue?

1.20.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

nodemcu v2

### Component causing the issue

N/A

### Example YAML snippet

```yaml
esphome:
  name: pool
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: <removed>
  password: <removed>

logger:

api:
  password: <removed>
ota:
  password: <removed>

dallas:
  - pin: D4

sensor:
  - platform: dallas
    name: "Pool Temperature"
    index: 0
```

### Anything in the logs that might be useful for us?

```txt
$ esphome run pool.yaml
INFO Reading configuration pool.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d pool
Processing pool (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
KeyError: 'framework-stm32cube':
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/builder/main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/uli/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/uli/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/uli/.platformio/platforms/espressif8266/builder/main.py", line 238:
    target_elf = env.BuildProgram()
  File "/Users/uli/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/uli/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/uli/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 1062:
    project.install_dependencies()
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/builder/tools/piolib.py", line 898:
    lm.install(spec)
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/package/manager/_install.py", line 48:
    pkg = self._install(
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/package/manager/library.py", line 93:
    if is_builtin_lib(spec.name):
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/commands/lib/helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/commands/lib/helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/platform/base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/platform/_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "/Users/uli/venvs/esphome2/lib/python3.9/site-packages/platformio/platform/_packages.py", line 22:
    owner=self.packages[name].get("owner"),
========================== [FAILED] Took 0.83 seconds ==========================
```

### Additional information

This was supposedly fixed in upstream platformio a while ago (https://github.com/platformio/platformio-core/issues/3777 ), but I still get the error.

oxan, 2021-08-04:
It seems platformio fixed this in the ststm32 repository, try updating the platforms you have installed (`pio platform update`).

ulido, 2021-08-04:
Thank you, this fixed the issue! I didn't know that I needed to update the platforms manually.

closed 2021-08-04 by ulido