esphome/issues #2290: ESP32 and Arduino 3 change failure mode of new[]

Habbie, 2021-08-03:
### The problem

This is not a bug report. This is a prevent-future-bug-reports report.

`new[]` on ESP8266, with Arduino 2.x (` - framework-arduinoespressif8266 3.20704.0 (2.7.4) `), returns a null pointer on failure. Most of the code will fail shortly after an allocation failure. Setting `-DNEW_OOM_ABORT` makes ESPHome abort at the point of alloc failure, instead of the null deref soon after.

I understand from @oxan that on ESP32, and in Arduino 3.x, the abort behaviour is default.

Then @jesserockz reports that code like this also exists, and relies on the null pointer:

```
void DisplayBuffer::init_internal_(uint32_t buffer_length) {
  this->buffer_ = new uint8_t[buffer_length];
  if (this->buffer_ == nullptr) {
    ESP_LOGE(TAG, "Could not allocate buffer for display!");
    return;
  }
  this->clear();
}
```

So, if ESPHome upgrade to Arduino 3.x, or an ESP32 user runs out of memory, or ESPHome adopts `-DNEW_OOM_ABORT`, those users will suddenly get weird crashes instead of a pretty clear error.

So, before the upgrade to 3.x, somebody should go over the components to see if any of them rely on being able to catch alloc failure.

### Which version of ESPHome has the issue?

all

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp01_1mm

### Component causing the issue

display components

ssieb, 2021-08-03:
Is there a code way to avoid the abort for these cases?

oxan, 2021-08-03:
> Is there a code way to avoid the abort for these cases?

`malloc` seems to be the only option that's available everywhere.

ptr727, 2021-08-04:
I'm no Arduino expert, but I've written plenty of C/C++ where the OOM and exception handlers can be overridden in the CRT and/or in the structured/C++ exception handlers for `malloc`, `new`, etc. I.e. it is entirely possible for `malloc` to throw unless you know for certain the handler is set to return NULL, same for `new`.
My 2c's is to conclusively determine how the CRT is configured such that you get the desired return NULL vs. throw behavior.

OttoWinter, 2021-08-05:
In C++ you can also set a global `new` operator: https://en.cppreference.com/w/cpp/memory/new/operator_new

I agree it would make sense to change the behavior so that the abort happens right away, and not later

oxan, 2021-08-05:
I checked, and contrary to my expectations adding `std::nothrow` to `new` suppresses the abort, so that's nice. There's two places where we actually check for failed allocations, `nextion_upload.cpp:278` and `display_buffer.cpp:17`.

ptr727, 2021-08-05:
Some CRT's have two versions of `new()/[]`, one with and one without std::nothrow, one throws and one returns NULL.
Some compilers ignore nothrow, some compilers use it as hint to optimize the code and foregoes exception unwrapping code making for faster smaller code.

As for defaulting to abort, I'd defer to a platform expert, on larger systems it is entirely possible to make an opportunistic allocation, that may fail now but succeed later, e.g. memory pressure, fragmentation, etc. In my experience writing defensively is best, always test for null or for expected exceptions, whatever pattern is used. the caller can always abort if need be, vs. having the platform abort on what could have been a recoverable condition.

oxan, 2021-08-05:
> Some CRT's have two versions of `new()/[]`, one with and one without std::nothrow, one throws and one returns NULL.
> Some compilers ignore nothrow, some compilers use it as hint to optimize the code and foregoes exception unwrapping code making for faster smaller code.

We only care about the combination of the ESP-IDF/Arduino runtime with gcc, which behaves as described above.

> As for defaulting to abort, I'd defer to a platform expert, on larger systems it is entirely possible to make an opportunistic allocation, that may fail now but succeed later, e.g. memory pressure, fragmentation, etc. 

The ESPs don't have a MMU, so overcommitment isn't an option.

> In my experience writing defensively is best, always test for null or for expected exceptions, whatever pattern is used. the caller can always abort if need be, vs. having the platform abort on what could have been a recoverable condition.

Theoretically yes, but in practice we already almost never check for allocation failure, changing all that will be a huge amount of work, and in most cases allocation failure isn't recoverable anyway, so it'd just amount to sprinkling abort calls all over the codebase. In cases where we are able to handle allocation failure, `nothrow` is available.

OttoWinter, 2021-08-10:
> In my experience writing defensively is best, always test for null or for expected exceptions, whatever pattern is used.

Most allocations in ESPHome are in stdlib containers (std::vector, std::string). That code assumes `new` throws on allocation error, and we can't (easily) change it so that it does handle it. So even if we check all instances in our own codebase, the majority would still be unchecked.

closed 2021-08-10 by OttoWinter

ptr727, 2021-08-10:
I didn't know using the C++ std library would be possible with exceptions disabled, but [appears](https://stackoverflow.com/questions/37700365/if-youre-in-the-we-dont-use-exceptions-camp-then-how-do-you-use-the-standar) it is, at least in GCC.

See GCC `-fno-exceptions` and `_GLIBCXX_THROW_OR_ABORT` in [code](https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/libstdc%2B%2B-v3/include/bits/c%2B%2Bconfig#L202):
```
#ifndef _GLIBCXX_THROW_OR_ABORT
# if __cpp_exceptions
#  define _GLIBCXX_THROW_OR_ABORT(_EXC) (throw (_EXC))
# else
#  define _GLIBCXX_THROW_OR_ABORT(_EXC) (__builtin_abort())
# endif
#endif
```