esphome/issues #2291: issues with new light model and transition.

tungmeister, 2021-08-04:
### The problem

Just updated to hass 2021.8 in order utilise the new lighting model I have some smart bulbs setup as RGBWW lights.
When using 'color_interlock: true' the hass UI displays as I would expect. No extra sliders for colour and whiteness with the separate tabs for colour and white temperature, selecting a colour will switch to it as expected. The only issue I'm having is that I have a 0.5s transition setup on the light as standard for all their changes. Due to the interlock when switching from WW to RGB there's quite a delay as the light has to turn off then turn on the RGB LEDs which results in a moment of total darkness. 
I appreciate that the interlock's main purpose is for lights that can't output to both RGB and WW channels at one or due to power limitations but without using it it's back to the additional sliders when changing to colour in order for it to display. (which isn't very family/user friendly)
An ideal solution would be to have an additional option for interlock such as 'crossfade: true' which would start the transition to the RGB LEDs at the same time as turning off the WW ones removing the point at which the bulb is completely off. Thoughts?

### Which version of ESPHome has the issue?

1.21 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8266

### Component causing the issue

light

oxan, 2021-08-05:
I'm not opposed to it, but it's more of a feature request than an bug report. 

I'm working on customizable transitions, which will make it possible to do this in a clean way, but that'll probably take a release or two to be finished.

tungmeister, 2021-08-05:
No problem wasn't sure if this was the intended behavior or if it should have transitioned hence posting it here.

Tugzrida, 2021-08-06:
I have an RGBW light(with colour interlock enabled) running ESPHome 1.20.4. Controlling it from HA 2021.8, when I set white brightness to 0 in order to show a colour, the white fades to black, and then the colour comes in instantly. When setting the white back to 100, the colour snaps to black, and then the white fades in.

The previous version of ESPHome I had flashed it with was 1.19.4 and with it, the colour fades in and out as expected, so it seems that this is a bug in the new colour model of 1.20.0.

Happy to open a new issue and do more testing if required, but I got quite lost trying to follow all the changes in HA and ESPHome recently so I'm not sure where this fits in and if it's expected with the current state of things.