esphome/issues #2292: Default_transition_length not being respected with constant_brightness: true

tungmeister, 2021-08-04:
### The problem

I have a few RGBWW lights set up with the default_transition_length being specified and ```constant_brightness: true```. This is respected when changing colour or WW temperature within each domain however if changing from RGB to WW or WW to RGB the transition is not respected and the change takes place instantly. I've tried making the change with a scene and specifying a transition length from within home assistant but the same result is observed. Setting ```constant_brightness: false``` results in the expected behaviour. This has only been an issue since hass 2021.8.

### Which version of ESPHome has the issue?

1.21 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8266

### Component causing the issue

light

### Example YAML snippet

```yaml
light:
  - platform: rgbww
    name: fireplace left
    constant_brightness: true
    red: output_red
    green: output_green
    blue: output_blue
    default_transition_length: 1s
    warm_white: output_warm_white
    cold_white: output_cold_white
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K
```

oxan, 2021-08-05:
> however if changing from RGB to WW or WW to RGB

What exactly do you mean by that? From your config it looks like you don't have the color interlock enabled, so you don't switch between the RGB and WW modes, they're both active at the same time.

I also can't reproduce this from the HA UI.

tungmeister, 2021-08-05:
I believe the issue must stem from a conversion process. I've just pulled my backup from yesterday (made a lot of hass changes today) and the lights were being controlled by the below scenes which were setup when the lights were running espurna (finally got round to migrating everything over to esphome a few weeks back):
```
  scene:
    - name: Snug Normal Lighting
      entities:
        light.desk_lamp:
          state: on
          brightness: 255
          color_temp: 425
        light.fireplace_right:
          state: on
          brightness: 255
          color_temp: 425
        light.fireplace_left:
          state: on
          brightness: 255
          color_temp: 425
    - name: Snug Dim Lighting
      entities:
        light.desk_lamp:
          state: on
          brightness: 230
          rgb_color: [255, 115, 0]
          white_value: 0
        light.fireplace_right:
          state: on
          brightness: 230
          rgb_color: [255, 173, 0]
          white_value: 0
        light.fireplace_left:
          state: on
          brightness: 230
          rgb_color: [255, 173, 0]
          white_value: 0
```

these were behaving fine before updating to 2021.8 but after the update with 'constant_brightness:true' set the transition from one scene to another was instantaneous - setting it to false returned the behavior back to normal with ```default_transition_length``` being obsered. I've since altered my scene configs to use ```rgbww_color:``` so the issue is no longer present for me though could still potentially a bug somewhere?

closed 2021-08-10 by jesserockz