esphome/issues #2293: RGBCCT with one channel color_temp, one channel brightness

Alfiegerner, 2021-08-04:
### The problem

I have bulbs (brand Kogan from Aus) with 5 channels:

- Red
- Green
- Blue
- Color Temp
- Brightness

Previously I've had these working with the following custom_component:
[RBGWWSinglepintempandbrightnessLightOutput.zip](https://github.com/esphome/issues/files/6934691/RBGWWSinglepintempandbrightnessLightOutput.zip)

A custom component was required as there was no way specify color_temp and brightness using one channel each.   The problem is this has stopped working in HA.  The color picker is absent from the UI as well as the color_temp slider.

If it's possible it would be great to do away with the custom component and have this type of bulb supported directly, but failing that I'm trying to get them working again.

I guess that the change with the supported_color_modes changes has broken this. I've seen the changes going into dev branch and wondering if someone could help me get this working?

Could someone confirm what I would need for traits, e.g old version.:

```
    LightTraits get_traits() override {
        auto traits = LightTraits();
        traits.set_supports_brightness(true);
        traits.set_supports_rgb(true);
        traits.set_supports_rgb_white_value(false);
        traits.set_supports_color_temperature(true);
// ******* CHANGE BELOW VALUES ******* 
        traits.set_min_mireds(154); //5700K
        traits.set_max_mireds(500); //3000K
// ******* Nothing more the change ******* 
        return traits;
    }

```

I'm guessing the write state would stay the same?

    void write_state(LightState *state) override {
        float red, green, blue, brightness, cwhite, wwhite, mathtemp;
        state->current_values_as_rgbww(&red, &green, &blue, &cwhite, &wwhite);
        this->pin_red_->set_level(red);
        this->pin_green_->set_level(green);
        this->pin_blue_->set_level(blue);
        state->current_values_as_brightness(&brightness);
        // do some maths to work our brightness and temp) 
      	 this->pin_brightness_->set_level(brightness);
         this->pin_temp_->set_level(mathtemp);
    }

### Which version of ESPHome has the issue?

1.19.4

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

light

OttoWinter, 2021-08-05:
The custom component should use the new color_mode API to register a `RGB_COLOR_TEMP` color mode.

Then for reading out the values you can just use `current_values` and read the RGB+color_temp properties manually + send them to the light.

OttoWinter, 2021-08-05:
Closing because this is an error in a custom component (and we don't maintain those) - feel free to ask questions though in the community forums or the discord

closed 2021-08-05 by OttoWinter