esphome/issues #2294: Latest esphome homeassistant addon broken

niahane, 2021-08-05:
### The problem

Today I tried to update my esphome addon from version 1.18 to 1.20.4 to see if the wireless connectivity problem of the previous versions was solved, now it is no longer possible to do anything.
If I click install I get a black screen with no entry, if I click log, nothing seems to work anymore.
I tried to logout and relogin but nothing, I tried logout and clear the browser history and relogin but nothing, I tried other browsers but the problem persists.

I understand that it is an open source system and you don't have to expect anything because most of the work is done only for passion, I love open source and I love esphome up to version 1.18.0 everything worked perfectly but in the latest versions it is only getting worse and worse, I wonder if anyone tests the software before releasing an official version?

### Which version of ESPHome has the issue?

1.20.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

Main interface

### Anything in the logs that might be useful for us?

```txt
Log are not visible
```

### Additional information

[image]
[image]

sabellwind, 2021-08-05:
Hi there,

I'm also experiencing the same thing.
Running Home Assistant Supervised 2021.8.1.
The add-on will not start from Supervisor, but all of my ESPHome devices are active.

Supervisor has the following logs:
`21-08-06 08:03:58 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/15ef4d2f_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/15ef4d2f_esphome/tmpqhl_n16p'
21-08-06 08:03:58 ERROR (MainThread) [supervisor.addons.addon] Add-on 15ef4d2f_esphome can't write options`

Pacococo, 2021-08-08:
Buenas, tengo el mismo problema que el anterior usuario, sabellwind, desde la actualización 2021.8.0, los dispositivos ESPHome funcionan, pero el add-on no es capaz de arrancar, dando el problema siguiente:
21-08-08 13:32:35 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmpjb7u72dk'
21-08-08 13:32:35 ERROR (MainThread) [supervisor.addons.addon] Add-on a0d7b954_esphome can't write options
21-08-08 13:34:05 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmp4una3_i_'
21-08-08 13:34:05 ERROR (MainThread) [supervisor.addons.addon] Add-on a0d7b954_esphome can't write options
21-08-08 13:46:40 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmpt_qpi603'
21-08-08 13:46:40 ERROR (MainThread) [supervisor.addons.addon] Add-on a0d7b954_esphome can't write options
No sé cómo solucionar el problema, porque han habido actualizaciones del add-on ESPHome, se actualiza, pero sigue sin arrancar. Espero noticias suyas, gracias

imoldovavan, 2021-08-09:
I also get this in the log...
`21-08-09 11:58:07 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory: '/data/addons/data/a0d7b954_esphome/tmphgrztnkp'
21-08-09 11:58:07 ERROR (MainThread) [supervisor.addons.addon] Add-on a0d7b954_esphome can't write options
21-08-09 11:58:07 WARNING (MainThread) [supervisor.addons] Can't start Add-on a0d7b954_esphome`

WebfinityDesign, 2021-08-23:
I've got the same problem. Any news on this?

kocsenc, 2021-08-23:
Easiest workaround for now is to uninstall and reinstall add on. All my configs were restored but YMMV.  Maybe snapshot before just in case.

WebfinityDesign, 2021-08-23:
It is the first time I install Esphome, so I have no snapshot to revert to. I tried uninstall/reinstall several times but that has no effect

casperklein, 2021-08-23:
Depending on your error message, you can just create the missing directory, that helped me. For example:

> 21-08-09 11:58:07 ERROR (MainThread) [supervisor.utils.json] Can't write /data/addons/data/a0d7b954_esphome/options.json: [Errno 2] No such file or directory

`mkdir /data/addons/data/a0d7b954_esphome/` should fix the problem.

WebfinityDesign, 2021-08-23:
The strange thing is I don't have an error message for ESPHome at all... (At least, I can't find any)

WebfinityDesign, 2021-08-24:
So today I installed the upgrade to 2021.8.1 in the hope that will solve the problem, but it didn't.
In the configuration there is a field for a host (adress?). Do I need to fill in something in it?

WebfinityDesign, 2021-08-25:
UPDATE: today I upgraded to 2021.8.2 but the problem is still the same

thebitpit, 2021-09-08:
I have the same black screen problem. Looking at the ESPHome log I see a lot of INFO 200 entries. I assume that indicates the make is happening. The feedback is just not showing up on the black window.

At one point the black window started working and I could see ESPHome interacting with my devices. Then it  stopped working and has not worked since. I have no idea what happened.

closed 2022-01-24 by github-actions[bot]

RHAD1969, 2022-02-04:
Same issue here. Re-installing does not help. Errors in log:

the-habu, 2022-03-16:
Same issue here. All up to date.