esphome/issues #2295: esphome libraries includes broken in last dev (from https://github.com/esphome/esphome/pull/2056)

SeByDocKy, 2021-08-05:
### The problem

Hi, after I installed the last dev build, one on my yaml codeusing external lib is no more compiling

### Which version of ESPHome has the issue?

dev cb8a6f6

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32 WROOM32

### Component causing the issue

core

### Example YAML snippet

```yaml
esphome:
  name: station_meteo
  platform: ESP32
  board: esp32dev

  includes:
    - veml6070_custom_sensor.h
    - tx23_custom_sensor.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6070"
    - "https://github.com/egzumer/Arduino-LaCrosse-TX23-Library"
    - SPI
```

### Anything in the logs that might be useful for us?

```txt
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome==1.21.0.dev0', 'console_scripts', 'esphome')())
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 750, in main
    return run_esphome(sys.argv)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 737, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 332, in command_run
    exit_code = write_cpp(config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 144, in write_cpp
    generate_cpp_contents(config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 156, in generate_cpp_contents
    CORE.flush_tasks()
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/__init__.py", line 616, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/local/lib/python3.7/dist-packages/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 136, in wrapped
    await coro(conf)
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/config.py", line 342, in to_code
    cg.add_library(None, None, lib)
  File "/usr/local/lib/python3.7/dist-packages/esphome/cpp_generator.py", line 552, in add_library
    CORE.add_library(Library(name, version, repository))
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/__init__.py", line 665, in add_library
    "".format(library, other)
ValueError: Adding named Library with repository failed! Libraries https://github.com/egzumer/Arduino-LaCrosse-TX23-Library and https://github.com/adafruit/Adafruit_VEML6070 requested with conflicting repositories!
```

OttoWinter, 2021-08-05:
Related to https://github.com/esphome/esphome/pull/2056

oxan, 2021-08-05:
@agners

closed 2021-08-07 by jesserockz