esphome/issues #2296: Pulse Effect not working

tungmeister, 2021-08-05:
### The problem

The pulse effect is not working with the new lighting mode - I have an RGBWW light setup that I've just attempted to assign a pulse effect to - when turning on the effect the light goes out but doesn't pulse staying visibly off.  Logging shows that the pulse effect has been activated it just doesn't work - I've tried using just the white channels or RGB with no success as well as the standard and a custom pulse. 

### Which version of ESPHome has the issue?

1.21 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

light

### Example YAML snippet

```yaml
light:
  - platform: rgbww
    name: ${display_name}
    red: output_red
    green: output_green
    blue: output_blue
    effects:
      - pulse:
          name: "pulsing"
          transition_length: 0.5s
          update_interval: 1s
    default_transition_length: 1s
    warm_white: output_warm_white
    cold_white: output_cold_white
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K
```

oxan, 2021-08-05:
It works for me (from HA). How did you start the effect?

tungmeister, 2021-08-05:
Selecting it from the light control popup card.  I've just tried triggering it from within the code when the light is turned on and it results in the same behavior - light turns visibly off and stays that way. Here's the full config:
```
substitutions:
  display_name: fireplace_right
  
esphome:
  name: fireplace_right
  platform: ESP8266
  board: esp8285
  esp8266_restore_from_flash: true
  
wifi:
  ssid: !secret SSID
  password: !secret pass
  domain: ".chobnet"
  use_address: 10.0.0.37
  ap:
    ssid: ${display_name}_AP
    password: !secret pass
    ap_timeout: 3min

captive_portal:

logger:

ota:
  password: !secret gen-pass

api:
  password: !secret gen-pass
  
switch:
  - platform: restart
    name: ${display_name} Restart

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO4
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_cold_white
    pin: GPIO14
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO12

light:
  - platform: rgbww
    name: ${display_name}
    red: output_red
    green: output_green
    blue: output_blue
    id: test
    effects:
      - pulse:
      - pulse:
          name: "pulsing"
          transition_length: 0.5s
          update_interval: 1s
    default_transition_length: 1s
    warm_white: output_warm_white
    cold_white: output_cold_white
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K
    on_turn_on:
      light.turn_on:
        id: test
        effect: pulsing
```

oxan, 2021-08-05:
Hmm, this seems to have something to do with the transition. If I disable the transition (i.e. `transition_length: 0s`), it works for me. Can you confirm that also fixes it for you?

tungmeister, 2021-08-05:
Latest version - i've just reinstalled so it's pulled the current zip. recompiled and flashed and it's still not working

oxan, 2021-08-05:
Hmm, I'm not familar with how the add-on works, does that really pull the last version from GitHub? Reason I'm asking is that I suspect you're hitting issues that were fixed with this commit: https://github.com/esphome/esphome/commit/ceb0564ebf9ebbab9d22babbcf4f695e8c92df6f

In any case, can you share the log output you get?

tungmeister, 2021-08-05:
The addon is pulling https://github.com/esphome/esphome/archive/dev.zip which can only be the latest version? 
the logs just show the pulse effect being specified:
```
[19:52:31][C][ota:032]:   Using Password.
[19:52:31][C][api:095]: API Server:
[19:52:31][C][api:096]:   Address: 10.0.0.37:6053
[19:52:58][D][light:038]: 'fireplace_right' Setting:
[19:52:58][D][light:110]:   Effect: 'Pulse'
```

closed 2021-08-07 by jesserockz