esphome/issues #2297: Tuya Light Min Value datapoint doesn't seem to persist if set in config

richardhopton, 2021-08-06:
### The problem

On esphome 1.20.4, when setting the min_value_datapoint to 3 and min_value to 10 in the tuya light config, these values don't seem to apply.  This means the dimmer can be set to a value lower than the MCU will accept causing the value to snap back to the MCU's minimum.

The MCU min_value_datapoint seems to start off at around 150 on all my TreatLife DS01 switched and 250 on my TreatLife DP01, which results in an achievable minimum brightness of 15% and 25% respectively.

Here's the config I am using.
```
esphome:
  platform: ESP8266
  board: esp01_1m
  name: dining_room_dimmer
  build_path: "builds/dining_room_dimmer/"

wifi: 
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  password: !secret api_password
ota:
  password: !secret ota_password
  id: config_ota

logger:
  baud_rate: 0 # (UART logging interferes with mcu)
  level: VERY_VERBOSE

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:
  id: tuya_light

light:
  - platform: tuya
    name: "Dining Room Dimmer"
    dimmer_datapoint: 2
    min_value_datapoint: 3
    switch_datapoint: 1
    min_value: 10
    max_value: 1000
```

Running this lambda on boot with a 10s delay, or as `turn_on_action` for a template switch does apply the setting successfully, and the change survives a re-flash but not a power cycle.
```
 - lambda: id(tuya_light).set_datapoint_value(3, 10);
```

Once this value is correctly set it is then possible for the dimmers datapoint (2 in my case) to be set to a value lower than the factory set min_value_datapoint, resulting in measurably lower output and less bright lighting.

### Which version of ESPHome has the issue?

1.20.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

TYWE3S (Generic 8266)

### Component causing the issue

tuya.light

ssieb, 2021-08-06:
The datapoint is getting sent in `setup()` before the initialization is done.  So it would only work on an MCU that accepts datapoint changes before the handshaking is done.

richardhopton, 2021-08-06:
Please note setting the min_value_datapoint, at least on my switches, results in the dimmer datapoint also being set to the same value.

closed 2021-12-19 by github-actions[bot]