esphome/issues #2298: ESP32-C3 deep sleep errors

markusg1234, 2021-08-06:
### The problem

Errors as seen in logs when trying to use deep sleep on ESP32-C3.

Regards,
Markus

### Which version of ESPHome has the issue?

V1.21.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Esp32-C3

### Component causing the issue

deep sleep

### Example YAML snippet

```yaml
esphome:
  name: esp32-c
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master

deep_sleep:
  run_duration: 60sec
  sleep_duration: 60min
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/platformio/platform-espressif32#feature/arduino-idf-master)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.5bb8177 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-riscv32-esp 8.4.0+2021r1
Dependency Graph
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
Compiling /data/test/.pioenvs/test/src/esphome/components/deep_sleep/deep_sleep_component.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/application.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/color.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/component.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/controller.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/esphal.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/helpers.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/core/log.cpp.o
src/esphome/components/deep_sleep/deep_sleep_component.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::begin_sleep(bool)':
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:5: error: 'esp_sleep_enable_ext0_wakeup' was not declared in this scope
     esp_sleep_enable_ext0_wakeup(gpio_num_t((*this->wakeup_pin_)->get_pin()), level);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:5: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
     esp_sleep_enable_ext0_wakeup(gpio_num_t((*this->wakeup_pin_)->get_pin()), level);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_sleep_enable_wifi_wakeup
src/esphome/components/deep_sleep/deep_sleep_component.cpp:81:5: error: 'esp_sleep_enable_ext1_wakeup' was not declared in this scope
     esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:81:5: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
     esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_sleep_enable_wifi_wakeup
*** [/data/test/.pioenvs/test/src/esphome/components/deep_sleep/deep_sleep_component.cpp.o] Error 1
========================== [FAILED] Took 1.64 seconds ==========================
```

### Additional information

Also deep sleep using the following, has similar errors.

deep_sleep:
   wakeup_pin: GPIO4
   run_duration: 60s

MFlasskamp, 2022-02-16:
The ESP32-C3 is missing some of the wakeup sources the ESP32 has:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#wakeup-sources
Until there is a more elegant implementation my bugfix is to exclude (`#ifdef...`) the code part for the ESP32-C3 variant in `esphome/components/deep_sleep/deep_sleep_component.cpp`:

```c
#if !defined(USE_ESP32_VARIANT_ESP32C3)
  if (this->wakeup_pin_ != nullptr) {
    bool level = !this->wakeup_pin_->is_inverted();
    if (this->wakeup_pin_mode_ == WAKEUP_PIN_MODE_INVERT_WAKEUP && this->wakeup_pin_->digital_read()) {
      level = !level;
    }
    esp_sleep_enable_ext0_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), level);
  }
  if (this->ext1_wakeup_.has_value()) {
    esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
  }

  if (this->touch_wakeup_.has_value() && *(this->touch_wakeup_)) {
    esp_sleep_enable_touchpad_wakeup();
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  }
#endif
```

Deep sleep of ESP32-C3 (LilyGo TTGO T-01C3 module) is working now.

agners, 2022-02-16:
There is a PR worked on, see https://github.com/esphome/esphome/pull/3066

p1ngod, 2022-02-19:
I have an Espressif C3 Mini Devkit at home, any need to test it?

agners, 2022-02-19:
Real world tests are always helpful!

closed 2022-02-19 by OttoWinter