esphome/issues #2304: RGBWW interlocked light not reporting color_temp

tungmeister, 2021-08-08:
### The problem

RGBWW light with ```color_interlock: true```  isn't reporting color_temp back to home assistant - it can be set from home assistant but never shows in the attributes/UI.

HA attributes:
```
...
color_mode: color_temp
brightness: 255
color_temp: 0
effect: None
...
```
the following shows in the hass log:

```Entity light.all_living_room_spots has incorrect color temperature 0.0````

### Which version of ESPHome has the issue?

v1.xx-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

light

### Example YAML snippet

```yaml
light:
  - platform: rgbww
    name: '${name}'
    id: '${device}'
    default_transition_length: !secret transrate
    red: output_red
    green: output_green
    blue: output_blue
    warm_white: output_warm_white
    cold_white: output_cold_white
    warm_white_color_temperature: 2800 K
    cold_white_color_temperature: 6200 K
    color_interlock: true
    effects:
      - pulse:
```

tungmeister, 2021-08-10:
@oxan I'd missed this before but the below is shown in the esphome log:
```
[15:42:48][W][light:141]: 'spot1' - This light does not support color mode Color temperature!
[15:42:48][I][light:393]: 'spot1' - Keeping current color mode Cold/warm white for call without color mode.
[15:42:48][D][light:349]: 'spot1' - Setting cold/warm white channels using white/color temperature values.
[15:42:48][D][light:038]: 'spot1' Setting:
[15:42:48][D][light:068]:   Color temperature: 317.0 mireds
[15:42:48][D][light:073]:   Cold white: 62%, warm white: 100%
```

OttoWinter, 2021-08-10:
What version of HA is this?

tungmeister, 2021-08-10:
@OttoWinter 2021.8.5 + latest from Dev branch

malteger, 2021-08-11:
This also happens for cwww lights. It is most noticeable in HA, as the color temperature slider will always be at the lowest value if the light is turned on or the controlling popup opens.

esphome version 1.21.0b1

HA version 2021.8.3

```yaml
- platform: cwww
  name: ${bulbname}
  id: cwww_light
  warm_white: output_warm
  cold_white: output_cold
  cold_white_color_temperature: 6500K
  warm_white_color_temperature: 2700K
  constant_brightness: true
```

esphome log:
```
[14:02:48][W][light:141]: 'Schreibtischlampe Malte' - This light does not support color mode Color temperature!
[14:02:48][D][light:349]: 'Schreibtischlampe Malte' - Setting cold/warm white channels using white/color temperature values.
[14:02:48][D][light:038]: 'Schreibtischlampe Malte' Setting:
[14:02:48][D][light:068]:   Color temperature: 364.0 mireds
[14:02:48][D][light:073]:   Cold white: 29%, warm white: 100%
[14:02:48][D][light:087]:   Transition length: 1.0s
```

Homeassistant state:
```
min_mireds: 153.84616088867188
max_mireds: 370.370361328125
supported_color_modes:
  - color_temp
color_mode: color_temp
brightness: 255
color_temp: 0
friendly_name: Schreibtischlampe Malte
supported_features: 40
```

oxan, 2021-08-11:
I haven't debugged this thoroughly yet, but this also happens _after_ setting the light to a given color temperature?

My first thought was this would happen because the color temperature on the ESPHome side is initialized to zero, but that should be overwitten once you actually set a color temperature.

malteger, 2021-08-11:
Yes, what happens is that the light is turned on with the last color_temperature known to esphome, but this is not reported back to HA, so there it is displayed as the lowest possible value. The light itself is fine, the issue is only with the value displayed in HA

closed 2021-08-15 by jesserockz