esphome/issues #2305: Automation Condition Error "Switch.is_on/off" when using 'Template Number'

3ative, 2021-08-08:
### The problem

Throws an error with condition, **switch.is_on/off**

_I hope this screencap is sufficient..._
E.G.
[image]

### Which version of ESPHome has the issue?

v1.20.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

Template Number

### Example YAML snippet

```yaml
number:
  - platform: template
    name: "Bedroom Curtains"
    update_interval: never
    optimistic: true
    min_value: 0
    max_value: 100
    step: 10
    on_value:
      if:
        condition:
          switch.is_on: test_swtich
        then:
          - stepper.set_target:
              id: curtain_motor_v2
              target: !lambda "return x *65;"
```

oxan, 2021-08-08:
You're aware there's a typo in your entity name?

3ative, 2021-08-08:
> You're aware there's a typo in your entity name?

I am aware. However, that's not the issue, as the ID is correct...
[image]

oxan, 2021-08-09:
That's a sensor, not a switch, so you can't apply the `switch.is_on` condition to it. Instead, you should make it a binary sensor if it isn't already, and use the `binary_sensor.is_on` condition.

3ative, 2021-08-09:
@oxan 
I'll have to differ, 'switch.is_on;' IS a valid condition for a sensor, I've used it many times before, on switches, sensors, etc - both directly in code (_as in my code example_) and from HA API calls (_I.E. my screenshot above_) 

E.G. Here is an instance I used in my 'freshmatic' project, 
[image]

3ative, 2021-08-09:
@oxan 
OIC, so when bringing in a 'switch.XXXX' ESPHome still uses it as a 'sensor' - coz that where it was brought in to... Got ya!  🙄

I have made the necessary code adjusts and everything is working as I wanted. 👍
Thank you for the assist.

Q: Should I close this thread or do you want to keep it open in order to highlight the "Better Error Message" tag?

closed 2021-12-08 by 3ative