esphome/issues #2307: Error every time HA starts: Error doing job: Task exception was never retrieved

iridris, 2021-08-09:
### The problem

Every time HA starts, I get the following error logged. This has persisted for several versions of both HA and ESPHome. It appears that this is related to a text_sensor that pulls in the state of a MyQ garage door from HA, if I comment out that text_sensor the error goes away.

This seems to just be a cosmetic bug as everything seems to work fine, I'd just like to eliminate the error at startup.

`2021-08-08 12:39:39 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 229, in send_home_assistant_state_event
    and event.data["old_state"].state
AttributeError: 'NoneType' object has no attribute 'state'`

### Which version of ESPHome has the issue?

1.20.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
text_sensor:
  - platform: homeassistant
    entity_id: cover.garage_door
    id: garage_display_sensor
    on_value:
      then:
        - display.page.show: !lambda |-
            if (id(garage_display_sensor).state=="open") {
              return id(garage_display_open);
            } if (id(garage_display_sensor).state=="opening") {
              return id(garage_display_opening);
            } if (id(garage_display_sensor).state=="closing") {
              return id(garage_display_closing);
            } if (id(garage_display_sensor).state=="closed") {
              return id(garage_display_closed);
            } else {
              return id(dunno);
            }
        - component.update: garage_display1
```

oxan, 2021-08-09:
Would be useful to know your HA version.

holdestmade, 2021-08-09:
Same issue here

iridris, 2021-08-09:
> 
> 
> Would be useful to know your HA version.

Currently on core-2021.8.4 but this issue has persisted through at least the June and July updates as well, I believe.

HenJou, 2021-08-10:
Hi I also got the same issue and if I can help with the debugging i will.
Running Ha ghcr.io/home-assistant/home-assistant:2021.8.3 on docker 20.10.8 on Raspberry pi 4. My esphome is running on a debian intel also on docker and version is 1.20.4. I can't tell which one of the 14 esp8266 is causing the problem.

my text_sensors on the esphome is all like this

> text_sensor:
>   - platform: template
>     name: $systemNameshort Uptime 
>     id: uptime_human
>     icon: mdi:clock-start 
>   - platform: wifi_info
>     ip_address:
>       name: $systemNameshort ESP IP Address 
>       id: ip

and the error on my HA is as follows

>   File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 229, in send_home_assistant_state_event,
>     and event.data["old_state"].state,
> AttributeError: 'NoneType' object has no attribute 'state',
> 2021-08-10 15:42:33 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved,
> Traceback (most recent call last):,

Spirituss, 2021-08-15:
Have the same issue. The messages appear always during HA restart and periodically even after.

oxan, 2021-09-11:
A fix has been merged into HA, will be available in 2021.10.

closed 2021-09-11 by oxan