esphome/issues #2309: Sonoff/Shelly bootloop/wifi connection issues with illegal instruction exception

oxan, 2021-08-10:
### The problem

There's been an influx of issues about Sonoff devices bootlooping/being unable to connect to wifi since v1.19 or v1.20. To save me and you from having to look a lot of issues, I want to centralize the discussion here.

If you have these symptoms, please:
* Obtain the full logs over the serial port, either using the `esphome logs` command or the dashboard.
* Check if there's an "Exception (0)" message in the output. If not, you don't have _this_ problem, please open a **new** issue.

If you have this issue, it'd be appreciated if you can try the following things and report back whether this solves this problem:
* If you upgraded from a version prior to v1.18, check whether v1.18 works.
* If you upgraded to a version beyond v1.19, check whether v1.19 works.
* Try setting the log level to INFO instead of the default DEBUG.
* If you have it enabled, try disabling the captive portal.

Please also include the full serial output you obtained, along with your full YAML, **between \<details\> and \</details\> tags**.

### Which version of ESPHome has the issue?

v1.19 or v1.20

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

Sonoff / Shelly

dattas, 2021-08-10:
Removing `arduino_version` resolved the issue for me, it may be useful to remove it from the sample configurations defined on the ESPHome website is that behavior causes problems: https://esphome.io/devices/sonoff_basic.html

Warter21, 2021-08-13:
I am still struggling with the new versions (1.19-1.20). It is working fine with 1.18.
If you need any more details, please let me know. Thanks

HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash - Wemos D1 Mini

RAM:   [======    ]  64.4% (used 52748 bytes from 81920 bytes)
Flash: [=====     ]  45.8% (used 478528 bytes from 1044464 bytes)

<details>
[12:29:01]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[12:29:01]
[12:29:01]Exception (29):
WARNING Exception type: Access to invalid address: STORE (wild pointer?)
[12:29:01]epc1=0x40229ab1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d epaper-display -t idedata
WARNING Decoded 0x40229ab1: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator<char> const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:611
[12:29:03]
[12:29:03]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[12:29:03]
[12:29:03]ctx: sys
[12:29:03]sp: 3fffeb90 end: 3fffffb0 offset: 0190
[12:29:03]3fffed20:  00000031 3fffb9b0 00000009 40229aac  
WARNING Decoded 0x40229aac: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator<char> const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:621
[12:29:03]3fffed30:  00000001 3fffede0 00000000 40229bb2  
WARNING Decoded 0x40229bb2: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:481
[12:29:03]3fffed40:  3fff5270 00000001 3fff4d0c 3731ee16  
[12:29:03]3fffed50:  00000000 00000000 00000009 4021a39c  
WARNING Decoded 0x4021a39c: String::concat(char)
[12:29:03]3fffed60:  00000009 00000020 3fffede0 3fffee00  
[12:29:03]3fffed70:  00000000 00000009 3fffb9b0 40229cb0  
WARNING Decoded 0x40229cb0: std::string::_M_replace_safe(unsigned int, unsigned int, char const*, unsigned int) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:687
[12:29:03]3fffed80:  3fffedd0 00000009 3fffee00 3fff4958  
[12:29:03]3fffed90:  3fffee00 00000009 3fffb9b0 40229d56  
WARNING Decoded 0x40229d56: std::string::assign(char const*, unsigned int) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:278
[12:29:03]3fffeda0:  00000000 3fff494c 3fffede0 40219e35  
WARNING Decoded 0x40219e35: String::invalidate()
[12:29:03]3fffedb0:  3fffede0 3fffb98c 3fffee00 3fff4ba0  
[12:29:03]3fffedc0:  3fffb8ac 3fffb9b0 3fffb9b0 40229d84  
WARNING Decoded 0x40229d84: std::string::assign(char const*) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:1134
[12:29:03]3fffedd0:  3fffee00 00000000 000003e8 40201236  
WARNING Decoded 0x40201236: std::_Function_base::_Base_manager<esphome::api::APIConnection::APIConnection(AsyncClient*, esphome::api::APIServer*)::{lambda(void*, AsyncClient*, void*, unsigned int)#4}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<esphome::api::APIConnection::APIConnection(AsyncClient*, esphome::api::APIServer*)::{lambda(void*, AsyncClient*, void*, unsigned int)#4}> const&, std::_Manager_operation) at api_connection.cpp
[12:29:03]3fffede0:  302e3000 312e302e 3fffb98c 40229d98  
WARNING Decoded 0x40229d98: std::string::operator=(char const*) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:557
[12:29:03]3fffedf0:  3fffb8ac 00000000 3fffb98c 40201d19  
WARNING Decoded 0x40201d19: esphome::api::APIConnection::APIConnection(AsyncClient*, esphome::api::APIServer*)
[12:29:03]3fffee00:  302e3031 312e302e 89200037 40201e84  
WARNING Decoded 0x40201e84: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), esphome::api::APIConnection::APIConnection(AsyncClient*, esphome::api::APIServer*)::{lambda(void*, AsyncClient*, void*, unsigned int)#4}>::_M_invoke(std::_Any_data const&, void*, AsyncClient*, void*, unsigned int) at api_connection.cpp
[12:29:03]3fffee10:  4021f5b0 1100000a 00000068 4021aca0  
WARNING Decoded 0x4021f5b0: precache
WARNING Decoded 0x4021aca0: operator new(unsigned int)
[12:29:03]3fffee20:  3fffb8ac 3fffb98c 3fff5d5c 40205129  
WARNING Decoded 0x40205129: std::_Function_handler<void (void*, AsyncClient*), esphome::api::APIServer::setup()::{lambda(void*, AsyncClient*)#1}>::_M_invoke(std::_Any_data const&, void*, AsyncClient*) at api_server.cpp
[12:29:03]3fffee30:  3fff5d6c 3fff63cc 3fffb97c 40213e3c  
WARNING Decoded 0x40213e3c: AsyncClient::AsyncClient(tcp_pcb*)
[12:29:03]3fffee40:  3fff5d6c 3fffb8ac 3fff526c 40100119  
WARNING Decoded 0x40100119: std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const
[12:29:03]3fffee50:  3fff5d6c 3fffb8ac 3fff526c 40213ec3  
WARNING Decoded 0x40213ec3: AsyncServer::_accept(tcp_pcb*, long)
[12:29:03]3fffee60:  3fff63cc 3fffb654 0000001b 3fff4ba0  
[12:29:03]3fffee70:  3fff526c 3fff4b84 3fff4b85 40213f10  
WARNING Decoded 0x40213f10: AsyncServer::_s_accept(void*, tcp_pcb*, long)
[12:29:03]3fffee80:  00000001 3fff2d30 00000000 4022db71  
WARNING Decoded 0x4022db71: tcp_process at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:942
 (inlined by) tcp_input at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:438
[... cut ...]
            //Fair
            it.print(x, y, id(mdi_wifi), TextAlign::BASELINE_CENTER, "󰤢");
            ESP_LOGI("WiFi", "Fair");
        } else if (id(wifisignal).state  >= -100) {
            //Weak
            it.print(x, y, id(mdi_wifi), TextAlign::BASELINE_CENTER, "󰤟");
            ESP_LOGI("WiFi", "Weak");
        } else {
            //Unlikely working signal
            it.print(x, y, id(mdi_wifi), TextAlign::BASELINE_CENTER, "󰤫");
            ESP_LOGI("WiFi", "Unlikely");
        }
      }
</details>

bkaufx, 2021-08-13:
Have you guys tried turning down the wifi power?  Default is 20 but Tasmota says do not set over 17.  I was having serious bootloop issues with the wifi ap on a different smart plug and lowering the wifi power solved my problems.

j-k-olofsson, 2021-08-14:
Removing arduino_version from the device config solved the problem for me (Sonoff basic R3)

0x3333, 2021-08-14:
I don’t have `arduino_version` on my config file. Issue persists. I reverted to 1.16, no issue so far.

Warter21, 2021-09-21:
my problem is gone with the latest version (2021.9.1).

dekamilv, 2021-09-26:
Sonoff basic (from the first ones)
esphome 2021.9.1

WiFi issue soled only by disabling captive portal in wifi settings.