esphome/issues #2311: color temperature change with constant_brightness has no longer a constant brightness

malteger, 2021-08-11:
### The problem

Starting with 1.21.0b1 the brightness of a cwww light during transitions is changing if you change the color temperature with `constant_brightness: true` (happens also for rgbww lights). In esphome 1.20 the brightness has been kept constant while doing the transition. This is most noticeable if you change the temperature from the lowest to the highest value or vice versa. outside of a transition the brightness is constant as in previous versions. 

It looks like both leds (cw and ww) start the transition and in the middle of the transition the light is significantly darker.

### Which version of ESPHome has the issue?

1.21.0b1 

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.3

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

cwww

### Example YAML snippet

```yaml
- platform: cwww
  name: ${bulbname}
  id: cwww_light
  warm_white: output_warm
  cold_white: output_cold
  cold_white_color_temperature: 6500K
  warm_white_color_temperature: 2700K
  constant_brightness: true
```

oxan, 2021-08-11:
Debugging notes;

This happens because the brightness is now constant at the level of the maximum of the cold/warm white values set (i.e. if cold_white = 80%, warm_white = 40%, then brightness is constant at 80% of single-channel output). This was necessary to allow brightness control of the cwww channels (not really necessary for the cwww platform, but required for the rgbww platform to allow white lights to be turned off). Setting a color temperature will always maximize one channel at the white value, so that works fine. However, the transition transitions cold+warm white individually, so the maximum value will dip below the value at the start/end. Not sure about the best solution.

malteger, 2021-08-11:
> Does constant brightness even make sense for the individual cold/warm white controls? Maybe it should only be applied when setting a color temperature.

The controls only show one brightness and one color temperature slider, I can not change the white channels individually. The same applies with rgbww lights in white mode. Otherwise you would be correct, I think there is no point in setting constant brightness if one has individual controls for each channel.

oxan, 2021-08-11:
> The controls only show one brightness and one color temperature slider, I can not change the white channels individually. The same applies with rgbww lights in white mode.

Yeah, I'm aware of that (though non-interlocked RGBWW lights should look like [this](https://user-images.githubusercontent.com/197154/129103804-6a60a74b-7a5b-48ef-a861-9ab764cae484.png)), but internally they still have separate cold/white warm controls.

oxan, 2021-08-12:
See #2152 -- this turned out to be more complicated than I thought.

malteger, 2021-12-11:
As this problem still persists in the latest stable version I think this should be kept open.

dizherui, 2022-01-23:
Let's keep this open

oxan, 2022-01-23:
I would appreciate some input on what I wrote in https://github.com/esphome/esphome/pull/2152 from people that use/care about this feature.

iBobik, 2023-04-24:
I expected this feature will allow me to use 100 W power for 140 W CCT strips, because I does not need to use both channels on full bright, but one of them yes.

So how to configure it to not go above 140% combined brightness?

clinta, 2023-09-01:
Hit by this bug today. I have a cwww light that cannot handle both at 100%. But constant brightness makes transitions terrible. I wonder if this could be solved at at the output component instead of the light. Having an option to limit the combined output level for two outputs.

clinta, 2023-09-05:
It is a shame that https://github.com/esphome/esphome/pull/2152 got stale and closed. I rebased this on the latest dev and just installed it on my light and it's a significant improvement. There is still a bit of brightness changes during transition due to the gamma issues, but it's much better than stock.

hramrach, 2025-01-07:
So the only problem is with non-interlocked RGBWW light. That itself is a workaround for another defect which I do not fully understand.

Maybe non-interlocked light cannot support constant brightness?

hramrach, 2025-01-07:
Also why is gamma applied to the cw-ww mixing?

The crossfade not working with gamma suggests that it should not be used.

That likely means setting the gamma of component lights to 1 when doing temperature change, and to the cwww light gamma when doing brightness change would likely work in general.

There are some edge cases like applying more adjustments while the transition effect is still in progress, and the non-interlocked rgbww lights.

Kubuxu, 2026-09-30:
Today I ran into a similar issue, though not exactly the same: `constant_brightness: false`, but the transition caused a noticeable brightness change.  I resolved this by calibrating gamma more accurately, moving away from the default of 2.8 (to 1.8 in my case).
AFAIK, gamma applies to CWWW LEDs because they are still non-linear, so you need the correct gamma set for the transition to happen at constant brightness. Extrapolating a bit, at the midpoint of the transition, you are moving from 100% WW to 50% WW and 50% CW. If your gamma is miscalibrated, 50% + 50% brightness won't match 100% single-channel brightness.

In my case, gamma=2.8 would show a brightness reduction during a transition.