esphome/issues #2318: Russian font

viktor6, 2021-08-13:
### The problem

Russian font does not work 

### Which version of ESPHome has the issue?

v1.20.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8.6

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

max7219digit

### Example YAML snippet

```yaml
globals:
  - id: display_page
    type: int
    restore_value: no
    initial_value: '1'
 
spi:
 clk_pin: D5                                  # это подключать к входу CLK на max7219
 mosi_pin: D7                              # это подключать к входу DIN на max7219

display:
 - platform: max7219digit
   cs_pin: D6                               # это подключать к входу CS на max7219
   num_chips: 8                        # количество матриц
   intensity: 8                            # яркость матрицы от 0 до 15
   update_interval: 5s
   scroll_speed: 200ms
   id: max7219d
   lambda: |-
          it.intensity(atoi(id(matrixbri).state.c_str()));
          if (id(display_page) == 1) {
            it.printf(0, 0, id(digit_font), "%s", id(pir_sensor).state ? "ON" : "OFF");
            }  else {
              if (id(display_page) == 2) {
              it.strftime(0, 0, id(digit_font), "%H:%M", id(hass_time).now()); 
              } else {
                if (id(display_page) == 3) {
                it.printf(0, 0, id(digit_font), "%.1f°C", id(temp1).state); 
                } else {
                   if (id(display_page) == 4) {
                   it.printf(0, 0, id(digit_font), "%s%%", id(humi1).state.c_str());
                   } else {
                   if (id(display_page) == 5) { 
                     it.printf(0, 0, id(digit_font), "%s", id(matrix_text).state.c_str()); 
                     id(display_page) = 0;
                     }
                   
                   } } } };
          id(display_page) ++;
[... 204 lines cut ...]
[18:49:32][D][homeassistant.sensor:024]: 'sensor.balkon_temperature': Got state 30.60
[18:49:32][D][sensor:127]: 'Kitchen Temp': Sending state 30.60000  with 1 decimals of accuracy
[18:49:32][D][homeassistant.text_sensor:017]: 'sensor.balkon_temperature': Got state '30.6'
[18:49:32][D][text_sensor:015]: 'Kitchen Humidity': Sending state '30.6'
[18:49:32][D][homeassistant.sensor:024]: 'sensor.balkon_temperature': Got state 30.60
[18:49:32][D][sensor:127]: 'Kitchen Temp': Sending state 30.60000  with 1 decimals of accuracy
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/opt/esphome/esphome/api/client.py", line 408, in run
    self._run_once()
  File "/opt/esphome/esphome/api/client.py", line 398, in _run_once
    msg.ParseFromString(raw_msg)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/python_message.py", line 1139, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/python_message.py", line 1206, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/decoder.py", line 619, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/decoder.py", line 572, in _ConvertToUnicode
    value = local_unicode(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 80: 'utf-8' codec can't decode byte 0xd0 in position 80: invalid continuation byte in field: SubscribeLogsResponse.message

jesserockz, 2021-08-14:
The default glpyhs are here: https://github.com/esphome/esphome/blob/5edebaf4686ceb30a4d0440b52d63811e9be6efd/esphome/components/font/__init__.py#L75-L77

For displays you will need to add the russian characters to the `glyphs` key:
```yaml
font:
  - file: "pixelmix.ttf"
    id: digit_font
    size: 8
    glyphs: ...
  - file: "Pixel-UniCode.ttf"
    id: digit_font_unic
    size: 16
    glyphs: ...
```

As for the logs crashing, that''s another issue and not really supported right now.

viktor6, 2021-08-14:
INFO Reading configuration /config/esphome/veneer_display.yaml...
INFO Detected timezone 'EET' with UTC offset 2 and daylight saving time from 27 March 03:00:00 to 30 October 04:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/veneer_display
Processing veneer_display (board: d1_mini; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Retrieving maximum program size /data/veneer_display/.pioenvs/veneer_display/firmware.elf
Checking size /data/veneer_display/.pioenvs/veneer_display/firmware.elf
RAM:   [=====     ]  54.3% (used 44444 bytes from 81920 bytes)
Flash: [=====     ]  45.7% (used 477180 bytes from 1044464 bytes)
========================= [SUCCESS] Took 6.82 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of veneer_display.local
INFO  -> 192.168.0.179
INFO Uploading /data/veneer_display/.pioenvs/veneer_display/firmware.bin (481328 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from veneer_display.local using esphome API
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
INFO Successfully connected to veneer_display.local
[22:50:25][I][app:105]: ESPHome version 1.20.4 compiled on Aug 14 2021, 22:47:56
[22:50:25][C][wifi:499]: WiFi:
[22:50:25][C][wifi:359]:   SSID: [redacted]
[22:50:25][C][wifi:360]:   IP Address: 192.168.0.179
[22:50:25][C][wifi:362]:   BSSID: [redacted]
[22:50:25][C][wifi:363]:   Hostname: 'veneer_display'
[22:50:25][C][wifi:367]:   Signal strength: -67 dB ▂▄▆█
[22:50:25][C][wifi:371]:   Channel: 8
[22:50:25][C][wifi:372]:   Subnet: 255.255.255.0
[22:50:25][C][wifi:373]:   Gateway: 192.168.0.1
[22:50:25][C][wifi:374]:   DNS1: 192.168.0.1
[22:50:25][C][wifi:375]:   DNS2: (IP unset)
[22:50:25][C][spi:081]: SPI bus:
[22:50:25][C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[22:50:25][C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[22:50:25][C][spi:085]:   Using HW SPI: YES
[22:50:25][C][logger:189]: Logger:
[22:50:25][C][logger:190]:   Level: DEBUG
[22:50:25][C][logger:191]:   Log Baud Rate: 115200
[... 88 lines cut ...]
    msg.ParseFromString(raw_msg)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/python_message.py", line 1139, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/python_message.py", line 1206, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/decoder.py", line 619, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/usr/local/lib/python3.7/dist-packages/google/protobuf/internal/decoder.py", line 572, in _ConvertToUnicode
    value = local_unicode(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 80: 'utf-8' codec can't decode byte 0xc2 in position 80: invalid continuation byte in field: SubscribeLogsResponse.message

WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 2 seconds
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 3 seconds
INFO Connecting to veneer_display.local:6053 (192.168.0.179)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 5 seconds
INFO Connecting to veneer_display.local:6053 (192.168.0.179)

closed 2021-12-27 by github-actions[bot]

DavWM2020, 2022-03-01:
Is the problem solved and how?