esphome/issues #2319: Broken fan speed control with PR #2059

jkl1337, 2021-08-14:
### The problem

Unfortunately the changes in PR #2059 broke Tuya fan devices such as the Treatlife DS03. The reason is that this device uses the enum type datapoint for the fan speed and the PR assumes the speed will always be an integer type datapoint. 

### Which version of ESPHome has the issue?

1.21.0b1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

TYWE3S

### Component causing the issue

tuya

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200

tuya:
  id: tuya_dev

light:
  - platform: tuya
    name: DS03 Light
    dimmer_datapoint: 10
    switch_datapoint: 9
    min_value: 10
    max_value: 1000
    id: lgt_ds03
    output_id: tuya_lgt_dev

fan:
  - platform: tuya
    name: DS03 Fan
    switch_datapoint: 1
    speed_datapoint: 3
    speed_count: 4
    id: fan_ds03
```

### Anything in the logs that might be useful for us?

```txt
Attempt to set datapoint 3 with incorrect type
```

nuttytree, 2021-08-18:
Do we know if some devices use integer and some enum or are all of them enum?  If they are all enum I can get a PR up quickly to change that.  If it is a mix we might have to add a config option for what type the device is using.

nuttytree, 2021-08-19:
I looked back through the history of the code for the Tuya fan component and it looks like back when the Tuya code required building the datapoint instead of having methods to send them it was building an enum datapoint for the speed.  I will get a PR up this morning to fix this.

closed 2021-08-24 by jesserockz