esphome/issues #2321: supported_color_modes changed with ESPHome 1.20.4 and HA 2021.8.6

dylan09, 2021-08-15:
### The problem

After updating my ESPHome driven RGBWW Lights to ESPHome 1.20.4 and update Home Assistant to version 2021.8.6 the attribute `supported_color_modes`has changed to only `rgbww``

Before the updates my lights have:
```
supported_color_modes:
  - color_temp
  - hs
  - rgbw
```

Now the only supported model is `rgbww`.
Before the update the settings dialog shows me a brightness slider, color wheel and a slider for white color temperature (in mired).
Now I have a brightness slider, color wheel and two sliders for cold-white and warm-white.

And in HomeKit I couldn't change color or white color temperature. Only brightness.
I have the color wheel and the temperature wheel in HomeKit. But both are without any function.

Is it possible to get back the old modes? I would like to have the possibility to change white color temperature using the old way?

BTW. I had to rewrite/change all my scenes because of the changed color modes.

Here is the light configuration:
```
light:
  - platform: rgbww
    name: ${name_prefix}
    id: ${id_prefix}
    red: out_red
    green: out_green
    blue: out_blue
    warm_white: out_warm_white
    cold_white: out_cold_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2200 K
    color_interlock: true # avoids simultaneous RGB and W/W
    default_transition_length: 500ms
    restore_mode: ALWAYS_OFF
``` 
 

### Which version of ESPHome has the issue?

1.20.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.6

### What platform are you using?

ESP8266

### Board

esp_wroom_02 

### Component causing the issue

light

oxan, 2021-08-15:
> And in HomeKit I couldn't change color or white color temperature. Only brightness.

This probably is https://github.com/home-assistant/core/pull/54595, will be fixed in HA 2021.8.7.

> I would like to have the possibility to change white color temperature using the old way?

ESPHome v1.21, which will be released in a few days, will restore this (as long as the color interlock is enabled).

> BTW. I had to rewrite/change all my scenes because of the changed color modes.

This is probably caused by https://github.com/home-assistant/core/issues/54216, not much ESPHome can do about that unfortunately.

closed 2021-08-15 by oxan

dylan09, 2021-08-15:
Ok. Will wait until next releases and test.
Thank you.