esphome/issues #2324: The number template cannot be discovered by the home assistant

jack9603301, 2021-08-16:
### The problem

The number template cannot be discovered by the home assistant

No mqtt declaration appears to have been issued to ha

### Which version of ESPHome has the issue?

v1.20.1 

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.8.4

### What platform are you using?

ESP8266

### Board

NODEMCU ESP8266

### Component causing the issue

MQTT Number, Number

### Example YAML snippet

```yaml
number:
  - platform: template
    name: "Template number"
    update_interval: never
    min_value: 0
    max_value: 100
    step: 1
```

jesserockz, 2021-08-18:
I don't use MQTT, but I set up the mqtt addon and an esphome node with a template number and the HA discovery info was successfully sent to mqtt..

jack9603301, 2021-08-18:
> I don't use MQTT, but I set up the mqtt addon and an esphome node with a template number and the HA discovery info was successfully sent to mqtt..

Can the configuration be given?

jesserockz, 2021-08-18:
I deleted it after it worked but same as you posted for number, and a very basic mqtt block with broker, username and password. That was all.

jack9603301, 2021-08-18:
I accidentally tried number during a configuration, but unfortunately, my ha did not find number

jack9603301, 2021-09-04:
[image]

jack9603301, 2021-09-05:
@jesserockz

jesserockz, 2021-09-05:
This issue was fixed in https://github.com/esphome/esphome/pull/2102

closed 2022-01-21 by github-actions[bot]